betaflight/betaflight #10511: Altitude hold mode

svefro, 2021-01-25:
### Is your feature request related to a problem? Please describe.
Alt hold mode removed from Betaflight

### Describe the solution you'd like
Re-implementation of a simple altitude hold mode based on Baro and or GPS.
This mode do not need to be as accurate as a dji drone but something that can be used to park the copter at a set altitude when fiddling around with fpv googles or something else.

I assume there already is a altitude controller since you have GPS rescue.
The quad i want alt hold on has both an internal barometer and a gps so both can be used.

Basicaly if gps rescue works, a simple alt hold mode should be easy to implement for those that are familiar with the code base.
I think alt hold mode should require Angle or Horizon mode active at the same time since enabling alt hold on it's own could cause "undefined behavior" in other modes :p

### Describe alternatives you've considered
i switched to iNav to get altidude hold but there are a lot more features i miss in inav than i betaflight so i would like to stick with betaflight.

mikeller, 2021-01-25:
This has been discussed extensively in the past - please search in past issues before opening new ones.

Closing as duplicate of #6675.

closed 2021-01-25 by mikeller

svefro, 2021-01-25:
I opened a new one since the old ones are closed.

I do not understand how you can close a feature request just like that.
It should remain open until someone has the time to implement it. unless it is against the greater good.

 it is clearly a feature that is wanted by the community.

I start to get a picture of why there is so many forks of cleanflight if this is the way feature suggestions is handled.

mikeller, 2021-02-07:
@svefro: This issue was closed because it is a duplicate of an existing issue. Discussion on this topic should happen on the original issue.