betaflight/betaflight #10515: Matek H743-Min

RRHN, 2021-01-25:
### Describe the bug

https://user-images.githubusercontent.com/77997843/105775865-d6814b80-5f67-11eb-917b-f6cc4ccf9f93.mp4

Hello

I do have an issue with my new Matek H743-Mini FC which I would like to use in a Fix Wing-Setup together with an TBS-Crossfire-Nano RX
The FC hat has the BF Firmware 4.2.3 installed, the BF Configurator is 10.7.0 and the Crossfire RX also with the newest FW installed(3.78).
I connected the Receiver to 8v, 5v and 4.5V.

I am able to get the signal from the Receiver incl the Telemetry and I can move the Servos and also can start the Motor so nearly everything is working.

The issue is that if I make any setting in the FC like OSD/PIDs/CLI or calibration and then store the setting the connection to the receiver is gone.
After the the Configurater restarts the settings are done but I have no contact to the receiver anymore. I need restart the FC totally by taking away all the power and power it up again.
This do not only accrue if I am using the Windows Betaflight Confugurater but also if I make the setting over the OSD with Stick commands.
I can fly my Wing but its a big Pain if I make Inflight calibration, which will then cuts off the receiver in flight.

I even tried to flash to the BF 4.3 FW but it don't like the version of my Configurator(can't find a newer), but I could also check that the same problem occurs with this FW.

Using a Frsky RS Receiver won't show this issue.

Is there a solution for this issue????

### To Reproduce

### Expected behavior

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
```
```
PASTE THE OUTPUT OF 'resource show all' HERE
```

### Setup / Versions
 
 - Flight controller: 
 
 - Other components:
 
 - How are the different components wired up:

haslinghuis, 2021-01-26:
Cannot reproduce on Matek F411 or Matek F722 Mini with Crossfire Nano (BF 4.3 / BFC 10.8).

Please provide a `diff` and `resource show all` as asked while creating this issue.

Auster, 2021-02-15:
Hi all,
I have the same issue with Matek H743-slim. It seems like this issue related to Matek H7 boards.
Tested with Serial 6 and Serial 2. The connection with RX dropped right after pressing the `Save` button.

```
# diff

# version
# Betaflight / MATEKH743 (M743) 4.2.3 Sep 27 2020 / 23:17:06 () MSP API: 1.43

# start the command batch
batch start

board_name MATEKH743

# feature
feature -AIRMODE
feature GPS

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# map
map TAER1234

# serial
serial 1 64 115200 57600 0 115200
serial 2 2 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# aux
aux 0 0 3 1700 2100 0 0
aux 1 2 0 1700 2100 0 0
aux 2 46 3 1700 2100 0 0
aux 3 13 2 1700 2100 0 0
aux 4 28 0 1300 1700 0 0
aux 5 35 2 1300 1700 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 14 23 27 29
vtxtable powerlabels 25 100 400 800

# master
set acc_calibration = 68,10,-229,1
set mag_hardware = NONE
set serialrx_provider = CRSF
set min_throttle = 1070
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 480
set failsafe_procedure = GPS-RESCUE
set align_board_roll = 180
set align_board_yaw = 180
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set osd_vbat_pos = 2090
set osd_link_quality_pos = 2081
set osd_tim_1_pos = 2455
set osd_tim_2_pos = 2391
set osd_remaining_time_estimate_pos = 2424
set osd_flymode_pos = 2529
set osd_vtx_channel_pos = 2465
set osd_current_pos = 2230
set osd_mah_drawn_pos = 2198
set osd_gps_speed_pos = 2305
set osd_gps_lon_pos = 2545
set osd_gps_lat_pos = 2533
set osd_gps_sats_pos = 2072
set osd_home_dir_pos = 2474
set osd_home_dist_pos = 2477
set osd_flight_dist_pos = 2409
set osd_altitude_pos = 2337
set osd_warnings_pos = 14793
set osd_avg_cell_voltage_pos = 2122
set osd_nvario_pos = 2273
set vtx_band = 5
set vtx_channel = 4
set vtx_power = 3
set vtx_freq = 5769

profile 0

rateprofile 0

# end the command batch
batch end
```

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 3
A01: MOTOR 4
A02: FREE
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: FREE
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: BEEPER
B00: MOTOR 1
B01: MOTOR 2
B02: GYRO_EXTI
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: SERIAL_TX 4
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_BATT
C01: ADC_CURR
C02: FREE
C03: FREE
C04: ADC_EXT
C05: FREE
C06: FREE
C07: FREE
C08: SDIO_D0
C09: SDIO_D1
C10: SDIO_D2
C11: SDIO_D3
C12: SDIO_CK
C13: FREE
C14: FREE
C15: GYRO_CS 1
D00: FREE
D01: FREE
D02: SDIO_CMD
D03: FREE
D04: FREE
D05: SERIAL_TX 2
D06: SERIAL_RX 2
D07: SPI_MOSI 1
D08: SERIAL_TX 3
D09: SERIAL_RX 3
D10: PINIO 1
D11: PINIO 2
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: USB_DETECT
E03: LED 1
E04: LED 2
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: GYRO_CS 2
E12: SPI_SCK 4
E13: SPI_MISO 4
E14: SPI_MOSI 4
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE
G00: FREE
G01: FREE
G02: FREE
G03: FREE
G04: FREE
G05: FREE
G06: FREE
G07: FREE
G08: FREE
G09: FREE
G10: FREE
G11: FREE
G12: FREE
G13: FREE
G14: FREE
G15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH1: BEEPER
TIM3:
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM4: FREE
TIM5:
    CH1: MOTOR 3
    CH2: MOTOR 4
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15: FREE
TIM16: FREE
TIM17: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 1
DMA1 Stream 1: MOTOR 2
DMA1 Stream 2: MOTOR 3
DMA1 Stream 3: MOTOR 4
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

mikeller, 2021-02-17:
From Slack:

mikeller￼  2 days ago
Yes we can. Thanks for the config and resource list - all looks normal. It looks like the connection to the RX is dropped on reboot of Betaflight. Can you try going into configurator with RX connected, go to CLI, type ‘exit’ and Enter, and see if this drops the connection as well? (edited) 
￼￼
Ilya Makarov  1 day ago
Hi @mikeller,
After the software restart, RX goes back to normal. Usually, I did save command in CLI instead of exit to temporary fix the problem.
Now I flashed BF 4.3 (aeb2a4f38). The problem still here but became floating. Usually, save operations works well from the OSD menu and configurator. But I reproduced a problem a few times from the OSD menu (VTX settings save & exit), and I can’t reproduce it from the configurator at all.
Previously, in 4.2.3, any save operation, from the configurator, broke RX.

mikeller, 2021-02-17:
So this only happens when you click a 'Save' button in configurator that does not reboot the flight controller? I.e. if you change a setting and then click a 'Save & Reboot' button the RX is connected after the reboot?

Auster, 2021-02-17:
@mikeller yes, FC still up and I can proof it by huge uptime in CLI (>100 sec, for example).

mikeller, 2021-03-21:
@RRHN, @Auster: Does any one of you guys have a scope to have a look at what the UART levels do when clicking 'Save'?

Or else, @MATEKSYS do you have TBS hardware to be able to investigate this?

MATEKSYS, 2021-03-21:
@mikeller  We don't have TBS systems in hands. 

I recall we got similar report on H743-SLIM from other user.  FYI
******The flight controller reboots every-time I click the save button as well as randomly; has issues with Betaflight FW4.2.0, 4.2.3 and 4.3.0dev.  I tested the ESC (Zeez Design 4-in-1 ESC) with 2 other flight controllers and there are no issues.
It seems that the rebooting happens every time I save, write to EEPROM. The radio becomes unresponsive and then I have to do a hard power reset. ********

We checked PWM output with oscilloscope.   there is a short interrupt of PWM output when clicking save. 
It happens on F405/F722 FC also,  but the interrupt period on F405/F722 is much shorter than on H7,  ESC connected to F405/F722 can't detect the short interrupt.

mikeller, 2021-03-21:
@MATEKSYS:

> We checked PWM output with oscilloscope. there is a short interrupt of PWM output when clicking save.

Yes, PWM output is disabled while the EEPROM is saved. But I am not sure how this should influence the TBS hardware that is connected to an UART. Or am I overlooking something here?
I am more worried about the UART TX going low during this time, or something similar.

haslinghuis, 2021-04-22:
@MATEKSYS any news on this.
@mikeller I have no delay on a Matek F411 or F722 with Crossfire using save, the sliders keep moving as I move the sticks

closed 2021-05-30 by github-actions[bot]