betaflight/betaflight #10519: ADD SUPPORT TCA9548A CHIP FOR İ2C İSSUES

luckyluckhcccp, 2021-01-28:
### Is your feature request related to a problem? Please describe.
 i have i2c problems. im trying connect two external sensor (bn880 & Bme280). but i have one i2c pad.

### Describe the solution you'd like
if we can use TCA9548A i2c multiplier we can solve it.

### Describe alternatives you've considered
im using  Tmotor Pacer F7 fc. there is one i2c pad. icant find any different solution.

etracer65, 2021-01-28:
I2C is a multi-device bus. No need to use any "multiplier" chip. You just connect the SDA/SCL for each device to the appropriate pad on the flight controller. The main thing to worry about is pull-up termination. SCL and SDA must have an external pull-up (and only one). So make sure both devices don't have pull-ups. For more details google about I2C and pull-up resistors.

luckyluckhcccp, 2021-01-28:
thnksfor interest @etracer65 . ill try saturday and write again.

etracer65, 2021-01-29:
There is nothing to report. I2C works as I've described. There is no problem connecting multiple devices to the SDA/SCL pads.

closed 2021-01-29 by etracer65