betaflight/betaflight #10521: GPS uses Auto Baud even when GPS Auto Baud is disabled.

Max-Plastix, 2021-01-30:
### Describe the bug

Using an oscilloscope, I see data from the FC to the GPS cycling through different bitrates in order, even when Auto Baud is disabled.

### To Reproduce

Under Ports: Select a UART for Sensor Input: GPS.  Set the Baud Rate to one fixed value (not AUTO), such as 115200.
Under Configuration:
Enable GPS
Disable "Auto Baud"

### Expected behavior

The FC should send data at ONLY the selected bitrate.

### Flight controller configuration

```
# version
# Betaflight / STM32F7X2 (S7X2) 4.2.6 Jan  5 2021 / 19:08:42 (a4b6db1e7) MSP API: 1.43
# config: manufacturer_id: EXUA, board_name: EXF722DUAL, version: c26e0c3f, date: 2019-10-23T13:46:31Z

# start the command batch
batch start

board_name EXF722DUAL
manufacturer_id EXUA

# name: RUN HD3

# resources
resource SERIAL_TX 5 NONE
resource SERIAL_RX 5 NONE

# feature
feature -AIRMODE
feature GPS
feature TELEMETRY

# serial
serial 0 2 115200 115200 0 115200
serial 1 0 115200 57600 0 115200
serial 2 1 115200 57600 0 115200
serial 3 64 115200 57600 0 115200

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 2 900 1300 0 0
aux 2 13 3 1700 2100 0 0
aux 3 26 0 1300 2100 0 0
aux 4 28 0 1700 2100 0 0
aux 5 31 6 1700 2100 0 0
aux 6 47 2 1300 1700 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# master
set gyro_lowpass2_hz = 300
set gyro_to_use = SECOND
set dyn_lpf_gyro_min_hz = 240
set dyn_lpf_gyro_max_hz = 600
set acc_trim_pitch = -12
set acc_trim_roll = -30
set acc_calibration = -20,-23,-1,1
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1000
set max_check = 2000
set rssi_channel = 12
set serialrx_provider = CRSF
set dshot_idle_value = 300
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = WAAS
set gps_auto_config = OFF
set pid_process_denom = 2
set osd_cap_alarm = 1000
set osd_tim1 = 1537
set osd_rssi_pos = 2497
set osd_tim_1_pos = 2543
set osd_flymode_pos = 2060
set osd_mah_drawn_pos = 2537
set osd_craft_name_pos = 2048
set osd_gps_speed_pos = 192
set osd_gps_lon_pos = 49
set osd_gps_lat_pos = 32
set osd_gps_sats_pos = 96
set osd_home_dir_pos = 256
set osd_home_dist_pos = 224
set osd_altitude_pos = 128
set osd_power_pos = 185
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 2476
set osd_esc_tmp_pos = 153
set osd_esc_rpm_pos = 25
set osd_core_temp_pos = 185
set gyro_2_align_yaw = 900
set name = RUN HD3

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 84
set dyn_lpf_dterm_max_hz = 204
set dterm_lowpass2_hz = 180
set anti_gravity_gain = 3000
set feedforward_transition = 60
set p_pitch = 56
set i_pitch = 95
set d_pitch = 39
set p_roll = 48
set i_roll = 90
set d_roll = 34
set p_yaw = 65

rateprofile 0

# rateprofile 0
set roll_rc_rate = 75
set pitch_rc_rate = 75
set yaw_rc_rate = 75
set tpa_rate = 40

# end the command batch
batch end
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: CAMERA_CONTROL
A01: FREE
A02: FREE
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: GYRO_CS 1
B00: FREE
B01: FREE
B02: GYRO_EXTI
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: PINIO 3
B09: FLASH_CS
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: ADC_CURR
C03: GYRO_CS 2
C04: LED 1
C05: FREE
C06: MOTOR 2
C07: MOTOR 4
C08: MOTOR 1
C09: MOTOR 3
C10: SERIAL_TX 4
C11: SERIAL_RX 4
C12: FREE
C13: PINIO 1
C14: PINIO 2
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5:
    CH1: CAMERA_CONTROL
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 
 - Flight controller: EXF722DUAL included in GEPRC's Run HD3 BnF Drone, from RDQ
 
 - Other components: TBS M8 GPS
 
 - How are the different components wired up:   TBS M8 GPS attached to UART1 (which has incorrectly reversed silkscreen for Tx/Rx)

I discovered this while trying to narrow down communication problems due to the incorrect silkscreen/documentation on UART1.  The Tx/Rx is reversed, but I wanted to hold the bitrate at a fixed value for observation/logging.  It appears I can not, and it will always mix in other bitrates.

etracer65, 2021-02-16:
I took a look at this and the logic seems to be working as expected.

What you are seeing is not the "auto-baud" activity. Instead you are seeing the normal UBLOX initialization. Since we don't know what baud rate the device is actually configured for, the initialization logic loops through all the available  baud rates and for each one sends the command to **configure the device to the desired baud rate** under the expectation that the device is listening to at least one of the baud rates. So say you configured 57600 baud. The initialization logic will set to 9600 baud, send the command to configure the GPS to 57600, set the baud to 19200, send the command to configure to 57600, etc. It loops through the options one time and then sets to the desired rate from that point forward.

If the GPS module is still not responding or it was previous working and then stopped responding, then the initialization will try again and repeat the above.

Where the `gps_auto_baud` comes into play is that if it's enabled, the "desired baud rate" will switch to the next possible one before repeating the initialization. So in the example above it will try again with 115200.

Max-Plastix, 2021-02-17:
Thank you for the thoughtful and detailed explanation of how this initialization works.

I misunderstood auto-baud, thinking it was similar to the uBlox U-Center autobaud (try them all; match the GPS and then stay there).
Thanks.

closed 2021-02-17 by Max-Plastix