betaflight/betaflight #10523: Support CRSF binding from CLI

Peter-Pichler, 2021-01-31:
### Is your feature request related to a problem? Please describe.

I'm always frustrated when I need to bind/update my Rx firmware after an update of the Tx firmware and the Rx bind button is not accessible through heatshrink.

### Describe the solution you'd like

As available for e.g. Spektrum receivers and described in "Support CRSF binding from CLI #8458" - please provide a CLI command to put the CRSF receiver into the correct mode (simulate bind button press).

### Describe alternatives you've considered

Disassemble all my quads every time, cut open the heatshrinked receivers, try many times to press the bind button, use tweezers...

Request #8458 has been closed due to inactivity but with the firmware update related problems caused by CRSF, it would be highly appreciated to get receivers into bind mode like other brands!

etracer65, 2021-01-31:
Feel free to provide specifications on how the CRSF protocol implements binding.

mikeller, 2021-02-01:
Closing as duplicate of #10523.

Please look at existing issues and read the discussion on them before wasting time opening duplicates.

closed 2021-02-01 by mikeller