betaflight/betaflight #10524: Canging resource Led Strip pin to PINIO to trigger causes lockup on the triggered device.

carbendi, 2021-02-02:
### Describe the bug
I resource LED STRIP pin to PINIO 1 and set up the PINIO BOX so I can trigger gopro recording from the radio using USER 1.
My problem is, if the gopro rec button is connected to PINIO 1 gopro does not power up properly when i plug the battery. It powers up only if i remove the power cable from gopro and plug it again. In the previous versions of betaflight (anything before 4.3) with the same setup didn't have this problem.

### To Reproduce
resource LED_STRIP none
pinio_config = 1,1,1,1
pinio_box = 40,255,255,255
resource PINIO 1 A08 (previous led strip pin)

Modes / User 1 Aux5 yellow line from 900 to 1600

Naked gopro;
 power from 9v pin
 ground from ground
 trigger from led strip pin
 
 Plug naked gopro connector (power ground signal)
 Plug lipo to quad 
 If auto start anabled on gopro connecting mode and ground  no llight from gopro it just freeses. 
 If no modding done to naked gopro, GoPro's led light up but it stays that wat go pro freeses. 
 

### Expected behavior
 If auto start anabled on gopro connecting mode and ground  led should blink 5 times and go pro should turn on.
  If no modding done to naked gopro, GoPro's led should blink 1 time indicating it has power and the power button on the go pro should power it normally. 

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
```

# version
# Betaflight / STM32F411 (S411) 4.3.0 Jan 12 2021 / 12:22:38 (81b0e2672) MSP API: 1.44
# config: manufacturer_id: BEFH, board_name: BETAFPVF411, version: 5e160c8b, date: 2019-12-11T10:26:31Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name BETAFPVF411
manufacturer_id BEFH
mcu_id 004d003e3539511434363532
signature 

# name: Hurricane 85X

# resources
resource LED_STRIP 1 NONE
resource PINIO 1 A08

# serial
serial 0 0 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200

# beeper
beeper -ARMED

# beacon
beacon RX_LOST
beacon RX_SET

# map
map RETA1234

# aux
aux 0 0 0 1675 2100 0 0
aux 1 1 2 1350 2100 0 0
aux 2 13 0 1250 1650 0 0
aux 3 26 5 1300 2100 0 0
aux 4 35 3 1700 2100 0 0
aux 5 40 4 900 1550 0 0

# adjrange
adjrange 0 0 1 900 2100 12 1 0 0
adjrange 1 0 3 900 1600 29 3 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 14 20 26 28
vtxtable powerlabels 25 100 400 500

# master
set gyro_lowpass2_hz = 325
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 100
set dyn_notch_max_hz = 350
set dyn_lpf_gyro_min_hz = 260
set dyn_lpf_gyro_max_hz = 650
set acc_calibration = -53,136,-273,1
set max_check = 1950
set serialrx_provider = GHST
set dshot_idle_value = 600
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set failsafe_delay = 10
set vbat_max_cell_voltage = 436
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 1
set yaw_deadband = 1
set thrust_linear = 25
set osd_warn_rssi = ON
set osd_vbat_pos = 2445
set osd_rssi_pos = 130
set osd_link_quality_pos = 2178
set osd_tim_2_pos = 2455
set osd_flymode_pos = 2114
set osd_throttle_pos = 2369
set osd_vtx_channel_pos = 2401
set osd_current_pos = 2103
set osd_mah_drawn_pos = 376
set osd_craft_name_pos = 2056
set osd_debug_pos = 33
set osd_pidrate_profile_pos = 2146
set osd_warnings_pos = 14602
set osd_avg_cell_voltage_pos = 2413
set osd_esc_tmp_pos = 2136
set osd_flip_arrow_pos = 2189
set osd_core_temp_pos = 2168
set osd_log_status_pos = 2092
set osd_stick_overlay_left_pos = 2208
set osd_stick_overlay_right_pos = 2231
set osd_stat_max_spd = OFF
set osd_stat_used_mah = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_esc_rpm = ON
set osd_stat_max_fft = ON
set osd_stat_total_time = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 6
set vtx_power = 4
set vtx_freq = 5843
set vcd_video_system = NTSC
set pinio_box = 40,255,255,255
set gyro_rpm_notch_harmonics = 1
set name = Hurricane 85X

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dterm_lowpass_hz = 120
set dterm_lowpass2_hz = 195
set iterm_relax_cutoff = 12
set p_pitch = 60
set d_pitch = 39
set f_pitch = 143
set p_roll = 55
set d_roll = 36
set f_roll = 135
set p_yaw = 60
set f_yaw = 135
set abs_control_gain = 5
set d_min_roll = 0
set d_min_pitch = 0
set d_min_advance = 25

profile 1

# profile 1
set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dterm_lowpass_hz = 120
set dterm_lowpass2_hz = 195
set iterm_relax_cutoff = 12
set p_pitch = 69
set d_pitch = 39
set f_pitch = 143
set p_roll = 63
set d_roll = 36
set f_roll = 135
set p_yaw = 50
set i_yaw = 85
set f_yaw = 135
set abs_control_gain = 5
set d_min_roll = 0
set d_min_pitch = 0
set d_min_advance = 25

profile 2

# profile 2
set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dterm_lowpass_hz = 120
set dterm_lowpass2_hz = 195
set p_pitch = 69
set d_pitch = 39
set f_pitch = 143
set p_roll = 63
set d_roll = 37
set f_roll = 135
set p_yaw = 68
set f_yaw = 135
set d_min_roll = 0
set d_min_pitch = 0
set d_min_advance = 25
set ff_interpolate_sp = AVERAGED_2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set thr_mid = 0
set rates_type = ACTUAL
set roll_rc_rate = 18
set pitch_rc_rate = 18
set yaw_rc_rate = 21
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 85
set tpa_breakpoint = 1750

rateprofile 1

# rateprofile 1
set thr_mid = 40
set roll_rc_rate = 144
set pitch_rc_rate = 144
set yaw_rc_rate = 144
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 64
set pitch_srate = 64
set yaw_srate = 64
set tpa_breakpoint = 1650

rateprofile 2

# rateprofile 2
set thr_mid = 0
set rates_type = ACTUAL
set roll_rc_rate = 18
set pitch_rc_rate = 18
set yaw_rc_rate = 18
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 85
set pitch_srate = 85
set yaw_srate = 85
set tpa_breakpoint = 1750

rateprofile 3

# rateprofile 3
set thr_expo = 30

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 1

# save configuration
save
```
PASTE THE OUTPUT OF 'resource show all' HERE
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FLASH_CS
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: PINIO 1
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: ADC_BATT
B01: ADC_CURR
B02: BEEPER
B03: SERIAL_TX 11
B04: MOTOR 1
B05: MOTOR 2
B06: MOTOR 3
B07: MOTOR 4
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: LED 2
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2: SERIAL_TX 11
TIM3:
    CH1: MOTOR 1
    CH2: MOTOR 2
TIM4:
    CH1: MOTOR 3
    CH2: MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: MOTOR 4
DMA1 Stream 4: MOTOR 1
DMA1 Stream 5: MOTOR 2
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 <!-- Specify your flight controller model (what type is it, where was it bought from, ...) -->
 - Flight controller: F405 2-4S AIO betafpv web site. But it behaves same way with my other Jhemcu GHF420AIO and jhemcu ghf411aio

 <!-- Specify other components attached to the flight controller (RX, VTX, brand / model for all of them, firmware version where applicable...) -->
 - Other components:
 <!-- Details about how all is wired -->
 - How are the different components wired up:

<!-- Add any other context about the problem that you think might be relevant here. -->

mikeller, 2021-02-08:
@carbendi: This sounds more of a GoPro problem than a Betaflight problem. Can you test what is happening when you power up the GoPro while the trigger input is directly connected to ground?

closed 2021-03-17 by github-actions[bot]

YuriDenison, 2022-02-05:
Hello.
Experiencing the same issue with BF 4.3 RC3 on various FCs:

FC:
- GEPRC GEP-F722-45A AIO
- IFlight Beast H7 55A AIO

On BF 4.2.9 / 4.2.11 everything worked fine, GoPro was successfully powered and turned on once battery is attached.
On BF 4.3 Gopro is powered (red led is on) but doesn't turn on at all.

Gopro - [geprc naked 8](https://geprc.com/wp-content/uploads/2021/05/geprc-gopro8-naked-User-Manual-Set-Guide.pdf).

Also tested on MAMBA DJI F722 MK2.
On 4.2.9 experienced the same issue, but after soldering `BF Led` pins on the board everything worked fine.
With 4.3RC3 also works fine on this FC.

YuriDenison, 2022-02-05:
@mikeller [mikeller](https://github.com/mikeller) Could you please take a look one more time into this issue?