betaflight/betaflight #10527: Request: Increase PID_GAIN_MAX to UINT8 max (255)

harry1453, 2021-02-02:
### Is your feature request related to a problem? Please describe.

I know this is quite a niche issue, but on one of my quads I'm now at 200 P gain on all 3 axis.
The quad is a Diatone Taycan weighing in at 600g with relatively lightweight / small 3 inch props, so it takes a lot of gain to get it to handle well.

As an aside, in a testament to RPM filtering, I have my D gains around 180 with D min disabled and my motors come down cold.

### Describe the solution you'd like

Raise `PID_GAIN_MAX` to 255 as all of the gains are stored as `uint8_t` so it would make sense to allow them to max out at 255.

If this gets accepted backporting the fix to 4.2 would be great but I know that's unlikely.

### Describe alternatives you've considered

Obvious alternative is running my own fork of the firmware but that's a pain.

Another alternative for quads that need massive (>255) gain would be to add a configurable constant multiplier to the PID sum. For example, the multiplier is 1, and if I run out of headroom for gain, I can half all of my PID coefficients and increase the multiplier to 2, resulting in the exact same PID sum with more headroom for adjusting individual coefficients. However this solution is a bodge so I'm not asking for it.

etracer65, 2021-02-02:
No quad needs gains this high. You've taken your tuning in the wrong direction.

harry1453, 2021-02-02:
Well, the quad still feels "loose" with 200 P gains, and is showing no signs of having too much gain at 200. Unfortunately this quad has no black box so I can't quantify "loose". With default gains the quad was extremely wobbly and increasing P and D has improved this but it's still not ideal, and ideally I'd like to find the point at which it is overtuned.

etracer65, 2021-02-02:
There's a point where the little motors/props are incapable of tracking setpoint based on the excessive weight. Increasing gains further will not solve that as the motors are already maxing out.

etracer65, 2021-02-02:
Maxing out the motor thrust. There is no more power to move the quad faster. You could set P to 1000 and it will make no difference.

If you want quicker response then increase F gains - not P. Also really high D gains dampen (slow down) the response.

harry1453, 2021-02-02:
Okay. Without blackbox there's no easy way to know unfortunately. And I should've been more clearer, when I said response I meant response to PID error, not to stick movements. The quad isn't really as stable as I'd like when just flying forward with no stick input. I understand about D gain, it's there because I'm increasing D proportionally with my P increases. I might try decreasing it a bit.

closed 2021-03-05 by harry1453