betaflight/betaflight #10531: HAKRC MINI DJI Stack F7 (MATEKF722SE ) onboard memory (dataflash) not detected

N4V1G4T0R, 2021-02-04:
### Describe the bug

On the 20x20mm HAKRC MINI DJI Stack F7 the Onboard Ram is not detected (MATEKF722**SE** is factory default firmware)

### To Reproduce

Just make a clean flash with the target MATEKF722SE on the HAKRC MINI DJI Stack F7 (which actually is a SE)

### Expected behavior
This is just a configuration issue. I already contacted the supplier to fix that, but until then you'll have to fix that manually. Following CLI Parameters will fix it:

### Flight controller configuration

```
resource SDCARD_CS 1 NONE
resource FLASH_CS 1 D02
set blackbox_device = SPIFLASH
set flash_spi_bus = 3
```

### Setup / Versions
 
 - Flight controller: HAKRC MINI DJI Stack F7 
 
 - Other components: stack with 4in1 ESC

etracer65, 2021-02-04:
The real MATEKF722SE has an SDCARD, not a flash chip. The solution is not to corrupt the MATEKF722SE target, but that HAKRC should stop producing bad clones and instead submit their own target that works for the board they're selling.

N4V1G4T0R, 2021-02-04:
true, true ...

still, ppl using this clone will need to change that manually, and now they know how ;-)

closed 2021-02-04 by N4V1G4T0R

mikeller, 2021-02-09:
@N4V1G4T0R:

> still, ppl using this clone will need to change that manually

Maybe that's a sign that they should buy hardware from other suppliers who actually put effort into maintaining targets for their products instead. :wink: