betaflight/betaflight #10536: Adjusting ESC Current Sensing and Voltage sensing when using ESC telemetry

BikeGuy1992, 2021-02-08:
### Is your feature request related to a problem? Please describe.
I currently use APD ESCs on an xclass build since they have been the most proven ESC to not fail compared to other brands targeted at xclass fliers. The biggest issue I have is the ESC telemtry for current sensing. The APD ESCs are not accurate at all. I have seen people asking APD about this and APD claims they are accurate since they test each one before being sent out. But others have said it isn't correct. One person recorded around 2000mAh used up in the OSD on one flight, then charged his batteries back up and the charger showed 3000mAh put back in. APD has said they are going to put a firmware out at some point that allows their ESC configuration tool to help fix this issue to adjust the ESCs to more accurately read current being pulled by the ESCs/motors so you don't over draw your packs. The problem with that is, it's been over a year and a half when doing research on the very subject when APD announced that. 

### Describe the solution you'd like
It got me thinking if APD is taking too long to get the firmware made for their ESCs and their ESC configuration tool, maybe Betaflight might have a solution to this problem. I know if you use an ESC that has a ADC current sensor on it, you can adjust the scaling and offset for it. So why shouldn't I be able to adjust the ESC telemetry data in Betaflight to get more accurate results? To my disappointment, you can't change it in Betaflight. In the power and battery tab in the configurator, it shows the number of ESCs being uses when "ESC Sensor" is selected for the current sensing part. But there is no way of being able to adjust this output in Betaflight. 

Would there be a way to implement something similar when adjusting an ADC current sensor (current scale and mAh offset) but for ESC telemtry instead so it isn't completely different to what is currently in Betaflight right now? I have seen that the KISS OSD only uses digital ESC telemetry to work with since they don't even support ADC current sensing at all. And, more importantly, they do have a way that allows the user to be able to adjust the ESC current sensing telemetry to be more accurate. Their way is more simple when in the "mAh correction menu." You adjust the percent of each ESC, but generally you set all four or however many you are using to the same value. The problem with it is it doesn't allow for finite adjustments. You can only change it 10% at a time, not down to 1% increments or even less than that (0.1% increments).

Basically, Betaflight would need to read the ESC telemetry data coming from the ESC (BLHeli_32, KISS/FETTEC, APD) then run it by a scale number to then show the output on the OSD to better accurately show proper current scale numbers and mAh or even wattage being pulled. 

1. ESC telemetry data is sent to FC
2. BF detects and reads telemetry info 
3. BF takes telemetry data (amperage) runs 
    through scale number
4. BF displays adjusted current draw in OSD
5. mAh and watt hours are more accurately 
    interpreted so batteries are not over drawn/ 
    wrecked/destroyed 

The same thing could be implemented for voltage sensing as well when using ESC telemtry, but generally from all the times I've seen people using voltage sensing via ESC telemetry it's close enough to work with (+/- 0.1v)

### Describe alternatives you've considered

Well, the only other option I have considered but it wouldn't work would be to get either a different PDB with a ADC current sensor on it and or change the ESCs. The main issue why the APD ESCs and the PDB doesn't have a physical ADC sensor on it is the amount of power these machines pull. People have tried using PDBs with the shunt resistor needed for current sensing like from Matek for instance, but they'll fall off. The heat that the PDB generates will cause the solder to liquefy again and the shunt resistor will fall off and then the whole rig will loose power. This is why APD doesn't use shunt resistors on the PDB or on the ESCs; it isn't feasible to do nor is it reliable. 

I have looked at the iFlight ESCs but they have been known to not be reliable at all compared to APD. And another company, FlyColor, is also coming out with xclass ESCs that can do current sensing. But, again, no one knows how reliable they're going to be until people start using them in real world applications. 

Please see the two attached images. 

First one is what the current power and battery tab looks like in Betaflight when 'ESC Sensor' is selected.

Second picture is what the OSD menu looks like on a KISS OSD system. This is the menu where you can adjust each individual ESC to get more accurate current sensing values when using an ESC that supports current via to digital telemetry.

[image]

[image]

etracer65, 2021-02-08:
Sensors on the ESC need to be calibrated by the ESC firmware. Just like how BLHeli32 Suite allows you to adjust the calibration of the current sensor. If your ESC vendor is not following up on their commitments perhaps you should try a different brand of ESC.

BikeGuy1992, 2021-02-08:
I gave my reasoning why I can't use a different brand of ESC.

mikeller, 2021-02-08:
@BikeGuy1992: There is no way that Betaflight can do anything to adjust the consumption by calibrating the amperage - in the case of ESC sensors, the consumption is tracked by the ESC and not by Betaflight - so if we wanted to provide a workaround for these broken ESCs then we'd have to add independent calibration for voltage, amperage, and consumption.
I think the right way to handle this is to return the ESCs that are not working correctly to the supplier or manufacturer and get them refunded or replaced with working ones.

BikeGuy1992, 2021-02-08:
Okay, thanks for the technical reply Mike. Appreciate it and wasn't sure the reason behind why Betaflight didn't have it initially.

closed 2021-03-18 by github-actions[bot]