betaflight/betaflight #10544: SPRacingF3 LED strip GUI not work

Kutilak, 2021-02-14:
### Describe the bug

Setup LED strip in GUI dont work. Nothing what I do in it. I use tutorial but it dont work. Also doesnt work SAVE. If I try click on other menu and than click back to LED strip it is always empty. And LED strip still light orange
### To Reproduce

In configuration enable LED strip and then try to setup some strip and save. LED strip still light orange.
### Expected behavior

Able to set color and setup function from GUI (for example battery level indication by LED strip color)
### Flight controller configuration

```
# diff
# Betaflight / SPRACINGF3 (SRF3) 4.0.6 Sep  1 2019 / 01:05:57 (2a64051a2) MSP API: 1.41 / FEATURE CUT LEVEL 6
batch start

feature -RX_PPM
feature -TELEMETRY
feature -RSSI_ADC
feature RX_SERIAL
feature MOTOR_STOP
feature LED_STRIP
beacon RX_LOST
beacon RX_SET
serial 2 64 115200 57600 0 115200
aux 0 0 2 900 1500 0 0
aux 1 1 0 900 1300 0 0
aux 2 2 0 1700 2100 0 0
aux 3 47 5 1700 2100 0 0
set gyro_sync_denom = 4
set acc_calibration = -70,13,-60
set rssi_channel = 14
set serialrx_provider = IBUS
set blackbox_device = NONE
set align_board_yaw = -90
set bat_capacity = 2000
set vbat_warning_cell_voltage = 360
set vbat_scale = 120
set ibatv_scale = 300
set ibatv_offset = 10
set ledstrip_race_color = BLUE
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_vbat_pos = 2519
set osd_rssi_pos = 2282
set osd_ah_pos = 206
set osd_current_pos = 2282
set osd_home_dir_pos = 2282
set osd_home_dist_pos = 2282
set osd_compass_bar_pos = 2091
set osd_altitude_pos = 2282
set osd_pid_pitch_pos = 419
set osd_warnings_pos = 2409
set osd_nheading_pos = 2282
set osd_esc_rpm_freq_pos = 417
profile 2

rateprofile 0

batch end
```
```
# resource show all
IO
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: ADC_BATT
A05: ADC_CURR
A06: MOTOR 1
A07: MOTOR 2
A08: LED_STRIP
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: MOTOR 3
A12: MOTOR 4
A13: FREE
A14: FREE
A15: FREE
B00: FREE
B01: FREE
B02: FREE
B03: LED 1
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: FREE
B11: SERIAL_RX 3
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: GYRO_EXTI
C14: FREE
C15: BEEPER
F00: FREE
F01: FREE
F03: FREE
F04: FREE

DMA:
DMA1 Channel 1: FREE
DMA1 Channel 2: LED_STRIP
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: FREE
DMA1 Channel 7: FREE
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE
```

### Setup / Versions
 
 - Flight controller: SP Racing F3 Acro
 
 - Other components: Default components from Eachine wizard x220 + LED strip 
 
 - How are the different components wired up: LED strip WS2812B data soldered on his pad 

It is 100% not HW problem, because comand: set ledstrip_race_color=BLUE  in CLI change color of strip to BLUE. 
If someone need some help with test or aditional information about setup or problem ask me for it.

mikeller, 2021-02-14:
@Kutilak: This is by design - due to flash space limitations the SPRACINGF3 does not support the full LED_STRIP capabilities - only race mode (all LEDs the same colour) and beacon mode (flashing white), and they need to be configured in the CLI, as explained in the documentation: https://github.com/pkruger/betaflight/blob/master/docs/LedStrip.md#led-strip-profiles

Kutilak, 2021-02-14:
But in older version (before 4.0.0) it work great. I dont understand why it replaced and by what.

Kutilak, 2021-02-14:
@mikeller It is posible to download and compile source code of SPRacingF3 ? If yes, where I find source for it and how to do it ?

etracer65, 2021-02-14:
@Kutilak As stated, the F3 processor ran out of storage space so less important features (like customizable LED strip) were removed in favor of more important flight control features. If you want to make your own version then feel free. The source code for 4.0.6 can be found here: https://github.com/betaflight/betaflight/releases/tag/4.0.6

Note that starting with Betaflight version 4.1 we no longer support F3 flight controllers and as such no support will be provided.

Kutilak, 2021-02-14:
OK. I try to build code myself. Thanks for describe.

closed 2021-03-25 by github-actions[bot]