betaflight/betaflight #10547: 4.2.7 the flight control will not be connected in the betaflight configuration after the firmware is swiped without opening the full disk erase and no restart sequence

Xiang-FPV, 2021-02-14:
### Describe the bug

4.2.7 the flight control will not be connected in the betaflight configuration after the firmware is swiped without opening the full disk erase and no restart sequence

### To Reproduce

The flight control system with parameters adjusted can remote control and take off does not connect to the computer according to boot. In the current version 4.2.6, turn off the full chip erase and no restart sequence, turn off the manual baud rate, brush the stm32f405 or stm32f7x2 firmware, automatically exit the DFU mode after brushing, the flight control indicator flashes, but the COM port is not checked, and the flight control system cannot be connected

### Expected behavior

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
```
###WARNING: NO CUSTOM DEFAULTS FOUND###

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.6 Jan  5 2021 / 19:08:42 (a4b6db1e7) MSP API: 1.43
###ERROR: diff: NO CONFIG FOUND###
# start the command batch
batch start

board_name HIFIONRCF7
manufacturer_id HFOR

# name: GEPRC Mark4

# resources
resource BEEPER 1 C15
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 B04
resource MOTOR 4 B03
resource MOTOR 5 C09
resource MOTOR 6 C08
resource PPM 1 A03
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource I2C_SCL 1 B06
resource I2C_SDA 1 B07
resource LED 1 A15
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource CAMERA_CONTROL 1 B08
resource ADC_BATT 1 C02
resource ADC_RSSI 1 C00
resource ADC_CURR 1 C01
resource PINIO 1 C14
resource PINIO 2 B09
resource FLASH_CS 1 C13
resource OSD_CS 1 B12
resource GYRO_EXTI 1 C04
resource GYRO_EXTI 2 C03
resource GYRO_CS 1 B02
resource GYRO_CS 2 A04

# timer
timer A03 AF3
# pin A03: TIM9 CH2 (AF3)
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer B04 AF2
# pin B04: TIM3 CH1 (AF2)
timer B03 AF1
# pin B03: TIM2 CH2 (AF1)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer C08 AF3
# pin C08: TIM8 CH3 (AF3)
timer A08 AF1
# pin A08: TIM1 CH1 (AF1)
timer B08 AF2
# pin B08: TIM4 CH3 (AF2)

# dma
dma ADC 3 0
# ADC 3: DMA2 Stream 0 Channel 2
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin B04 0
# pin B04: DMA1 Stream 4 Channel 5
dma pin B03 0
# pin B03: DMA1 Stream 6 Channel 3
dma pin C09 0
# pin C09: DMA2 Stream 7 Channel 7
dma pin C08 0
# pin C08: DMA2 Stream 2 Channel 0
dma pin A08 0
# pin A08: DMA2 Stream 6 Channel 0
dma pin B08 0
# pin B08: DMA1 Stream 7 Channel 2

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature LED_STRIP
feature OSD

# beacon
beacon RX_SET

# serial
serial 2 64 115200 57600 0 115200
serial 3 1 115200 57600 0 115200
serial 4 2 115200 115200 0 115200
serial 5 8192 115200 57600 0 115200

# led
led 0 5,5::CTO:13
led 1 6,5::CTO:13
led 2 7,5::CTO:13
led 3 8,5::CTO:13
led 4 9,5::CTO:13
led 5 10,5::CTO:4
led 6 5,7::CB:1
led 7 6,7::CB:1
led 8 7,7::CB:1
led 9 8,7::L:13
led 10 9,7::L:13
led 11 5,8::AT:13
led 12 6,8::AT:13
led 13 7,8::AT:13
led 14 8,8::AT:13
led 15 9,8::AT:13
led 16 5,9::AT:13
led 17 6,9::AT:13
led 18 7,9::AT:13
led 19 8,9::AT:13
led 20 9,9::AT:13
led 21 5,10::AT:13
led 22 6,10::AT:13
led 23 7,10::AT:13
led 24 8,10::AT:13
led 25 9,10::AT:13

# mode_color
mode_color 6 0 3
mode_color 6 1 5

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 1 1300 1700 0 0
aux 2 2 1 1800 2100 0 0
aux 3 46 5 1700 2100 0 0
aux 4 13 2 1700 2100 0 0
aux 5 35 3 1700 2100 0 0
aux 6 49 4 1700 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 200 400 600 1000
vtxtable powerlabels 25 200 400 800 1W

# rxfail
rxfail 9 s 1900

# master
set gyro_lowpass2_hz = 275
set gyro_to_use = SECOND
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_lpf_gyro_min_hz = 220
set dyn_lpf_gyro_max_hz = 550
set acc_calibration = -93,-74,-98,1
set mag_align_yaw = 1800
set mag_hardware = NONE
set baro_bustype = I2C
set baro_i2c_device = 1
set serialrx_provider = CRSF
set adc_device = 3
set blackbox_device = SPIFLASH
set dshot_idle_value = 500
set dshot_burst = ON
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_procedure = GPS-RESCUE
set current_meter = ADC
set battery_meter = ADC
set ibata_scale = 450
set beeper_inversion = ON
set beeper_od = OFF
set small_angle = 90
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_rescue_min_sats = 5
set gps_rescue_allow_arming_without_fix = ON
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 2
set osd_warn_link_quality = ON
set osd_warn_over_cap = ON
set osd_cap_alarm = 1100
set osd_vbat_pos = 2497
set osd_link_quality_pos = 2103
set osd_rssi_dbm_pos = 2136
set osd_tim_2_pos = 2475
set osd_flymode_pos = 2273
set osd_throttle_pos = 2297
set osd_crosshairs_pos = 2285
set osd_ah_pos = 2158
set osd_current_pos = 2518
set osd_craft_name_pos = 2505
set osd_gps_speed_pos = 2488
set osd_gps_sats_pos = 2092
set osd_home_dir_pos = 2081
set osd_home_dist_pos = 2082
set osd_altitude_pos = 2401
set osd_warnings_pos = 14729
set osd_avg_cell_voltage_pos = 2465
set osd_flip_arrow_pos = 2414
set osd_stat_tim_1 = ON
set osd_stat_min_batt = OFF
set osd_stat_battery = ON
set osd_stat_max_esc_rpm = ON
set osd_stat_min_link_quality = ON
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 3
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5806
set max7456_spi_bus = 2
set dashboard_i2c_bus = 1
set pinio_box = 40,41,255,255
set flash_spi_bus = 3
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_sensor_align = CW90
set gyro_1_align_yaw = 900
set gyro_2_spibus = 1
set gyro_2_sensor_align = CW90
set gyro_2_align_yaw = 900
set name = GEPRC Mark4

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 77
set dyn_lpf_dterm_max_hz = 187
set dterm_lowpass2_hz = 165
set i_pitch = 65
set d_pitch = 28
set i_roll = 65
set d_roll = 28
set p_yaw = 38

rateprofile 2

# end the command batch
batch end
```
PASTE THE OUTPUT OF 'resource show all' HERE
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: FREE
A04: GYRO_CS 2
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: LED_STRIP
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: LED 1
B00: MOTOR 1
B01: MOTOR 2
B02: PREINIT 1
B03: MOTOR 4
B04: MOTOR 3
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: CAMERA_CONTROL
B09: PINIO 2
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: GYRO_EXTI
C04: FREE
C05: FREE
C06: SERIAL_TX 6
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: FLASH_CS
C14: PINIO 1
C15: BEEPER
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: LED_STRIP
TIM2: FREE
TIM3: FREE
TIM4:
    CH3: CAMERA_CONTROL
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: LED_STRIP
DMA2 Stream 7: FREE

### Setup / Versions
 
 - Flight controller: 
 
 - Other components:
 
 - How are the different components wired up:

mikeller, 2021-02-14:
@Xiang-FPV: I just tested 4.2.7 on F722, and it works just fine for me.

> `###WARNING: NO CUSTOM DEFAULTS FOUND###`

It looks like you selected the wrong target - flash again (by pressing the boot button when powering up), and make sure to select the correct target (**not STM32F7X2**).

Xiang-FPV, 2021-02-14:
I am a foreign player. In betaflight configuration, I have to use VPN to refresh the firmware directly, so I am used to download the firmware in GitHub and swipe F7 flight control firmware into stm32f7x2, and then enter CIL to import unified targets to swipe flight control firmware. F4 is the same way. Small version updates are used to not turning on the full chip erase, because there is no need to import unified targets repeatedly, and there is no need to spend time adjusting parameters. I guarantee that I did not select the wrong target file when I swipe, because I have been swiping from 4.2.0 to 4.2.6 and there is no problem,until I swipe from 4.2.6 to 4.2.7. After writing firmware, the flight control status light flashes normally, and the flight control electronic self-test Normally, only the flight control is plugged into the computer, and the com serial port is not recognized. There is no response to the remote control. Then I turn on the full chip erase, swipe the firmware to connect with betaflight configuration, then import unified targets, and then import the backup files of the previous flight control system. In this way, flight control can be activated normally. Betalight can be connected, and the remote control responds normally Later, I changed to another plane and didn't open the whole chip to erase the firmware, The above problem occurred again, Once again, I was forced to open the whole chip to erase the firmware

mikeller, 2021-02-15:
@Xiang-FPV: The officially supported way to install Betaflight firmware is through Betaflight configurator - this will add the required hardware configuration to the firmware during the flashing process.
Flashing hex files outside of this may work, but it is not supported, and you are on your own in making this work.

Winky1st, 2021-02-15:
I apologize if I write in the wrong thread but i dont want to open a duplicate thread.
Same happen on KakuteF7, once updated to 4.2.7 there is no more usb connection. (no erase all and no restart-sequence selected)
I updated from 4.2.5 (working perfectly) to 4.27, 
To unlock situation I had to manually press DFU button and re-flash 4.2.6 that is perfectly working.

haslinghuis, 2021-02-15:
@Winky1st Please try again with 4.2.8 (sometimes it's needed to use full chip erase between versions)

Winky1st, 2021-02-17:
> @Winky1st Please try again with 4.2.8 (sometimes it's needed to use full chip erase between versions)
Ty very much 4.2.8 is perfectly working. ty

closed 2021-03-28 by github-actions[bot]