betaflight/betaflight #10549: Cannot connect to serial port HGLRC411

gizdolok, 2021-02-14:
### I just upgraded my HGLRC411 FC from 4.2.5 to 4.2.7. Now I cannot connect to to the FC anymore, the serial port is no longer recognized.

### Upgrade FC from 4.2.5 to 4.2.7.

### Should be able to connect to FC.

### Flight controller configuration

```

```

### Setup / Versions
 
 - Flight controller: 
 - https://www.hglrc.com/collections/fc/products/fd411-flight-controller
 
 - Other components:
 
 - How are the different components wired up:

gizdolok, 2021-02-14:
after installing version 4.2.6, now I can connect to the FC again.

mikeller, 2021-02-14:
@gizdolok: There was a problem with the F411 Unified Target for 4.2.7
. It has been replaced with a fixed target, please try flashing it again.

gizdolok, 2021-02-14:
yeah @mikeller I've just tried again and works now.

gizdolok, 2021-02-14:
Thanks a lot, all good 👍

closed 2021-02-14 by gizdolok