betaflight/betaflight #10556: PID Pitch/Roll ratio slider

og54, 2021-02-16:
### Is your feature request related to a problem? Please describe.
Quads with a high moment of inertia in one axis compared to the other might have issues getting the slider tuning working properly. A long body with a heavier camera up front could need a completely different PD gain on pitch than on roll, while a quad with a more centralized mass where the inertia is more or less the same in both axis would need more similar gains.

### Describe the solution you'd like
A slider that would scale the pitch axis PID compared to the roll axis PID.

### Describe alternatives you've considered
In the current versions of the configurator we have to do pitch axis PID gains more or less individually if our drone has a higher moment of inertia. Or tune them individually (leaving the other axis out of tune), noting every value then feeding them in manually when done

klutvott123, 2021-02-20:
roll/pitch ratio slider is coming https://github.com/betaflight/betaflight/pull/9119

haslinghuis, 2021-02-20:
BFC 10.8 and BF 4.3 will rock the planet

haslinghuis, 2021-02-20:
BFC 10.8 / BF 4.3 will rock the planet!

closed 2021-03-30 by github-actions[bot]