betaflight/betaflight #10557: Save gyro calibration in EEPROM; allow manual only gyro calibration

useroff, 2021-02-16:
### Is your feature request related to a problem? Please describe.
It is known, that the gyro origin offsets measured during the gyro calibration process are temperature dependent.
DJI drones heat their IMU during the (lengthy) calibration process and PX4 firmware allow users to freeze and heat the drones to get the gyro temperature compensation curves.
During a flight they constantly get the gyro sensor temperature readings and then apply the gyro offsets, corresponding to that temperature.
Betaflight does not do that.
It gets gyro offsets during the power-up (when the gyro is cold) and then use those offsets throughout the flight (when gyro is heated), which leads to the gyro drift.
Betaflight users flying the ACRO mode blame the gyro drift to their setups.
For the ANGLE mode pilots the gyro drift is obviously a Betaflight "feature".

### Describe the solution you'd like
The simple solution is to disable forced gyro calibration on power-up, to allow users calibrate the gyro (with the stick commands) when it is heated to the working temperature, then save the gyro calibration data to EEPROM and reuse those data on the subsequent power-ups, as it is done with the accelerometer calibration data.
(Still perform automatic gyro calibration if there is no gyro calibration data saved in EEPROM.)

The users will only get the gyro drift during a short period at the beginning of the flight (when gyro is warming up) and not for the rest of the flight (when the gyro temperature is rather stable).
Certainly, the gyro working temperature may vary due to the external conditions, but in most cases this simple and dirty solution is much more accurate than the current Betaflight implementation.

### Describe alternatives you've considered
(To completely get rid of the temperature drifts it is necessary to implement the gyro/accel/baro temperature compensation curves as PX4 firmware do.)

I have tried the stated approach on Mobula6 whoop (it had ugly gyro drift) - it works as expected: at room temperature the noticeable gyro drift lasts for about 40 seconds after cold takeoff (gyro temp 25C), steadily decreasing (up to gyro working temp 65C).
Since I am a git newbie, I have not uploaded the code to git (just not to break something), but the changes are quite straightforward.

etracer65, 2021-02-16:
I don't think this is a good idea. The IMU already has internal compensation for temperature. Further the MEMS structure "ages" significantly with abuse (like crashes) so the zero offset calibration performed at power-cycle is important. If you want a calibration close to flight, you can enable `gyro_cal_on_first_arm` so that the calibration cycle will run immediately before the motors start spinning on the first arming of a battery.

Also I think you're confusing "gyro drift" with what's actually happening for level mode users. This is also frequently incorrectly called "accelerometer drift". In reality neither are happening. What's actually happening is that the accelerometer is seeing too much physical vibration (frame, motors, props) and the background acceleration readings are too high so the accelerometer data is not trusted. The IMU attitude estimate (the roll/pitch angles) is calculated based on dead-reckoning from the gyro as the accelerometer data can't always be trusted even in normal situations because of flight maneuvers. So since the attitude is only based on gyros, the attitude estimate gradually drifts over time. However if the accelerometer data is trusted (experiencing low G-forces) then it is used as correcting data to converge the attitude estimate back towards reality as a feedback loop. The problem comes if the accelerometer data is not trusted like when there is too much vibration. Then there will be no corrective action and the attitude will only be based on gyro data and the "level" will drift over time. The solution in every case is to reduce the physical vibrations felt by the IMU.

useroff, 2021-02-16:
Ok, I see your point: the IMU should have the internal temperature correction, all evil is from the vibration.
I have carried out the following experiment:
I have put the quad on the workbench and connected it to the Betaflight configurator (the motors are off - no vibration at all).
Then I go to the Motors tab and select Gyro as the sensor source (I have also added smartport temperature telemetry for mpu6000, to get the gyro temperature readings).
I've got the following readings (axes X, Y, Z):
At 25C: 2.66,  3.17,  0.98
At 65C: 0.24, -2.68, -0.49
[image]
So in my case the difference is more than 2 deg/sec for X, more than 5 deg/sec for Y, and about 1.5 deg/sec for Z. Definitly above the noise level (~ ±0.24 deg/sec).
This result is very well reproducible.
5 deg/sec rotation is quite noticeable for me.
(Remember: no vibration - lying steady on the workbench.)

etracer65, 2021-02-16:
That gyro behavior definitely doesn't look normal and it's likely damaged.

useroff, 2021-02-16:
I have also checked mpu6000 specification to find out what temperature correction does it have. 
And it states that mpu6000 gyroscope zero-rate output (ZRO) variation over temperature (-40°C to +85°C) is _typically_ ±20 º/s.
Typical variation of 40 deg/sec is very promising :)

etracer65, 2021-02-16:
The image you provided shows a damaged gyro and not normal behavior. Here's what a normal MPU6000 looks like:

[image]

useroff, 2021-02-16:
At what CPU temperature that image was taken and how close is CPU to IMU on that board?

etracer65, 2021-02-16:
Normal flight controller, normal spacing, been on for about 30 minutes. There is no generalized problem here that needs fixing. You have a bad gyro. Or alternately a terrible flight controller design that gets the components too hot (and I doubt this is the problem).

useroff, 2021-02-16:
After being turned on for 30 minutes that gyro also has horizontal readings (but I don't want to wait for 30 minutes before takeoff).
It is a whoop and it obviously has extremely tight layout.
I wonder how many such "damaged" gyros are out there...

etracer65, 2021-02-16:
You miss the point. The graph after 30 minutes is exactly the same as immediately after the device was powered on and all components were room temperature. You have a defective gyro. Try another flight controller.

haslinghuis, 2021-02-16:
@useroff @etracer65 is right. Never saw that bad reading and deviation. My development boards are connected for hours.

useroff, 2021-02-16:
> My development boards are connected for hours.

The point is not to heat the quad for hours (when it gets temperature equilibrium).
The point is the gyro has different zero readings for different temperatures (and it is normal).
But Betaflight gets gyro zero offsets right after the startup.
And if you did not heat the quad for hours, but took it from the shelf, you will get the gyro temperature drift.

useroff, 2021-02-16:
Pay attention to the plot scale (at the default scale you won't see anything but a straght line).
Also, I have freezed the quad to 0ºC, before I have taken the plots (that is also why they look **so** "damaged").
Nevertheless, the numbers I presented above are correct:
Mobula 6 gyro drift:
At 25C: 2.66,  3.17,  0.98
At 65C: 0.49, -2.68, -0.49

I have another whoop on a different board (absolutely typical brushed Eachine QX65):
[image]
As you can see, it also has a temperature drift:
X: -1.46 -> -5.61
Y:  0.73 ->  2.44
Z:  0.98 ->  3.41

etracer65, 2021-02-16:
MPU6000 powered on and calibrated when components were room temperature (22C). Graph after the board has been running for over an hour. Notice that the traces are still properly calibrated using the zero-offset calculated from initial power when it was "cold". Nothing to fix here.

[image]

useroff, 2021-02-16:
@etracer65, I am glad you have no problems with your setup.
But Betaflight is also for whoops, right?
I have 2 different whoops and both have gyro temperature drift (more or less expressed).
Other whoop owners' reviews are welcomed.
Take the gyro plot of your whoop at Scale: 10, Refresh: 1000 ms for the first 5 minutes after the cold startup.
If you have recently powered it up (even via USB) let it cool down firstly for 15 minutes.

mikeller, 2021-02-17:
I don't think that this should be discarded as unfounded or unneeded this easily. The fact alone that PX4 includes the gyro in their elaborate thermal calibration procedure (https://docs.px4.io/master/en/advanced_config/sensor_thermal_calibration.html) should be an indicator that there is a justification for this - normally PX4 uses a sound and fact based approach.
While MPUs normally have an internal thermal compensation, I would not expect this to be based on individual per-chip calibration and a custom calibration curve being generated and stored on every chip at the factory. So individual chips may still show thermal drift even after the generic compensation has been applied.

Unfortunately it will be hard to test the hypothesis that this is of value for Betaflight by showing that a significant thermal drift exists by correlating gyro temperature and gyro output, as there currently is no way to track the gyro temperature - this would first have to be implemented more or less from scratch.

useroff, 2021-02-17:
> there currently is no way to track the gyro temperature

MPU6000 does have the temperature sensor, see "MPU-6000/MPU-6050 Register Map and Descriptions" (registers 65 and 66).

mikeller, 2021-02-17:
@useroff: I think most gyros have this functionality. But that does not change the fact that Betaflight does not support reading it at the moment.

useroff, 2021-02-17:
> Betaflight does not support reading it at the moment.

Betaflight does support gyro temperature readings (see temperatureFn member of gyroDev_t).
I have easilly added temperatureFn for MPU6000 and got gyro temperature readings on my Radiomaster TX16S.

mikeller, 2021-02-17:
@useroff: Betaflight does support anything you can imagine - if you first write the code that implements it. :wink:
Not having this functionality available in the firmware that users already use will make it hard to ask them to collect log data that can be used to gather statistically relevant data on gyro temperature drift.

useroff, 2021-02-17:
It is actually already implemented in Betaflight for MPU3050.
I can push the code for MPU6000 if anybody is willing to pull it.

mikeller, 2021-02-17:
@useroff: MPU3050 is a legacy gyro that isn't used in any current hardware, so having support there doesn't help much. And even this is just driver support, so changes will be needed to read the temperature and do something with the data. I think periodic reading and sticking it into a debug log will be a good start - this way users can generating logs showing temperature readings and gyro drift.

artur-iluk, 2021-03-02:
IMHO the problem is in understanding of "internal compensation"
In the documentation of MPU6050 there is no single word about the thermal compensation with the use of temp sensor.

Some [ test ](https://www.i2cdevlib.com/forums/topic/53-mpu-6050-temperature-compensation/) shown that MPU6050 probably do NOT use the on-board temperature sensor for temperature compensation. 

"If you use MPU6050's DMP, you dont have to care for temperature calibration. I have been checking MPU6050 vs a Microstrain 3DM-GX2, and it seems so. If not using DMP you have to compensate temperature, it is critical if you are using it in something like a quadrotor or airplane, etc. What I did is to leave the MPU6050 still reading values and temperature changing, then I obtain a linear (or cuadratic, don't remember) offset depending on temperature, and every time I read the sensor I also check temperature and compensate it."

So the sensor is probably used for compensation ONLY when you are using on-board DMP processor (MPU6050 is equipped with it). If you are using raw gyro/accel values, if should be compensated by yourself.
It is in line with observation of @useroff, and in line with action of PX4 developers.

There is inverse correlation of gyro drift with temperature shown [here](https://reprapdad.wordpress.com/2014/11/10/mpu-6050-warm-up-and-dmp-yaw-drift/) maybe it can be taken as a base for simply temp sensor correction in betaflight (after some tests...)

Good point is, that is probably will increase the accuracy mainly of yaw estimation (pitch and roll are partially naturally compensated by gravity), while yaw drift is not important for standard betaflight user. It IS important for navigation purposes and for this reason probably PX5 pay more attention to that issue.

useroff, 2021-03-02:
> There is inverse correlation of gyro drift with temperature shown [here](https://reprapdad.wordpress.com/2014/11/10/mpu-6050-warm-up-and-dmp-yaw-drift/) maybe it can be taken as a base for simply temp sensor correction in betaflight

The correlation is not inverse, but chip specific (see https://github.com/betaflight/betaflight/issues/10557#issuecomment-780148732).

> that is probably will increase the accuracy mainly of yaw estimation (pitch and roll are partially naturally compensated by gravity)

At least in level modes, the pitch and roll are clearly affected by the gyro dift.

useroff, 2021-03-03:
> If you use MPU6050's DMP, you dont have to care for temperature calibration.

The problem is there is no documented way to use DMP on MPU6000/MPU6050.
It can be enabled on MPU6500. But MPU6000 does not react to that command.

Invensense's proprietary Motion Driver (MPL) is claimed to have runtime gyro (re)calibration/temperature compensation:

>Gyro Calibration (Fast No Motion)
>Run-time calibration routine. Once a no motion state is detected the gyro 
>calibration will trigger. Calibration will complete within .5 seconds of no 
>motion state detection. 

>Gyro Temperature Compensation 
>After each gyro calibration, the MPL will record the internal temperature. 
>After several data points the MPL will be able to build a multi-point 
>temperature slope for the gyro and apply it along with the calibration biases. 
>This will compensate for gyro drift due to temperature.

But it's nowhere mentioned that these features are offloaded to DMP.

artur-iluk, 2021-03-03:
So to conclude:
1. DMP can probably solve the temperature calibration, but is not used in betaflight because it is not oficially documented (however there is [unofficial](https://github.com/jrowberg/i2cdevlib/tree/master/Arduino/MPU6050) library to use DMP in 6050 with the newest MotionApps v6.12
2. MPU6050 without DMP is rather NOT use temperatyre sensor (documented for similar MPU6000 [here](https://wiki.dronecode.org/_media/u01.imu_temperature_compensation_-_hessel_van_der_molen.pdf)). I've seen somethere (cannot find it now) the graph of MPU6050 gyro drift with DMP, and perfect gyro stabilization takes 15s (instead 15min)
3. The "pure" MPU6050 (and all other gyros/accels if equipped with internal temperature sensor) can be calibrated more or less manually with the use of internal temperature sensor. It requires:
a) new code to allow writing of MPU temperature reading to the debug channel in blackbox (maybe possible now?)
b) new code to enter calibration coefficients for specific sensor into betaflight
Procedure of calibration (when data is collected) is documented on PX4 site.
It can improve stability of gyros without waiting 15min for temperature stabilization (gyros can be calibrated at power on). And will make it temperature resistant (for cold/hot days)

The question is, if it is worth of effort. Probably for some user - yes.

artur-iluk, 2021-03-04:
A have reviewed  debug possibility, and the most simple solution is logging the temperature of MPU6050 (and maybe other equipped with temp sensor) as a debug value no 4, then in blackbox ACCELERATION or GYRO_RAW is selected.
- DEBUG1,2,3 are then XYZ components of ACC/GYRO
- DEBUG4 can be temperature of MPU (Requested Feature)
All data are available in betaflight, debug4 slot is free, it will not affect normal operation - so it should be quite easy and efficient.

With such feature, we can with time take from users easily temperature dependency of ACC/GYRO for various setups, and later decide if it is worth to incorporate the temp calibration (more time consuming)

useroff, 2021-03-04:
@artur-iluk, There is no sense to add gyro temperature to the Blackbox data for the in-flight logging, since you do not know the unbiased gyro readings to compare to.
To find out the gyro drift (and the initial data for the compensation curves) the quad should be still. So Betaflight Configurator's Tethered Logging and Sensors features will be enough.
In-flight, the gyro temperature is needed to know the in-flight temperature equilibrium (at what temperature to calibrate the gyro) and how far you are from it.
For that, the gyro temperature should be in transmitter's telemetry sensors (and, maybe, in the OSD data for the FPV).

artur-iluk, 2021-03-05:
@useroff You are right, in-flight blackbox gyro bias cannot be compared to anything, but:

1. With the in-flight blackbox logging of IMU temperature, you will know what will be a range of IMU temperature in a real flight for your setup -> you will now if you need temperature compensation. 
It is hard to test it on the table. For many users, like @etracer65 or @haslinghuis, it is probably stable, almost air temperature, but for some setups it can raise from 20C to 60-70C, as in mentioned by you Mobula6 whoop (BTW, temperature of what did you mentioned, IMU?). So in-flight IMU temp logging make sense to see if you should care about thermal compensation or gyro re-calibration after warm-up.

2. The second reason to include IMU temp as fourth value (after XYZ gyro) in the blackbox DEBUG is simplicity. Even if you want to make only a table test with Tethered Logging in Configurator, it will not require ANY changes in Configurator software - just switch on DEBUG in Tethered Logging and select GYRO in blackbox setting - it will work perfectly on current Configurator software.

3. Last reason to do it by blackbox logging - not everyone can use Telemetry feature (more expensive equipment) and not everyone is using OSD (for the same reason). The blackbox feature can be used on almost every platform.

closed 2021-04-23 by github-actions[bot]