betaflight/betaflight #10560: vbat_hysteresis docs/range or behaviour is incorrect (depending upon your perspective).

casrya, 2021-02-17:
### Describe the bug
vbat_hysteresis is actually in 0.01V increments not 0.1V as stated in the configurator therefore the default value of 1 basically does nothing (0.01V). I discovered this after testing showed little effect so I looked at the source code and identified how it actually worked.

Solution: either multiply batteryConfig()->vbathysteresis * 10 in the source code (and change the param range in the configurator) or change the docs to match actual behaviour.

P.S I actually stuffed around with this (googling etc.) before I went to the source code since I couldn’t believe that no one had reported it in all this time .. or maybe it is a new bug since I am new to betaflight? FWIW a value of 65 (0.65V) works well on 1S Mobula 7 and a bit more for 2S.

### To Reproduce
use vbat_hysteresis param and observe actual versus documented behaviour

### Expected behavior
Described above

### Flight controller configuration
N/A since relates to the value of a single param vbat_hysteresis

### Setup / Versions
Mobula7 stock CRAZYBEEF4

mikeller, 2021-02-17:
Good find @casrya!

This is indeed a bug - it was introduced in https://github.com/betaflight/betaflight/pull/7255 when the resolution of the battery voltage was changed from 0.1V to 0.01V. `batteryConfig()->vbathysteresis`, which is added into the battery voltage was not touched by this, but the significance of it changed from 0.1V to 0.01V as well.

Not sure what the best fix is here:
- leaving the code as is and amending the documentation to indicate a 0.01V resolution will yield a 2.5V maximum hysteresis - applied for high cell count batteries this might not be enough;
- changing the resolution back to the original 0.1V reduces the resolution quite drastically, and 0.1V resolution might not be enough for the nowadays popular 1S models.

etracer65, 2021-02-17:
Probably best to fix the documentation to indicate the current behavior (0.01V resolution) and then expand it to a `uint16` for 4.3.

It's not used in any MSP so nothing to worry about there.

etracer65, 2021-02-17:
Besides the docs and default value (should be 10, not 1), there are potential underflow bugs.
https://github.com/betaflight/betaflight/blob/26b74c335028ea1dbb394032f047cb094f0c3365/src/main/sensors/battery.c#L237

All the types in this are unsigned so if the hysteresis value exceeded the warning voltage then it will underflow and fail. In the current 0-250 (0-2.50V) range it's not likely to happen but if the range is expanded then it's a real problem.

I'll work on a fix.

etracer65, 2021-02-17:
See #10563

Note that `vbat_hysteresis` is not actually functioning as a `hysteresis`. In fact it's simply an offset to the user's configured warning voltages. See my additional comments on the PR.

PR #10563 (2021-02-21): Fix vbat_hysteresis docs and potential calculation underflow
Fixes #10560 

Documentation and comments were never updated when the vbat scaling was changed from 0.1v to 0.01v units. Also fixed a potential underflow if the warning or critical cell voltages were set lower than `vbat_hysteresis`.

Note that the default of 1 (0.01v) remains unchanged. The original intent of the default was to mean 0.1v so you would think the setting should now be 10 to represent 0.10v. But that would introduce undesirable changes to the battery warnings - particularly for 1S flyers. In testing it was surprising fo [... cut]
closed 2021-02-21 by mikeller