betaflight/betaflight #10561: new Unified Target need for  STM32F401CCU6   MCU

height888, 2021-02-17:
### Is your feature request related to a problem? Please describe.
Low cost simplification for airplane type, without OSD/VTX/BLACKBOX...

### Describe the solution you'd like
new Unified Target  for  STM32F401CCU6   MCU

### Describe alternatives you've considered
256K FLASH is enough

DusKing1, 2021-02-17:
256K IS NOT  ENOUGH

etracer65, 2021-02-17:
If 256K was enough then we'd still be supporting F3's.

height888, 2021-02-17:
It's possible  for airplane application without OSD/VTX/BLACKBOX/， success test is on it STM32F411

etracer65, 2021-02-17:
You miss the point. We don't support MCU's with less than 512K flash any more. It's too much work for us to maintain and that's why F3's (with 256K) were dropped.

height888, 2021-02-17:
I understand. Thank you

closed 2021-02-17 by height888