betaflight/betaflight #10564: no RQLY / RFMD of CRSF in OSD on target MATEK405CTR

lasdevegas, 2021-02-17:
### Describe the bug

no values rqly / RFMD of CRSF in OSD

### To Reproduce

configure nano crsf , verify sensors and select link quality in osd

### Expected behavior

i would have the values of RFMD/RQLY in OSD but i have always the icon with the value 0

### Flight controller configuration

```
# version
# Betaflight / STM32F405 (S405) 4.2.8 Feb 15 2021 / 12:08:36 (101738d8e) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF405CTR, version: f173f9bc, date: 2019-10-02T14:43:13Z

# start the command batch
batch start

board_name MATEKF405CTR
manufacturer_id MTKS

# resources
resource MOTOR 1 C09
resource MOTOR 2 C08
resource MOTOR 3 C07
resource MOTOR 4 C06
resource MOTOR 5 NONE
resource SERIAL_TX 11 A15

# feature
feature -SOFTSERIAL
feature GPS
feature RSSI_ADC

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 128 115200 57600 0 2000000
serial 2 2048 115200 57600 0 115200
serial 3 2 115200 115200 0 115200
serial 4 1024 115200 57600 0 115200

# aux
aux 0 0 3 1700 2100 0 0
aux 1 1 2 900 1150 0 0
aux 2 2 2 1300 1700 0 0
aux 3 13 0 1675 2075 0 0
aux 4 26 2 1725 2100 0 0
aux 5 35 2 1300 1700 0 0
aux 6 36 1 1650 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# master
set gyro_lowpass_hz = 0
set gyro_lowpass2_hz = 0
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 90
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 750
set acc_calibration = 74,13,-116,1
set min_check = 1020
set max_check = 2000
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set blackbox_device = SERIAL
set dshot_idle_value = 240
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_delay = 10
set align_board_yaw = 180
set vbat_scale = 109
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_sbas_integrity = ON
set gps_auto_config = OFF
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_min_sats = 5
set deadband = 2
set pid_process_denom = 2
set osd_warn_batt_warning = OFF
set osd_warn_core_temp = OFF
set osd_warn_launch_control = OFF
set osd_vbat_pos = 2059
set osd_rssi_pos = 84
set osd_link_quality_pos = 2066
set osd_tim_2_pos = 2050
set osd_mah_drawn_pos = 2072
set osd_gps_speed_pos = 2551
set osd_gps_sats_pos = 2536
set osd_home_dir_pos = 2540
set osd_home_dist_pos = 2545
set osd_esc_rpm_pos = 2401
set osd_esc_rpm_freq_pos = 291
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_alt = ON
set osd_stat_flight_dist = ON
set osd_stat_total_dist = ON
set debug_mode = FFT_FREQ
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 4
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5806
set gyro_1_align_yaw = 2700

profile 0

# profile 0
set dyn_lpf_dterm_curve_expo = 7
set vbat_pid_gain = ON
set pid_at_min_throttle = OFF
set iterm_relax_cutoff = 13
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set p_pitch = 79
set d_pitch = 12
set f_pitch = 0
set p_roll = 69
set d_roll = 12
set f_roll = 0
set p_yaw = 90
set f_yaw = 100
set d_min_roll = 0
set d_min_pitch = 0
set ff_interpolate_sp = AVERAGED_3

rateprofile 0

# rateprofile 0
set roll_rc_rate = 140
set pitch_rc_rate = 140
set yaw_rc_rate = 140
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 65
set pitch_srate = 65
set yaw_srate = 65
set tpa_rate = 75
set tpa_breakpoint = 1400
set tpa_mode = PD

# end the command batch
batch end
```

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: LED 2
A15: FREE
B00: FREE
B01: ADC_RSSI
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: LED 1
B10: OSD_CS
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: PREINIT 2
C02: GYRO_CS 1
C03: GYRO_EXTI
C04: ADC_CURR
C05: ADC_BATT
C06: MOTOR 4
C07: MOTOR 3
C08: MOTOR 2
C09: MOTOR 1
C10: SERIAL_TX 3
C11: FREE
C12: FREE
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

```

### Setup / Versions
 
 - Flight controller:   MATEKF405CTR ( MKFS ) 
  
 - Other components: CRSF nano RX V4.08 , rush tank VTX
 
 - How are the different components wired up:
 My CRSF nano RX is wired up to the pins TX2 and RX2

My quad is functional and fly good but i haven't the values RFMD/RQLY on my OSD ( telemetry sensors are ok on my radio ) .
For example , the battery current drawn is ok on my osd and on my radio .
For information , i have downgrade betaflight to 4.2.0 and it's always not ok 
I have downgrade betaflight to 4.1.7 ( MATEKF405CTR )  and it's ok !!( i have the values of RQLY/RFMD in my OSD with the same configuration )

etracer65, 2021-02-17:
Turn off RSSI ADC on the Configuration tab.

lasdevegas, 2021-02-18:
Thanks !! It's ok now

closed 2021-02-18 by lasdevegas