betaflight/betaflight #10568: FrskyD / FrskyX connection reset when throttle reach 1880

grostim, 2021-02-18:
### Describe the bug

Not sure if this issue is related to betaflight and/or Multiprotocol module:
I have an ongoing ticket overthere: [https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/530](url)

I am using a CrazyBeeFR4FR controller board (which have an embedded SPI Frsky receiver).
my remote use a multiprocol module.
They both bind properly but as soon as i move the throttle stick beyond 1880, the connection is lost and the controller switch to failsafe.

### Flight controller configuration

```
# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / STM32F411 (S411) 4.2.8 Feb 15 2021 / 12:09:04 (101738d8e) MSP API: 1.43
# config: manufacturer_id: HAMO, board_name: CRAZYBEEF4FR, version: 56f796fb, date: 2019-10-26T09:47:21Z

# start the command batch
batch start

board_name CRAZYBEEF4FR
manufacturer_id HAMO

# feature
feature -RX_PARALLEL_PWM
feature -ANTI_GRAVITY

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 1 2048 115200 57600 0 115200

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 1 900 1200 0 0
aux 2 2 1 1350 1650 0 0
aux 3 13 3 1700 2100 0 0
aux 4 35 2 1800 2100 0 0

# adjrange
adjrange 0 0 1 900 1200 12 1 0 0

# master
set gyro_lowpass2_hz = 0
set acc_trim_pitch = -14
set acc_trim_roll = -4
set acc_calibration = 97,-4,-201,1
set max_check = 2000
set rssi_channel = 15
set serialrx_provider = SBUS
set rx_max_usec = 2250
set bat_capacity = 250
set ibata_scale = 1175
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 1
set yaw_deadband = 1
set pid_process_denom = 1
set osd_warn_batt_not_full = OFF
set osd_warn_batt_warning = OFF
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_vbat_pos = 2433
set osd_rssi_pos = 2362
set osd_tim_2_pos = 2423
set osd_flymode_pos = 2457
set osd_throttle_pos = 2392
set osd_current_pos = 320
set osd_mah_drawn_pos = 2400
set osd_craft_name_pos = 2442
set osd_warnings_pos = 2345
set osd_avg_cell_voltage_pos = 353
set osd_disarmed_pos = 2314
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set frsky_spi_tx_id = 105,47
set frsky_spi_offset = -73
set frsky_spi_bind_hop_data = 4,65,126,187,13,74,135,196,22,83,144,205,31,92,153,214,40,101,162,223,49,110,171,232,58,119,180,6,67,128,189,15,76,137,198,24,85,146,207,33,94,155,216,42,103,164,225,0,0,0
set frsky_x_rx_num = 48
set gyro_1_align_yaw = 900

profile 0

# profile 0
set dterm_lowpass2_hz = 0
set vbat_pid_gain = ON
set p_pitch = 52
set p_roll = 51
set p_yaw = 72
set i_yaw = 55
set f_yaw = 100

rateprofile 0

# rateprofile 0
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 130
set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 50

# end the command batch
batch end
```
```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: RX_SPI_CS
B00: ADC_BATT
B01: ADC_CURR
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: LED
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: RX_SPI_EXTI
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3: MOTOR 1
TIM3: FREE
TIM4:
    CH1: MOTOR 2
    CH2: MOTOR 3
    CH3: MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 
 - Flight controller: CrazybeeFR4FR (supplied with EachineUS65 Pro)
 
 - Other components:  CaddxAnt cam
 
 - How are the different components wired up: (Stock setup from Eachine US65Pro)

grostim, 2021-02-18:
Not cool bot :(

mikeller, 2021-02-21:
@grostim: From https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/530#issuecomment-781623466 this seems to have been resolved to a configuration issue in multiprotocol?

grostim, 2021-02-21:
Absolutely. This was completely my fault.
I'm closing the issue.

Keep up the good work guys !

closed 2021-02-21 by grostim