betaflight/betaflight #10571: Flashed FC from 4.1 to 4.2.8 - FC now unresponsive and not connecting to BF

mzungumrefu, 2021-02-19:
### Describe the bug
 As title...went to update firmware on my Aikon F7 FC, followed Bardwells instructions - firmware upload seem to work - I got a green 'success' on the firmware flasher page - connected and there was a message about no ESC protocol selected, went in changed to dshot600, save and reboot, now can't connect and FC not registering. Note Betaflight never asked me if I wanted to 'apply custom defaults'. Also not getting beeps confirming ESC and FC are connected. Please advise on the solution! Any advice much appreciated!!! Adam

### To Reproduce

### Expected behavior
 I expected to have BF 4.2.8 on my FC 

### Flight controller configuration

```
# 

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.1 Nov 15 2019 / 12:57:05 (1e5e3d369) MSP API: 1.42
# manufacturer_id: AIKO   board_name: AIKONF7   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name AIKONF7
manufacturer_id AIKO
mcu_id 003c00443338511139393832
signature 

# name: SUPER G

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY
feature OSD

# serial
serial 0 64 115200 57600 0 115200
serial 2 1 115200 57600 0 115200

# aux
aux 0 0 1 1725 2100 0 0
aux 1 1 0 1725 2100 0 0
aux 2 2 0 1300 1700 0 0
aux 3 27 3 1725 2100 0 0
aux 4 13 2 1750 2100 0 0

# master
set gyro_lowpass2_hz = 425
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set dyn_lpf_gyro_min_hz = 340
set dyn_lpf_gyro_max_hz = 850
set acc_calibration = -34,24,97
set serialrx_provider = SBUS
set sbus_baud_fast = ON
set min_throttle = 1070
set dshot_idle_value = 400
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 480
set yaw_motors_reversed = ON
set pid_process_denom = 1
set osd_warn_batt_not_full = OFF
set osd_remaining_time_estimate_pos = 449
set osd_flymode_pos = 2234
set osd_mah_drawn_pos = 2488
set osd_craft_name_pos = 2543
set osd_compass_bar_pos = 98
set osd_battery_usage_pos = 2505
set gyro_rpm_notch_harmonics = 1
set name = SUPER G

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 119
set dyn_lpf_dterm_max_hz = 289
set dterm_lowpass2_hz = 255
set anti_gravity_gain = 1100
set iterm_relax = OFF
set iterm_relax_cutoff = 15
set throttle_boost = 0
set p_pitch = 50
set d_pitch = 30
set f_pitch = 60
set p_roll = 45
set i_roll = 90
set d_roll = 25
set f_roll = 60
set d_yaw = 10
set f_yaw = 5
set d_min_roll = 0
set d_min_pitch = 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set thr_mid = 30
set roll_rc_rate = 127
set pitch_rc_rate = 127
set yaw_rc_rate = 120
set roll_expo = 40
set pitch_expo = 40
set roll_srate = 72
set pitch_srate = 72
set yaw_srate = 75
set tpa_rate = 70
set tpa_breakpoint = 1500

rateprofile 1

# rateprofile 1
set roll_rc_rate = 80
set pitch_rc_rate = 80
set yaw_rc_rate = 80
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```
```
PASTE THE OUTPUT OF 'resource show all' HERE - Can't complete without connecting
```

### Setup / Versions
 
 - Flight controller: Aikon F7 FC - https://droneislife.co.uk/flightcontrollers/Aikon-F7-30x30-FC
  DJI Digital - Aikon F7 - Aikon AK32 V3 4-in-1 35A 2-6S ESC
 - Other components:
 
 - How are the different components wired up: - JST connectors and soldering 

Please help ASAP - I wanted to fly this weekend!

mzungumrefu, 2021-02-19:
Operating system macOS Big Sur.

mzungumrefu, 2021-02-19:
I flashed STM32F7X2 as per the version on the diff.....I noticed there was also an Aikon F7 specific option too - should I have flashed that instead?

klutvott123, 2021-02-19:
Yes, you should have selected the Aikon F7. Then you will get the message about applying custom defaults

haslinghuis, 2021-02-19:
Which configurator is used? Maybe try https://github.com/betaflight/betaflight-configurator-nightlies.
After flashing and (re)connect you must apply custom defaults. It configurator does not show this option you have flashed the wrong target.

mzungumrefu, 2021-02-20:
> Yes, you should have selected the Aikon F7. Then you will get the message about applying custom defaults

Despite CLI naming my firmware as ''Betaflight / STM32F7X2'' (S7X2) 4.1.1 Nov 15 2019 / 12:57:05 (1e5e3d369) MSP API: 1.42

Note Not Aikon F7 although that's mention as the ''board_name: Aikon F7''

klutvott123, 2021-02-20:
STM32F7X2 is the unified target used for that flight controller. Flashing the Aikon F7 target will still flash your flight controller with STM32F7X2 but it let's the configurator know which custom defaults to apply

mzungumrefu, 2021-02-20:
> Which configurator is used? Maybe try https://github.com/betaflight/betaflight-configurator-nightlies.
> After flashing and (re)connect you must apply custom defaults. It configurator does not show this option you have flashed the wrong target.

I'm using Configurator: 10.7.0

mzungumrefu, 2021-02-20:
> STM32F7X2 is the unified target used for that flight controller. Flashing the Aikon F7 target will still flash your flight controller with STM32F7X2 but it let's the configurator know which custom defaults to apply

Ok thanks. Any quick ideas on how to get the FC into boot loader mode to reflash now that it's not connecting? Without taking the quad apart

haslinghuis, 2021-02-20:
Pushing boot button or shortcut boot pins

mzungumrefu, 2021-02-20:
> Pushing boot button or shortcut boot pins

Reachable through frame. 👌 😅 . Thanks

mzungumrefu, 2021-02-20:
Sorted! Thanks guys!

closed 2021-02-20 by mzungumrefu