betaflight/betaflight #10572: VTX switch control does not allow for more than 10 settings

DBetic-Pilot, 2021-02-21:
### Describe the bug
When using the CLI commands to control VTX via a PWM switch, you are unable to index past 9. When attempting use any index greater than 9, the error 

### To Reproduce

### Expected behavior

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
```
```
PASTE THE OUTPUT OF 'resource show all' HERE
```

### Setup / Versions
 
 - Flight controller: 
 
 - Other components:
 
 - How are the different components wired up:

etracer65, 2021-02-21:
There are only 10 and that's how it's designed. This is not a bug. There have always only be 10 options. Your goal of having 45 selectable channels is not possible.

DBetic-Pilot, 2021-02-21:
So would this then be a feature request, or just a "go pound sand?"

etracer65, 2021-02-21:
There is no need to add this to Betaflight. Your best option is to make your own custom version.

mikeller, 2021-02-21:
Let's leave this as a feature request, and see if it gets some traction, or if somebody comes up with a good idea on how to do this - I do not think that 100 discrete ranges will be usable with the required precision on an analogue input - selecting a channel will become hit and miss.

DBetic-Pilot, 2021-02-21:
THanks mikeller, completely agree that there is no really need for 100 combinations, but figured it was a limitation in the array size being limited to a single digit. My initial thought was a tuning knob for your VTX frequency, but even if you have your 6 primary frequency options and 5 power levels, you are already short of capability. 
I'm not much of a programmer and am not sure where i would start to figure out how to make this change, but am happy to help however possible if there is interest in improvement here.

etracer65, 2021-02-21:
@DBetic-Pilot Some clarification on why I said there's no need to add this and implied that it's unlikely that Betaflight would have this change.

The first big issue is that every range we add uses memory. Adding lots (like 100!) adds a lot of memory use. In this case about 600 bytes of memory which is a huge amount in the context of Betaflight features. This would be memory that couldn't be used for other likely more useful features.

Next increasing the number of ranges will mean more processing every loop. The logic has to iterate through all the ranges and check for any that apply. Doing that every time for 10 is not so bad. Doing it for 100 will likely introduce performance problems and require reengineering of the logic.

So to justify the excessive memory use and performance impacts there would have to be compelling arguments on why it makes sense to implement this. The first might be that lots of people would use this. This seems exceptionally unlikely. The CLI `vtx` command has been around since the Cleanflight days and I can't recall a single request to increase the number of entries - even a little, let alone to 100!

So the next argument is that your suggested use case would be very useful. Again I would disagree with this. Further I think it's a very bad idea. First it's going to be extremely difficult to select a specific band/channel with a pot or slider on your radio. Trying to precisely set it to select on of 45 entries will be very hard. Next, what happens if you accidentally bump the control (or your radio jitters) while in flight? You'll lose video and be in a dangerous situation. Lastly how is this going to be useful? As you turn the knob and sweep through bands/channels you'll be blasting anybody else in the vicinity and knocking them out of the air. I'm not sure how you could even use this on the ground because as soon as you touch the knob you'll lose video in your goggles.

So the suggestion of creating your own custom version if you want to do this was not intended to be facetious, but rather as a real way that you can get what you want and try it out. There are many youtube videos out there on how to build your own version as well as documentation here: https://github.com/betaflight/betaflight/tree/master/docs/development

The effort to make this "work" in the simplest sense is trivial - just changing one line of code. So the objections are not based on this but rather on the value and cost to implementing things in Betaflight that are useful to the most people.

And if you want to build your own version, the line to change is here:

https://github.com/betaflight/betaflight/blob/7cce28131f9e1ed8cf286f52fef4836786c10c5c/src/main/io/vtx_control.h#L31

Note that there is no point is increasing to beyond 40. The minimum channel range step is 25 so that yields a max of 40 discrete channel ranges.

DBetic-Pilot, 2021-02-21:
Bruce,
Thanks for the very thorough and professional response. I clearly mis-associated the tone on your initial posts and apologize for my rather snarky response. Based on what you have said, i completely agree that this probably has no place in the primary branch of code, and is my use case is absolutely a 1-off where i am utilizing this functionality in order to easily test VT receiver scan functions, which as you pointed out means it is not beneficial to the greater betaflight community. 

Unless there are others that feel more than 10 are needed, i think its perfectly fine to close the request at this point, as i can either compile after making the change above, or find another work-around for my use case.

thanks again for the clarification.

closed 2021-02-21 by DBetic-Pilot

SPooKfpv, 2021-07-01:
A minimum of 20 to change channel, band and power.

Using 3 channels - Have logic setup so channel/ band only works in pitmode. Its safe, it should work. 

# vtx powers
vtx 0 8 0 0 1 900 1150
vtx 1 8 0 0 2 1200 1750
vtx 2 0 0 0 3 900 900
vtx 3 8 0 0 4 1800 2100
# vtx band
vtx 4 11 0 0 0 950 1050
vtx 5 11 1 0 0 1100 1250
vtx 6 11 2 0 0 1300 1500
vtx 7 11 3 0 0 1550 1700
vtx 8 11 4 0 0 1750 1900
vtx 9 11 5 0 0 1950 2100
# vtx channel
vtx 10 9 0 1 0 950 1050
vtx 11 9 0 2 0 1100 1150
vtx 12 9 0 3 0 1200 1350
vtx 13 9 0 4 0 1400 1500
vtx 14 9 0 5 0 1550 1650
vtx 15 9 0 6 0 1700 1800
vtx 16 9 0 7 0 1850 1930
vtx 17 9 0 8 0 1980 2100

Now im gonna have to figure out how to get that line of code into my betaflight build.....

gumileeev, 2023-03-23:
I would like to support those who ask to increase the ability to adjust the VTX to the number 20 (19).
The radio equipment Jumper t pro, Radiomaster TX16s has convenient 6-position switches, 7 channels can be assigned to them. On the rotary control, I assign the power in, at least 4 levels. There are already 10 in total, and we did not meet the allotted 9 settings. Probably someone else would add a switch band. I will be glad if changes are made. If not, we will use what we have. This is a very good implementation, in inav this is not enough.

PR #14630 (2025-09-03): allow custom define for MAX_CHANNEL_ACTIVATION_CONDITION_COUNT
- please test/verify
- resolves #10572
- resolves #14541 

## Summary by CodeRabbit

* **New Features**
  * Added support for multiple VTX channel activation conditions, enabling configuration by band, channel, power, and range.
  * Increased flexibility for VTX behavior with an array of configurable conditions (default capacity of 10; supports higher counts within system limits).
  * Public configuration options updated to reflect the new activation condition fields and list-based setup.