betaflight/betaflight #10574: T-Motor F7 FC - I2C BMP280 Problems

addo60, 2021-02-21:
### Describe the bug

### To Reproduce

### Expected behavior

### Flight controller configuration

```
diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.8 Feb 15 2021 / 12:10:35 (101738d8e) MSP API: 1.43
# config: manufacturer_id: TMTR, board_name: TMOTORF7, version: e02dd6f2, date: 2020-11-04T11:31:26Z

# start the command batch
batch start

board_name TMOTORF7
manufacturer_id TMTR

# name: addo60_bee

# feature
feature -RX_PARALLEL_PWM
feature INFLIGHT_ACC_CAL
feature RX_SERIAL
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE

# map
map TAER1234

# serial
serial 2 1 115200 57600 0 115200
serial 4 64 115200 57600 0 115200

# aux
aux 0 0 1 1900 2100 0 0
aux 1 1 3 1900 2100 0 0
aux 2 2 3 1400 1600 0 0
aux 3 13 0 1900 2100 0 0
aux 4 35 2 1900 2100 1 0
aux 5 36 4 1900 2100 0 0

# master
set gyro_lowpass2_hz = 450
set dyn_notch_width_percent = 0
set dyn_notch_q = 270
set dyn_notch_min_hz = 90
set dyn_notch_max_hz = 350
set dyn_lpf_gyro_min_hz = 360
set dyn_lpf_gyro_max_hz = 900
set acc_calibration = 47,2,70,1
set baro_bustype = I2C
set baro_spi_device = 0
set baro_i2c_device = 3
set baro_i2c_address = 76
set baro_hardware = BMP280
set baro_noise_lpf = 350
set min_check = 1010
set max_check = 1990
set rc_smoothing_auto_smoothness = 7
set fpv_mix_degrees = 25
set serialrx_provider = CRSF
set dshot_idle_value = 90
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set align_board_roll = 180
set bat_capacity = 1400
set vbat_warning_cell_voltage = 340
set vbat_scale = 144
set vbat_divider = 13
set ibata_scale = 208
set small_angle = 180
set deadband = 3
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_cap_alarm = 1200
set osd_vbat_pos = 2467
set osd_rssi_pos = 2403
set osd_rssi_dbm_pos = 389
set osd_flymode_pos = 2133
set osd_g_force_pos = 2117
set osd_current_pos = 2124
set osd_mah_drawn_pos = 2435
set osd_altitude_pos = 134
set osd_pid_roll_pos = 2446
set osd_pid_pitch_pos = 2478
set osd_pid_yaw_pos = 2414
set osd_warnings_pos = 14602
set name = addo60_bee

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 133
set dyn_lpf_dterm_max_hz = 323
set dterm_lowpass2_hz = 285
set anti_gravity_gain = 4000
set i_pitch = 81
set d_pitch = 34
set f_pitch = 86
set p_roll = 41
set i_roll = 77
set d_roll = 32
set f_roll = 81
set i_yaw = 81
set f_yaw = 81
set d_min_roll = 21
set d_min_pitch = 23

rateprofile 0

# rateprofile 0
set roll_rc_rate = 120
set pitch_rc_rate = 120
set roll_expo = 35
set pitch_expo = 35
set yaw_expo = 53
set roll_srate = 65
set pitch_srate = 65

# end the command batch
batch end

# 
```
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: LED 1
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: FREE
B09: CAMERA_CONTROL
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: ADC_CURR
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: MOTOR 1
C07: MOTOR 2
C08: FLASH_CS
C09: FREE
C10: SERIAL_TX 3
C11: SERIAL_RX 3
C12: SERIAL_TX 5
C13: BEEPER
C14: PINIO 1
C15: FREE
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 3
    CH2: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11:
    CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: DSHOT_BITBANG 2
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Setup / Versions
 
 - Flight controller: T-Motor F722 FC
 
 - Other components: Caddx Vista
 
 - How are the different components wired up:

etracer65, 2021-02-21:
Your settings are incorrect. Should be:
```
set baro_i2c_device = 2
set baro_i2c_address = 0
set baro_hardware = AUTO
```

Also not sure why you changed `baro_noise_lpf = 350` and would recommend setting back to the default of 600.

addo60, 2021-02-21:
Okay, thanks - i'll try this out

closed 2021-02-21 by addo60