betaflight/betaflight #10575: Request support for VCM5883L compass

TTRNFPV, 2021-02-22:
[VCM5883L_driver.zip](https://github.com/betaflight/betaflight/files/6019199/VCM5883L_driver.zip)

### Is your feature request related to a problem? Please describe.

### Describe the solution you'd like

### Describe alternatives you've considered

etracer65, 2021-02-22:
Why? Can you provide testing hardware like breakout boards or Betaflight compatible flight controllers that contain this part?

Google doesn't show much about this other than the manufacturer page in Chinese.

TTRNFPV, 2021-02-22:
I can provide testing hardware. Because there are too many compasses that are out of production. It would be great if you can support

mikeller, 2021-02-22:
@TTRNFPV: The numbers suggest that it's yet another clone of the HMC5883L - have you tried if it works with the HMC5883L driver?

TTRNFPV, 2021-03-03:
HI. I tested it and it didn't work at all.

jflyper, 2021-03-09:
A driver code in
https://github.com/DFRobot/DFRobot_QMC5883/blob/master/DFRobot_QMC5883.h
suggests that VMC part is closer to QMC part, but there are numerous small differences starting with device ID lol

mikeller, 2021-03-24:
@jflyper: Do you have time to fiddle around with this part?

TTRNFPV, 2021-03-24:
HI. The VCM5883L compass is being tested recently. May I ask what you suggest. Or is there any good way

jflyper, 2021-03-25:
Why is VCM5883L so unpopular on the net? I can only get a handful of hits from google.

TTRNFPV, 2021-04-06:
I was testing it recently. They all can't be used normally.Do you have suggestions for compass？

closed 2021-05-13 by github-actions[bot]