betaflight/betaflight #10580: Corrupted Config : osd_units = 25###

koklimabc, 2021-02-22:
I'm newbie on betaflight and really need to ask help from you all expertised guide and help here.

I've upgraded JHEF7DUAL to "Betaflight version 1.1.1 (legacy)" in moment, 

My problems here:

1. Failed : Wired Video In to VIN of FC from Camera, and VIOUT of FC to Video port of VTX
    Success : Wired Camera Signal directly to Video Signal of VTX 

2. In CLI, When I'd typed out "diff all" to check out and ended out with an errors as "CORRUPTED CONFIG: osd_units" under set osd_units,

 # master
set gyro_lowpass2_hz = 325
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set dyn_lpf_gyro_min_hz = 260
set dyn_lpf_gyro_max_hz = 650
set motor_pwm_protocol = DSHOT600
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_units = 
**###ERROR: CORRUPTED CONFIG: osd_units = 25###**
set osd_rssi_alarm = 96
set osd_cap_alarm = 742
set osd_alt_alarm = 0
set osd_vbat_pos = 14360
set osd_rssi_pos = 14337
set osd_flymode_pos = 14842
set osd_gps_sats_pos = 14816
set osd_home_dir_pos = 14827
set osd_home_dist_pos = 14828
set osd_flight_dist_pos = 14821
set osd_altitude_pos = 14834
set debug_mode = SMARTAUDIO
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 1
set vtx_freq = 5658

Kindly appreciate somebody could help me to fix up on this problem, Thanks a lot.

closed 2021-02-22 by github-actions[bot]

koklimabc, 2021-02-22:
problem fix : set to Imperial

mikeller, 2021-02-23:
@koklimabc: The configuration is potentially still corrupted (something has caused this), so the recommendation is to reset the configuration to defaults, and then re-apply your custom settings.