betaflight/betaflight #10583: OSD info wrong when used VTX AUX function - VTX AUX does not save channel

uPz3TT3r, 2021-02-25:
### Describe the bug

If the video channel is changed with vtx aux function, it is not saved and will change after reboot, this leads to inconsistent state.
If you change video channel via betaflight osd, or you change the default start channel on vtx tab in betaflight GUI, both saves the new channel as new default and go back to that channel after reboot. vtx aux channel function even changes that variable in betaflight gui, but after reboot it is resetted.
In my opinion if the user changes the video channel, that channel should be saved as new default channel.

If you set vtx default channel on betaflight vtx tab to "none" and save, after it you switch channel via vtx aux function to raceband 3, the OSD in googles show "R:3:25" what is correct, if you reboot the OSD shows "R:  :25" but vtx still sending on raceband3.

So in my opinion after switching with vtx aux function, that value should be saved to get a save OSD status after reboot and in the upper case there should be "R:3:25" in the OSD.

### To Reproduce

Define AUX 6 for vtx aux function, in my case i would like to switch video channels via "special function - overwrite" of flysky nirvana.
vtx 0 5 0 0 0 1000 1525 
vtx 1 5 0 1 1 1550 1575
vtx 2 5 0 2 1 1600 1625
vtx 3 5 0 3 1 1650 1675
vtx 4 5 0 4 1 1700 1725
vtx 5 5 0 5 1 1750 1775
vtx 6 5 0 6 1 1800 1825
vtx 7 5 0 7 1 1850 1875
vtx 8 5 0 8 1 1900 1925

If you set vtx default channel on betaflight vtx tab to "none" and save, after it you switch channel via vtx aux function to raceband 3, the OSD in googles show "R:3:25" what is correct, if you reboot the OSD shows "R:  :25" but vtx still sending on raceband3.

### Expected behavior

So in my opinion after switching with vtx aux function, that value should be saved to get a save OSD status after reboot and in the upper case there should be "R:3:25" in the OSD.

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
no diff dump
PASTE THE OUTPUT OF 'resource show all' HERE
no resource show all output

### Setup / Versions
 <!-- Specify your flight controller model (what type is it, where was it bought from, ...) -->
 - Flight controller: Mamba F405, Fortini F4
 <!-- Specify other components attached to the flight controller (RX, VTX, brand / model for all of them, firmware version where applicable...) -->
 - Other components: TBS unify Race HV
 <!-- Details about how all is wired -->
 - How are the different components wired up:

<!-- Add any other context about the problem that you think might be relevant here. -->

etracer65, 2021-02-25:
> So in my opinion after switching with vtx aux function, that value should be saved to get a save OSD status after reboot and in the upper case there should be "R:3:25" in the OSD.

Yes, that's your opinion. But that's not how the feature is designed to work and it is working as it should. Any entry in the `vtx` CLI command is only active while its channel range conditions are set. If none of the `vtx` entries are matched then the band/channel/power return to the default settings. The `vtx` settings controlled by aux channel are an "override" to the default settings while the condition is met. Your suggestion would be unworkable as there'd never be a reasonable way to know what the vtx will be set to. Was an aux channel active before? Was it not? What if you had many entries set up in the CLI `vtx` command? I realize you think you want this behavior but you really don't and if you think about it you'll see why.

And in your example if you set the aux channels on your radio back to the same configuration as when you switched to "R:3:25" then it will be that state when you power up the next time. If your radio switches are not the same then it would be confusing for it to actually be on this channel since it doesn't match the configuration for your radio channels and CLI `vtx` settings. This all works the same as the settings on the "Modes" tab. For example, would it make sense for your quad to "stay" in angle mode just because you enabled it on the last battery even though the radio switches don't enable it now?

uPz3TT3r, 2021-02-25:
Ok, i accept a little bit the idea, but i think the default vtx value in betaflight should not be modified by the vtx aux function to show thats it has not the same rights as osd and gui.

And i think its still a bug that you can set it to "none" on the vtx tab. Change via aux to another channel for example raceband 3 and after the restart the betaflight osd on the googles show "R:  :25" and everthing is still sending on raceband 3. In this case the fc knows that the vtx is still on raceband 3 (via vtx data) and so the osd info is wrong.

etracer65, 2021-02-25:
Setting the band/channel/power to "none" (0 in the CLI) means "don't try and set a value" to Betaflight. Or leave the value alone and don't try and change it. It doesn't mean "set it to nothing".

Further the OSD display shows what Betaflight is trying to set the VTX to. Not what the VTX is already set to. So it's showing that it's not trying to set the channel and the VTX remains however it was previously set.

So everything seems to be working as designed.

uPz3TT3r, 2021-02-26:
With smartaudio the fc is able to know on with channel the vtx really is on, right?
And my understanding of OSD information is that they should show the actual real world status. Battery, rssi, time, warnings everthing is shown as it is really is, but vtx channel is shown as what the fc likes it to be....? No comment for that, in my opinion its just stupid in that way

etracer65, 2021-02-26:
> And my understanding of OSD information is that they should show the actual real world status. Battery, rssi, time, warnings everthing is shown as it is really is, but vtx channel is shown as what the fc likes it to be....? No comment for that, in my opinion its just stupid in that way

If you think you can do it better then feel free to submit a pull request.

uPz3TT3r, 2021-02-26:
Ok perfect. I am so sure that you never have tried to reproduce that and get the feeling thats it could  be really wrong as it is. YOU know how it should be, so lets do it in YOUR way, because YOU use that feature every day ... I can feel that. Thank you so much.

uPz3TT3r, 2021-02-26:
You wrote: "The vtx settings controlled by aux channel are an "override" to the default settings while the condition is met." Is my understanding correct that this is just a temporary overwrite? I dont understand why changing vtx channel via betaflight osd is a permanent overwrite of the vtx default channel and changing vtx channel with aux is just temporary... that is my problem, it makes no sense for me why.

If you connect your fc to usb and go to betaflight and change vtx via aux, the default channel value is changed. Same behavior as when doing this via betaflight osd. But one is not saved and the other one is saved...and because of that the vtx will normally stay on that channel i think its a bit different behavior than the "angle" mode or something, because there is no other device involved with the angle mode, but with the vtx it is.

You wrote: "And in your example if you set the aux channels on your radio back to the same configuration as when you switched to "R:3:25" then it will be that state when you power up the next time."  Yes thats true, that is working
You wrote: "f your radio switches are not the same then it would be confusing for it to actually be on this channel since it doesn't match the configuration for your radio channels and CLI vtx settings." AND thats the point, therefore we have the option to set a value area and give it band=0 channel=0 to not change it and stay on the last "default" channel, which is normally the last one set by OSD or GUI or VTX AUX ... so when VTX AUX video change would do a save of the default channel, as both other functions do, everthing would be ok and working nice

etracer65, 2021-02-26:
There is no bug and it's working as designed. If you don't like the way it's working and think you can make it better then develop the need code changes, test, and submit a pull request for consideration.

closed 2021-02-26 by etracer65