betaflight/betaflight #10591: R-XSR ACCST D16 v2.1.1 firmware doesn't work correctly with beta flight 4.2.6, telemetry is lost after few sec

porlock, 2021-02-27:
### Describe the bug
R-XSR ACCST D16 v2.1.1 firmware doesn't work correctly with beta flight 4.2.6, telemetry is lost after few sec
reverting back to R-XSR ACCST D16 v2.1.0 fix the issue.

### To Reproduce
Setup telemetry on R-XSR ACCST D16 v2.1.1 f.port and beta flight 4.2.6

### Expected behavior
Telemetry is rock solid and don't break up.

### Flight controller configuration

```
# 

# diff all

# version
# Betaflight / STM32F405 (S405) 4.2.6 Jan  5 2021 / 19:07:23 (a4b6db1e7) MSP API: 1.43
# config: manufacturer_id: FPVM, board_name: BETAFLIGHTF4, version: f92ddc38, date: 2020-04-28T07:37:21Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name BETAFLIGHTF4
manufacturer_id FPVM
mcu_id 005400485453501720383148
signature 

# name: CLOUD-149

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY
feature LED_STRIP

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# serial
serial 0 2048 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# led
led 0 3,8::AO:2
led 1 4,8::CO:2
led 2 5,8::CO:2
led 3 6,8::L:2
led 4 7,8::L:2
led 5 8,8::CO:2
led 6 9,8::CO:2
led 7 10,8::AO:2

# mode_color
mode_color 6 1 2
mode_color 7 0 5

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1300 1700 0 0
aux 2 2 1 900 1300 1 0
aux 3 13 2 1700 2100 0 0
aux 4 36 2 1300 1700 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 100 200 400

# master
set acc_calibration = -40,69,136,1
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set dshot_idle_value = 300
set motor_pwm_protocol = DSHOT600
set failsafe_procedure = AUTO-LAND
set bat_capacity = 1300
set current_meter = ADC
set battery_meter = ADC
set ibata_scale = 80
set osd_warn_batt_not_full = OFF
set osd_warn_rssi = ON
set osd_warn_link_quality = ON
set osd_vbat_pos = 79
set osd_rssi_pos = 2202
set osd_link_quality_pos = 2113
set osd_tim_1_pos = 33
set osd_flymode_pos = 2553
set osd_anti_gravity_pos = 2081
set osd_vtx_channel_pos = 2101
set osd_current_pos = 103
set osd_craft_name_pos = 2538
set osd_pidrate_profile_pos = 2528
set osd_warnings_pos = 14633
set osd_avg_cell_voltage_pos = 2168
set osd_battery_usage_pos = 2088
set osd_disarmed_pos = 2283
set osd_stat_rtc_date_time = ON
set osd_stat_tim_2 = OFF
set osd_stat_endbatt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_min_link_quality = ON
set osd_stat_flight_dist = ON
set osd_stat_total_time = ON
set osd_stat_min_rssi_dbm = ON
set vtx_band = 1
set vtx_channel = 8
set vtx_power = 1
set vtx_freq = 5725
set name = CLOUD-149

profile 0

profile 1

# profile 1
set p_pitch = 41
set d_pitch = 34
set p_roll = 38
set d_roll = 32
set p_yaw = 27
set d_min_roll = 18
set d_min_pitch = 20

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

# rateprofile 1
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 5
set roll_srate = 77
set pitch_srate = 77
set yaw_srate = 72

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: SERIAL_TX 2
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: BEEPER
B05: LED 1
B06: LED_STRIP
B07: FREE
B08: PULLDOWN 1
B09: FREE
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: MOTOR 4
C09: MOTOR 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: INVERTER 2
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM4:
    CH1: LED_STRIP
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3: MOTOR 4
    CH4: MOTOR 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: MOTOR 2
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 1
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: MOTOR 4
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: MOTOR 3

# 
```

### Setup / Versions
XF20-F405 F4 2-6S Flight Controller from banggood
R-XSR - ACCST D16 v2.1.1 F.Port

It's looks like exactly same issue that was for FrSky XSR 4 years back in 2017, that was described by some guy on this video
https://youtu.be/od-Qrd4Lulo

klutvott123, 2021-02-27:
It's been reported here https://github.com/betaflight/betaflight/issues/10496. Looks like a RX firmware bug. I would recommend reporting it to Frsky https://github.com/FrSkyRC/Firmware-Test/issues

daleckystepan, 2021-03-02:
Already reported to FrSky through my local vendor.

klutvott123, 2021-04-03:
Reported to Frsky https://github.com/FrSkyRC/Firmware-Test/issues/118

closed 2021-05-11 by github-actions[bot]