betaflight/betaflight #10592: betaflight bug

tibsel, 2021-02-27:
### Describe the bug
today I was flying my drone, it worked great, I charged my batteries again and wanted to fly again but my osd was not working,
I thought I burnt out my osd chip, but that's not the case I saw this tread https://github.com/betaflight/betaflight/issues/8964
and if I try it my osd comes back, go out again plugged it in and osd is gone again, apparently, if i power cycle it forgets that specific command

### To Reproduce
i dont know how to reproduce the behavior, the flight controller in question is a mamba f405 with 30a esc, runcam swift2 and a eachine tx1200 1w vtx. i updated it to the latest version of betaflight a couple of days ago that bf 4.2.8 at the time.

### Expected behavior
i expect the flight controller to just put the osd over my video signal like it is supposed to

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
``` version
# Betaflight / STM32F405 (S405) 4.2.8 Feb 15 2021 / 12:08:36 (101738d8e) MSP API: 1.43
# config: manufacturer_id: DIAT, board_name: FURYF4OSD, version: 45b6e90d, date: 2020-11-27T19:02:33Z

# start the command batch
batch start

board_name FURYF4OSD
manufacturer_id DIAT

# resources
resource OSD_CS 1 A15

# feature
feature DISPLAY

# map
map TREA1234

# serial
serial 0 64 115200 57600 0 115200

# aux
aux 0 0 0 1325 1725 0 0
aux 1 2 1 1300 1700 0 0
aux 2 35 1 1700 2100 0 0
aux 3 36 2 1700 2100 0 0

# master
set gyro_lowpass2_hz = 325
set dyn_lpf_gyro_min_hz = 260
set dyn_lpf_gyro_max_hz = 650
set acc_calibration = 15,8,27,1
set fpv_mix_degrees = 20
set dshot_idle_value = 500
set dshot_bidir = ON
set yaw_motors_reversed = ON
set osd_link_quality_pos = 14433
set osd_flymode_pos = 14381
set osd_ah_sbar_pos = 14542
set osd_ah_pos = 14414
set osd_current_pos = 14401
set osd_mah_drawn_pos = 14360
set osd_avg_cell_voltage_pos = 14497
set gyro_1_align_yaw = 1800

profile 1

rateprofile 0

# end the command batch
batch end

```
PASTE THE OUTPUT OF 'resource show all' HERE
```
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 1
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: BEEPER
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: OSD_CS
B00: MOTOR 2
B01: MOTOR 3
B02: FREE
B03: FLASH_CS
B04: LED 2
B05: LED 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: CAMERA_CONTROL
B10: SERIAL_TX 3
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: INVERTER 1
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 1
    CH2: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11:
    CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 1
DMA2 Stream 3: DSHOT_BITBANG 2
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
### Setup / Versions
 <!-- Specify your flight controller model (what type is it, where was it bought from, ...) -->
 - Flight controller: mamba f405 mk2 30a 4in1 esc from banggood.
 <!-- Specify other components attached to the flight controller (RX, VTX, brand/model for all of them, firmware version where applicable...) -->
 - Other components:
 - rx stock frsky receiver 
 - vtx tx1200 eachine
 - camera runcam swift 2
 <!-- Details about how all is wired -->
 - How are the different components wired up:
just like they were before they broke
so camera plus to 9v camera - to ground signal to cam in 
vtx to the pins where it is designed for together with smart audio

<!-- Add any other context about the problem that you think might be relevant here. -->

etracer65, 2021-02-27:
The issue you linked to is not related to your problem and is specific to another flight controller. You absolutely should not do `resource OSD_CS 1 A15` for your flight controller. This guaranteed not to work. The proper setting is the default for this board:
```
resource OSD_CS 1 B12
```

Since you've messed up the settings it's recommended that you reset all settings with `defaults` in the CLI and reconfigure from scratch (don't past this `diff` back in).

Also the "DISPLAY" feature is not the OSD - turn that off. It's an external OLED display panel that you would connect via I2C and rarely used.

Lastly, this flight controller doesn't power the OSD chip when only connected to USB. So for the OSD to be detected and work, the battery must be connected first and then the USB. If you connect the USB first and then the battery then the OSD won't work.

tibsel, 2021-02-27:
Thanks alot for the fast response and the good explenation i dont get it why peaple know this stuff but its going way above my head. It works now so im able to fly my packs empty tomorrow. About the lcd i didnt know it was turned on. I turned it off. Thanks for the great suport and that you take the time to suport beginners like me.

tibsel, 2021-02-28:
it was fixed for 2 flights and it's back now retried it did not work tried reflashing firmware reset to defaults it did not work...

etracer65, 2021-02-28:
Then  it sounds like you have a hardware issue with a failing OSD chip. Suggest you contact the manufacturer or replace the flight controller.

https://github.com/betaflight/betaflight#hardware-issues

closed 2021-02-28 by etracer65

taurusturk, 2023-03-23:
Hello,
I just bought eachine tx1200 vtx. I soldered everything properly. but the image goes away at 1mt. pit closed. Can you do a reset again.vtx to the motherboard. Is there a hex file?