betaflight/betaflight #10593: 4.2.8 FC freeze after few seconds

MattxFR, 2021-03-02:
### Describe the bug
Problem with 4.2.8 firmware on FC lumenier lux mini F7 freeze in its state after a short while: activity led stop flashing (on or off randomly), if a motor is running, it will continue running until main power cutoff, if change tab in betaflight "waiting for data" will appear forever. voltage sensor freeze on one value.

i reverted to BF 4.1.3 which seems to fix the issue. I will probably try more BF versions to run a more up to date software, but its scary, if it happens in flight the motors will keep running until battery dies !

### To Reproduce
plus with usb, and switch tabs until you cannot. alternatively use the motor tab to send manually order to motors until you see it freeze.

### Expected behavior
To no freeze

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
```
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.3 Jan 16 2020 / 11:35:53 (543a5e1ba) MSP API: 1.42
# manufacturer_id: LMNR   board_name: LUXMINIF7   custom defaults: YES

# start the command batch
batch start

board_name LUXMINIF7
manufacturer_id LMNR

# serial
serial 0 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200

# master
set acc_calibration = 52,20,-228
set motor_pwm_protocol = DSHOT300

profile 0

rateprofile 0

# end the command batch
batch end
```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: CAMERA_CONTROL
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: BEEPER
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 1
B07: MOTOR 3
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: PINIO 1
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_BATT
C01: ADC_CURR
C02: FREE
C03: FREE
C04: GYRO_CS 1
C05: FREE
C06: FREE
C07: FREE
C08: MOTOR 2
C09: MOTOR 4
C10: FREE
C11: SERIAL_RX 3
C12: FREE
C13: FREE
C14: FREE
C15: LED 1
D00: FREE
D01: FREE
D02: OSD_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: CAMERA_CONTROL
TIM2: FREE
TIM3: FREE
TIM4:
    CH1: MOTOR 1
    CH2: MOTOR 3
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3: MOTOR 2
    CH4: MOTOR 4
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: TIMUP 8
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 
 - Flight controller: LUMENIER LUX MINI F7
 
 - Other components: same with or without aything plugged
 
 - How are the different components wired up: same with or without aything plugged

It is the second FC, i though the first one was defective because i had difficulties even to flash the firmware. if was even worse on the other FC and i tried 4.2.6 as well as 4.2.8 with same issue. Note the above CLI info are after bug fix i do not want to risk being blocked with the defect on 4.2.8 as it may freeze before the firmware flash

etracer65, 2021-03-09:
To be clear, you can only replicate this in the Configurator right? It's a Configurator error - not that the flight controller is "freezing". The Configurator is the one becoming unresponsive. So there's not a risk of this happening during flight. If the flight controller actually "froze" then the motors couldn't possibly keep spinning.

If you were to close and reopen the Configurator you would be able to connect again.

MattxFR, 2021-03-09:
Hello Bruce,

Update on the topic:
I have tried a second board from the same supplier with same issue. Then I have purchased another brand (furious fpv racepit mini F7) which works beautifully under bf4.2 out of the box.
My conclusion is lumenier have a quality problem and betaflight is fine, unless there is some sort of incompatibility with 4.0 but it came already with it on so doesn't make sense.
The problem of freeze in a certain state occurs also without betaflight connected so it's not only the configurator, it's the board freezing in a certain state and this is deadly dangerous, could freeze full throttle...
Cheers, matt

Matthias Favrel

closed 2021-03-10 by MattxFR