betaflight/betaflight #10599: VTX remembers last power state rendering "low power on disarm: on until first arm" useless

badgyro, 2021-03-04:
### Describe the bug
When I power up my quad, VTX is using the same power level as was using when disconnecting the battery.

For example:
When I disconnect the battery while my VTX is transmitting at 800 mW and then connect it again my VTX will be still at 800 mW despite `low power on disarm` set to `ON` or `ON until first arm`.

But that doesn't mean that `low power on disarm` is not working. When I arm the quad VTX power is switched to correct level. And when I disarm VTX power is switched to 25 mW as expected.

All of that mean that `low power on disarm` set to `ON` almost work as it should. But set to `ON until first arm` never works.

When I disarm while having `low power on disarm` set to `ON` power gets reduced to 25 mW, then I unplug the battery. VTX remembers this 25 mW somehow and when I plug it again I get expected 25 mW. However when I disarm while `low power on disarm` is set to `ON until first arm` power is not reduced and I unplug the battery while VTX is at e.g 200 mW. Then plugging it back in I get 200 mW instead of expected 25 mW.

### To Reproduce
1. Set the `low power on disarm` to `ON` or `ON until first arm`.
2. Set the VTX power to something higher than the lowest setting. (My VTX has 25, 200, 400, 800 mW and for example I will set it to 200 mW).
3. Disconnect the battery making sure that you are disconnecting while VTX is at level 2 or more.
4. Connect the battery and check what mW is OSD and VTX showing. In my case VTX LED is showing 200 mW and OSD is showing 200 mW.

### Expected behavior
When I connect the battery I expect VTX power to be lowest possible, 25 mW in my case. FC should enforce that power is switched to the lowest.

### Flight controller configuration
```
# version
# Betaflight / STM32F7X2 (S7X2) 4.2.8 Feb 15 2021 / 12:10:35 (101738d8e) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF722MINI, version: f173f9bc, date: 2019-10-02T14:43:13Z

# start the command batch
batch start

board_name MATEKF722MINI
manufacturer_id MTKS

# feature
feature ESC_SENSOR

# beacon
beacon RX_SET

# serial
serial 2 8192 115200 57600 0 115200
serial 3 16384 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# aux
aux 0 0 0 1450 1550 0 0
aux 1 1 1 950 1050 0 0
aux 2 13 3 1950 2050 0 0
aux 3 35 2 1450 1550 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 25 200 400 600
vtxtable powerlabels 25 200 400 800

# master
set gyro_lowpass2_hz = 300
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_max_hz = 350
set dyn_lpf_gyro_min_hz = 240
set dyn_lpf_gyro_max_hz = 600
set acc_calibration = 205,29,345,1
set min_check = 1005
set max_check = 1995
set fpv_mix_degrees = 15
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_poles = 12
set bat_capacity = 850
set ibata_scale = 100
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 1
set osd_warn_rssi = ON
set osd_warn_link_quality = ON
set osd_warn_rssi_dbm = ON
set osd_cap_alarm = 750
set osd_tim1 = 1
set osd_tim2 = 0
set osd_vbat_pos = 6199
set osd_rssi_pos = 6177
set osd_link_quality_pos = 6209
set osd_rssi_dbm_pos = 33
set osd_flymode_pos = 6529
set osd_g_force_pos = 6305
set osd_throttle_pos = 6425
set osd_vtx_channel_pos = 6593
set osd_current_pos = 6230
set osd_mah_drawn_pos = 6328
set osd_motor_diag_pos = 6393
set osd_power_pos = 6296
set osd_warnings_pos = 6497
set osd_battery_usage_pos = 497
set osd_disarmed_pos = 6465
set osd_esc_tmp_pos = 6488
set osd_rtc_date_time_pos = 6630
set osd_stat_rtc_date_time = ON
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_min_link_quality = ON
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 2
set vtx_low_power_disarm = ON
set vtx_freq = 5806
set vcd_video_system = PAL
set gyro_rpm_notch_harmonics = 2

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 84
set dyn_lpf_dterm_max_hz = 204
set dterm_lowpass2_hz = 180
set p_pitch = 69
set d_pitch = 44
set f_pitch = 143
set p_roll = 63
set d_roll = 40
set f_roll = 135
set p_yaw = 68
set f_yaw = 135
set d_min_roll = 0
set d_min_pitch = 0

rateprofile 0

# end the command batch
batch end
```
```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: SERIAL_RX 6
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: PREINIT 2
D00: FREE
D01: FREE
D02: FLASH_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 - FC: Matek F722-miniSE
 - VTX: Matek VTX-mini, firmware 1.1.0 (latest 04.03.2021)

### How are the different components wired up:
[image]

I do not have RF power meter to make sure what VTX is actually outputting. And I do not have any other FC or VTX to try.
This is my only quad.

etracer65, 2021-03-05:
Tested and can't replicate the problem. Used a different Tramp protocol VTX (Speedybee TX500) and it behaves as expected. If the VTX is powered off while in high-power mode during arming it correctly switched back to the lowest power (25mw) on the next power cycle. There is a couple second delay after initial power before it switches and that's normal as the flight controller firmware is booting and detecting/initializing devices and switches once it's fully operational.

I don't have a Matek VTX-mini to test so the problem may be specific to that VTX. Combined with the fact that other VTXs seem to work (and no other problem reports from users) along with the fact that there were recent firmware updates to the VTX-mini to fix other power-up related problem it seems likely. Pinging @MATEKSYS.

AndreaCCIE, 2021-03-08:
I have the same problem (and many others related to tramp protocol already discussed here) on my Matek VTX mini.

MATEKSYS, 2021-03-13:
upgrade VTX-mini firmware  http://www.mateksys.com/?portfolio=vtx-mini#tab-id-8

badgyro, 2021-03-14:
> upgrade VTX-mini firmware http://www.mateksys.com/?portfolio=vtx-mini#tab-id-8

As I wrote earlier, I'm using version 1.1.0 which is the newest as of today (03/14/2021).

closed 2021-04-21 by github-actions[bot]