betaflight/betaflight #10600: FC Freeze - fall out of sky

daleckystepan, 2021-03-05:
### Describe the bug
FC freezes for 4 seconds and than operates normally. My drone fall out of the sky.

### To Reproduce
Unfortunately still no idea after hours of testing. Pretty random but I have some conditions which were satisfied where the issue happens:
1) Very small speed or hovering
2) Low altitude (up to 3 meters)
3) Near the pilot/transmitter (up to 10 meters)

### Expected behavior
Flight without FC freezes and subsequent crashes.

### Flight controller configuration
```
# version
# Betaflight / STM32F7X2 (S7X2) 4.2.8 Feb 15 2021 / 12:10:35 (101738d8e) MSP API: 1.43
# config: manufacturer_id: IFRC, board_name: IFLIGHT_F722_TWING, version: bc19b7dc, date: 2020-04-01T04:37:12Z

# start the command batch
batch start

board_name IFLIGHT_F722_TWING
manufacturer_id IFRC

# name: Cidora SL5

# mixer
mixer CUSTOM
mmix 0  1.000 -1.000  0.843 -1.000
mmix 1  1.000 -1.000 -0.843  1.000
mmix 2  1.000  1.000  0.843  1.000
mmix 3  1.000  1.000 -0.843 -1.000

# feature
feature -SOFTSERIAL
feature -OSD
feature GPS
feature TELEMETRY
feature LED_STRIP
feature ESC_SENSOR

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 4 1024 115200 57600 0 115200

# led
led 0 0,0:S:G:12
led 1 1,0:S:AW:0
led 2 2,0:S:AW:0
led 3 3,0:S:AW:12
led 4 12,0:N:C:6
led 5 13,0:N:C:6
led 6 14,0:N:C:6
led 7 15,0:N:C:6

# mode_color
mode_color 6 0 3
mode_color 6 1 12
mode_color 6 7 12

# aux
aux 0 0 0 900 1300 0 0
aux 1 1 2 1700 2100 0 0
aux 2 2 2 1300 1700 0 0
aux 3 5 3 1700 2100 0 0
aux 4 13 1 1700 2100 0 0
aux 5 31 3 900 1300 0 0

# master
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 90
set dyn_notch_max_hz = 350
set acc_trim_pitch = 8
set acc_trim_roll = -1
set acc_calibration = 38,-41,-2,1
set align_mag = CW270FLIP
set mag_align_pitch = 1800
set mag_align_yaw = 2700
set mag_calibration = 436,243,-15
set min_check = 1000
set max_check = 2000
set rssi_scale = 200
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set blackbox_p_ratio = 16
set dshot_idle_value = 450
set dshot_bidir = ON
set failsafe_delay = 5
set vbat_warning_cell_voltage = 340
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_descend_rate = 100
set yaw_deadband = 5
set debug_mode = FPORT
set gyro_2_align_yaw = 900
set name = Cidora SL5

profile 0

# profile 0
set anti_gravity_gain = 5000
set feedforward_transition = 30
set iterm_rotation = ON
set throttle_boost = 8
set p_pitch = 61
set i_pitch = 108
set d_pitch = 50
set f_pitch = 114
set p_roll = 55
set i_roll = 102
set d_roll = 47
set f_roll = 108
set p_yaw = 60
set i_yaw = 108
set f_yaw = 108
set level_limit = 75
set d_min_roll = 30
set d_min_pitch = 34
set idle_min_rpm = 21

rateprofile 0

# rateprofile 0
set thr_mid = 25
set thr_expo = 25
set rates_type = ACTUAL
set roll_rc_rate = 15
set pitch_rc_rate = 15
set yaw_rc_rate = 15
set roll_expo = 50
set pitch_expo = 65
set yaw_expo = 45
set roll_srate = 80
set yaw_srate = 50

# end the command batch
batch end
```

```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: CAMERA_CONTROL
A01: LED_STRIP
A02: FREE
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: GYRO_EXTI
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: GYRO_CS 1
B00: FREE
B01: FREE
B02: GYRO_EXTI
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: PINIO 3
B09: FLASH_CS
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: PREINIT 4
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: ADC_CURR
C03: GYRO_CS 2
C04: LED 1
C05: FREE
C06: MOTOR 2
C07: MOTOR 4
C08: MOTOR 1
C09: MOTOR 3
C10: SERIAL_TX 4
C11: SERIAL_RX 4
C12: FREE
C13: PINIO 1
C14: PINIO 2
C15: BEEPER
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2: LED_STRIP
TIM3: FREE
TIM4: FREE
TIM5:
    CH1: CAMERA_CONTROL
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: LED_STRIP
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 - Flight controller: IFLIGHT SucceX-D F7 V2.1 TwinG Flight Controller (HD Version), TARGET: IFLIGHT_F722_TWING
 - Other components and how they are wired 
ESC: IFLIGHT SucceX 50A 2-6S BLHeli_32 Dshot1200 4-in-1 ESC, TARGET: iFlight-BL32-4IN1, FW: 32.7
UART GPS and I2C Compass: Matek m8q-5883, FW: Ublox 3.01
FPORT Receiver: Frsky R-XSR, FW: ACCST FPORT 2.1.0 (not the latest 2.1.1 beacause of telemetry issue, another story)
no camera or vtx

### What I have tried
* different batteries
* changed to 8 CH 9ms and Failsafe No pulses according to https://github.com/betaflight/betaflight/wiki/SBus-FPort-and-RC-Smoothing-when-using-OpenTX
* Blackbox data analyzing

### What I have found
* Screenshot from crash
[image]
[image]
* Blackbox from crash
[Blackbox.zip](https://github.com/betaflight/betaflight/files/6093595/Blackbox.zip)
* Raw blackbox analysis where we can see that usually 16 cycles took around 2ms. During the crash 16 cycles took more than 4000ms.
[image]

It seems to me that something blocks the CPU to process cycles unfortunately I don't know how to find what can cause it. 

I am willing to debug this issue. I can update firmware, compile firmware I even have oscilloscope but I have no ideas :)

Thank you for your time and support

daleckystepan, 2021-03-08:
Finally I have some news about my issue. Inspired by https://github.com/betaflight/betaflight/pull/10605 I plug in my USB after crash and my compass max/us was about 4000ms. So the issues is probably in compass, BF, I2C or something like this.

Meanwhile I have updated to latest BF 4.3 without any improvement.

daleckystepan, 2021-03-08:
```
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms
00 - (         SYSTEM)      9       7       1    0.5%    0.5%        24
01 - (         SYSTEM)    750      12       1    1.4%    0.5%      3169
02 - (           GYRO)   7937      42      23   33.8%   18.7%    710239
03 - (         FILTER)   7937      39      17   31.4%   13.9%    524816
04 - (            PID)   7937      93      44   74.3%   35.4%   1513488
05 - (            ACC)    743      28      15    2.5%    1.6%     54576
06 - (       ATTITUDE)    367      30      10    1.6%    0.8%     22083
07 - (             RX)     88      70      25    1.1%    0.7%     10323
08 - (         SERIAL)     79  377553      34 2983.1%    0.7%     12651
09 - (       DISPATCH)    751      12       0    1.4%    0.0%      1791
10 - (BATTERY_VOLTAGE)     39      16       2    0.5%    0.5%       376
11 - (BATTERY_CURRENT)     39      12       1    0.5%    0.5%       262
12 - ( BATTERY_ALERTS)      5       8       2    0.5%    0.5%        36
13 - (         BEEPER)     79      12       2    0.5%    0.5%       499
14 - (            GPS)     79      73      10    1.0%    0.5%      3438
15 - (        COMPASS)      9 4096402   25452 3687.2%   23.4%    201073
16 - (           BARO)    732      39       2    3.3%    0.6%      7696
18 - (       ALTITUDE)     28      24      10    0.5%    0.5%      1654
20 - (      TELEMETRY)    188      16       0    0.8%    0.0%      2759
21 - (       LEDSTRIP)     79      46       7    0.8%    0.5%      2217
24 - (     ESC_SENSOR)     79      15       1    0.6%    0.5%       708
25 - (            CMS)     19      10       1    0.5%    0.5%       124
26 - (        VTXCTRL)      5       8       1    0.5%    0.5%        15
27 - (        CAMCTRL)      5       7       1    0.5%    0.5%        11
29 - (    ADCINTERNAL)      1      12       4    0.5%    0.5%        15
30 - (       PINIOBOX)     18      13       2    0.5%    0.5%       131
RX Check Function                  14       3                      1005
Total (excluding SERIAL)                      3846.7%  102.6%

```

jflyper, 2021-03-09:
It's most likely that I2C communication with the mag device are failing for some reason and the mag/I2C driver is trying to recover from the error.

Looking at the CLI status report, we see that

I2C Errors: 9305
Uptime: 4271 [sec]
COMPASS sampling rate: 10 [Hz]
COMPASS sampling time (total): 201.073 [sec]

Time out values hard coded inside the I2C driver is 25 [msec], so 9305 timeouts will take 232 [sec], which is roughly equal to the total sampling time.

This means there are frequent errors occurring at I2C driver level, at an average rate of once per 4271[sec] / (201 [sec] * 10 [Hz]) = 2 [sec].

The max driver time of 4 second is very special, and I don't have an explanation at the moment, but 4 second is only a fraction of time compared to the total sampling time of 201 seconds, so we have to assume average rate of one error per every 2 seconds.

jflyper, 2021-03-09:
I want to know if the I2C Error count increases on the bench, too.

daleckystepan, 2021-03-09:
We can divide it to 3 stages:
1) After FC boot: 6 errors
2) Normal operation: still 6 errors and working without any issue
3) Issue: Suddenly (I dont know how to do this manually) mag freezes, baro freezes, I2C errors start to increase forever

I experienced same behavior just plugged to the USB (no battery backs connected, no power to ESC)

jflyper, 2021-03-09:
> I experienced same behavior just plugged to the USB (no battery packs connected, no power to ESC)

I suspect a hardware originated issue in I2C mag. How strong is it pulled up, and how long is the wire from FC to GPS/mag puck?

daleckystepan, 2021-03-09:
SCL line is under 3V at the end of transfer at maximum and around 1V (maybe 0.8V) at minimum. Regardless of connected compass or activated baro. Undershoots and also weired phenomenon: voltage goes constatntly down during transfer and up after transfer. Two screenshots attached. Resistance between 3V3 and SDA/SCL is arount 2.2kOhm without compass and around 1.8kOhm with compass connected (measured without power supply).
[image]

[image]

daleckystepan, 2021-03-10:
Compass datasheet: This device supports standard and fast speed modes, 100kHz and 400kHz, respectively.
And SCL from o-scope: around 983kHz
set i2c2_overclock = OFF    set freqeuncy around 446kHz

We will see. More testing is needed.

daleckystepan, 2021-03-10:
Still some errors but no lockup and not increasing forever
```
# status
MCU F722 Clock=216MHz, Vref=3.27V, Core temp=52degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 4058, max available: 16384
Devices detected: SPI:2, I2C:2
Gyros detected: gyro 1, gyro 2
GYRO=ICM20689, ACC=ICM20689, BARO=BMP280, MAG=QMC5883
OSD: NONE
System Uptime: 1665 seconds, Current Time: 2021-03-10T20:39:06.612+00:00
CPU:10%, cycle time: 125, GYRO rate: 8000, RX rate: 33, System rate: 9
Voltage: 15 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 43
SD card: Not configured
Arming disable flags: RXLOSS CLI MSP RPMFILTER

# tasks
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms
00 - (         SYSTEM)     10       2       0    0.5%    0.0%         9
01 - (         SYSTEM)    992       7       0    1.1%    0.0%      1307
02 - (           GYRO)   8000      28      23   22.9%   18.9%    299337
03 - (         FILTER)   8000      24      16   19.7%   13.3%    217729
04 - (            PID)   8000      61      44   49.3%   35.7%    600847
05 - (            ACC)    992      19      14    2.3%    1.8%     23544
06 - (       ATTITUDE)    495      16      11    1.2%    1.0%      9500
07 - (             RX)     33      54      48    0.6%    0.6%      2704
08 - (         SERIAL)    100    4136       2   41.8%    0.5%      5106
09 - (       DISPATCH)    993       3       1    0.7%    0.5%       713
10 - (BATTERY_VOLTAGE)     50       5       2    0.5%    0.5%       155
11 - (BATTERY_CURRENT)     50       3       0    0.5%    0.0%        66
12 - ( BATTERY_ALERTS)      5       3       2    0.5%    0.5%        14
13 - (         BEEPER)    100       4       2    0.5%    0.5%       215
14 - (            GPS)    100      30       8    0.8%    0.5%      1435
15 - (        COMPASS)     10     282     281    0.7%    0.7%      4657
16 - (           BARO)    123      48      31    1.0%    0.8%      6209
18 - (       ALTITUDE)     40      15      11    0.5%    0.5%       708
20 - (      TELEMETRY)    247       2       1    0.5%    0.5%       977
21 - (       LEDSTRIP)    100      27       5    0.7%    0.5%       842
24 - (     ESC_SENSOR)    100       3       1    0.5%    0.5%       210
25 - (            CMS)     20       3       2    0.5%    0.5%        51
26 - (        VTXCTRL)      5       2       1    0.5%    0.5%         6
27 - (        CAMCTRL)      5       1       1    0.5%    0.5%         4
29 - (    ADCINTERNAL)      1       4       4    0.5%    0.5%         6
30 - (       PINIOBOX)     20       3       2    0.5%    0.5%        52
RX Check Function                   3       2                       107
Total (excluding SERIAL)                       107.5%   79.8%
```

daleckystepan, 2021-03-10:
Error visible on graphs
[image]

daleckystepan, 2021-03-10:
Issue can be here. It returns
983kHz for 800kHz request
446kHz for 400kHz request

It shouldn't return higher frequency than requested for safety reason.

https://github.com/betaflight/betaflight/blob/bc4372588b43ccf93373b0c466a18b3be2fa7d17/src/main/drivers/bus_i2c_hal_init.c#L352

jflyper, 2021-03-11:
Hmm... `i2cClockTIMINGR` is written based on ST reference manual. May be there are something that is overlooked.

daleckystepan, 2021-03-11:
These issue emerged during debugging:
1. I2C overclocking enabled at 983kHz by default
2. QMC5883 max speed is 400kHz so BF violates datasheet
3. BF I2C frequency calculation (i2cClockTIMINGR) returns higher frequency than requested

daleckystepan, 2021-03-11:
I have tried to compare I2C frequency calculation between iNav and BF. It seems to be the same.

I also tried to verify calculation with reference manual (https://www.st.com/resource/en/reference_manual/dm00305990-stm32f72xxx-and-stm32f73xxx-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) however I don't understand it enough to be sure.

hydra, 2021-03-11:
* Did you measure the I2C pull-up resistor values?
* Did you try with an even lower I2C clock speed?
* Did you manage to capture scope traces of the SDA/SCL lines on your scope?
*

daleckystepan, 2021-03-11:
- Yes. 1.8kOhm with compass connected, 2.2kOhm with just baro on the board
- Yes. Tried without overclocking with 446kHz. Still I2C errors but no FC freezes anymore. Haven not tried with frequencies lower than 446 kHz - not possible without code modification.
- Yes. SCL attached

[image]

hydra, 2021-03-12:
I'd be tempted to try a slightly stronger pull-up as those rise times dont look very clean.

however reducing the I2C clock speed will allow the i2c signal sampling to happen later, once the signals have settled, so suggest making code modifications before changing any pullups.

daleckystepan, 2021-03-12:
1.8kOhm seems to be good value according to documents I have found. Usually 1kOhm is minimal value and we are close with 1.8kOhm.

Rise time is about 138 ns according to my o-scope. That is withing 300ns ranger according to I2C specs.

I have bigger concerns about frequency.

From (https://www.ti.com/lit/an/slyt770/slyt770.pdf?ts=1615587063498):
> In rare cases, extremely fast rise/fall times could falsely
trigger an electrostatic discharge (ESD) cell on the SDA/
SCL pin of edge-rate-triggered I2C devices and latch the
SDA/SCL lines low until the ESD cell recovers. This situation usually occurs when an I2C slave rated for 400 kHz is
on an active I2C bus that is operating at 1 MHz or faster.
False triggers are a result of improper I2C slave placement/
isolation; do not place 400-kHz slaves on I2C buses operating faster than 400 kHz.

mikeller, 2021-03-15:
Interesting case re overclocking:
- yes, this puts the I2C clock outside of the acceptable range for some devices. But despite this this is the first time that problems are reported because of it - it looks like most devices work well with it;
- disabling overclocking by default will have a negative impact on loop time stability, as the I2C driver in Betaflight is busy-waiting, so it is not really desirable.

daleckystepan, 2021-03-15:
- I have found some discussions (BF, iNav and other forums) about issues with I2C devices (mainly compass) - undetected, not working, not stable etc. Maybe a lot of them don't know what cause it (or what is the cause of FC freeze), maybe a lot of them simply disable it. I don't know. Just guessing. There is not much people with o-scope at home.
- Completely agree that causing jitter in loop time is very bad approach how to solve this issue.

What can we do in fact?
- Ignore the issue (definitely valid reason if I am the only one)
- Rewrite I2C driver with usage of interrupts/dma and don't use busy-waiting so we can decrease frequency under 400kHz?
- Use state machines and communicate with I2C devices in parts so loop time will not be much affected?
- Any other ideas?

mikeller, 2021-03-16:
Agreed with your list of options. The reality is that the value of I2C is rather marginal in Betaflight (as compared to other projects like iNav that rely more on the use of peripherals like barometer, compass, ...), so 'do nothing' seems to be a reasonable first approach. Rewriting the driver could be done, but would probably come at the cost of additional flash space usage, which isn't a great option.

To my understanding of your comments, if you set `i2c2_overclock = off` and add external terminators then it becomes usable for you? If so then I think this is a valid workaround, and seems to be needed for certain peripherals.

daleckystepan, 2021-03-16:
I would switch "do nothing" to something like "write errata" or "document this" and link this issue. We can hopefully help some people.

If I switch `i2c2_overclock = off`:
- It makes a little bit of  jitter in loop time
- still I2C errors, sometimes bad read from compass
- no FC freezes anymore

My final solution:
- disable compass completely (no real usecase - just GPS rescue and MAG mode AFAIK)

klutvott123, 2021-03-17:
I just connected a scope to the i2c bus on a matekf722 and am seeing the same high frequencies as @daleckystepan.

Here
https://github.com/betaflight/betaflight/blob/b30f16fe91fa5a0c836e65289d9e0ab081c3d1f2/src/main/drivers/bus_i2c_hal_init.c#L217
The fall time is assumed to be 100ns(same as rise time tr). In reality the fall time is closer to something like 10ns. Fall time is used in ```SCLH```/```SCLL``` calculation. Higher fall time results in lower counter values and higher frequency.

Looks like this with fall time set to 10ns:

[image]
[image]

Of course it's just another best guess, but for most setups it's probably a better guess than 100ns. Maybe we should change it to stay closer to specs.

klutvott123, 2021-03-17:
Looks like I didn't JUST change the fall time. 
https://github.com/betaflight/betaflight/blob/b30f16fe91fa5a0c836e65289d9e0ab081c3d1f2/src/main/drivers/bus_i2c_hal_init.c#L263-L264
I removed the -1's from here because in the reference manual I don't see a -1. @jflyper? 

https://github.com/betaflight/betaflight/blob/b30f16fe91fa5a0c836e65289d9e0ab081c3d1f2/src/main/drivers/bus_i2c_hal_init.c#L252-L253
I also changed these to only include the delay from 2 clock periods instead of 3.

> The duration of tSYNC2 depends on these parameters:
– SCL rising slope
– When enabled, input delay induced by the analog filter.
– When enabled, input delay induced by the digital filter: DNF x tI2CCLK
– Delay due to SCL synchronization with I2CCLK clock (2 to 3 I2CCLK periods)

Nothing more. I promise.

daleckystepan, 2021-03-17:
Fall time estimation around 10ns is much more realistic than 100ns. Agreed. This is candidate for PR.

Here is the reason for `-1`
[image]

which is in reality here
https://github.com/betaflight/betaflight/blob/b30f16fe91fa5a0c836e65289d9e0ab081c3d1f2/src/main/drivers/bus_i2c_hal_init.c#L256

This seems to be redundant
https://github.com/betaflight/betaflight/blob/b30f16fe91fa5a0c836e65289d9e0ab081c3d1f2/src/main/drivers/bus_i2c_hal_init.c#L263-L264

However I am not expert at this topic. Just had and issues and deep dive as much as I can. Now I am hitting the border.

klutvott123, 2021-03-17:
@daleckystepan Yes, I see now. When calculating these
https://github.com/betaflight/betaflight/blob/b30f16fe91fa5a0c836e65289d9e0ab081c3d1f2/src/main/drivers/bus_i2c_hal_init.c#L258-L259
the result is truncated and then we subtract 1. So worst case we're actually subtracting 2 and ending up with a much higher frequency.

klutvott123, 2021-03-23:
@daleckystepan The max/us of 4096402 youre seeing is likely because of the i2c timeout
https://github.com/betaflight/betaflight/blob/76d9babab6105fb896abec7897eb195e17141f25/src/main/drivers/bus_i2c_impl.h#L28-L30

0x1000 = 4096. 

We're sending this value as the timeout to be used by the HAL functions like here
https://github.com/betaflight/betaflight/blob/76d9babab6105fb896abec7897eb195e17141f25/src/main/drivers/bus_i2c_hal.c#L118

The thing is that the HAL functions expect the timeout to be in sys ticks. 1 sys tick is 1 millisecond so the actual timeout is 4096ms. So it's just spinning away doing nothing for 4 seconds when it times out. 

I just made a PR with a fix for this https://github.com/betaflight/betaflight/pull/10631

closed 2021-03-24 by mikeller

daleckystepan, 2021-03-24:
Thank you guys for your help and PR.