betaflight/betaflight #10601: Cannot change RSSI dBm value below 1 in Betaflight OSD menu

ponuryzrzeda, 2021-03-07:
### Describe the bug
For osd_rssi_dbm_alarm allowed range is between -130 and 20. 
With the default, in the OSD menu (BTFL menu -> OSD -> Alarms) i've got a value of 196 for a OSD dBm alarm. Cannot change this below 1 and when save/exit is selected,  failed value is saved in the osd_rssi_dbm_alarm variable. Without saving remains valid, negative value.
CLI:
```
# get osd_rssi_dbm_alarm 
osd_rssi_dbm_alarm = 196
###ERROR: get: CORRUPTED CONFIG: osd_rssi_dbm_alarm = 196###
Allowed range: -130 - 20
Default value: -60
```
### To Reproduce
In Betaflight OSD menu just Save/Exit even without changing anything.

### Expected behavior
OSD menu should support negatives values and set proper value for osd_rssi_dbm_alarm

### Flight controller configuration
```
# version
# Betaflight / STM32F405 (S405) 4.2.8 Feb 15 2021 / 12:08:36 (101738d8e) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF405STD, version: 310c299f, date: 2020-05-22T01:08:06Z

# start the command batch
batch start

board_name MATEKF405STD
manufacturer_id MTKS

# name: Realacc220

# resources
resource MOTOR 5 NONE
resource SERIAL_TX 11 A15
resource I2C_SCL 3 A08
resource I2C_SDA 3 C09

# feature
feature -ANTI_GRAVITY
feature GPS
feature RSSI_ADC

# beeper
beeper -RX_LOST_LANDING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_SET

# map
map TAER1234

# serial
serial 2 2 115200 57600 0 115200
serial 4 8192 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# aux
aux 0 0 0 1950 2050 0 0
aux 1 1 1 950 1050 0 0
aux 2 13 2 1950 2050 0 0
aux 3 35 2 1450 1550 0 0
aux 4 47 1 1450 1550 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 25 200 400
vtxtable powerlabels 25 200 500

# rxfail
rxfail 3 h

# master
set acc_calibration = 0,-3,-7,1
set min_check = 1005
set max_check = 1995
set fpv_mix_degrees = 30
set rx_min_usec = 925
set rx_max_usec = 2075
set blackbox_p_ratio = 64
set min_throttle = 1015
set dshot_idle_value = 300
set motor_pwm_protocol = DSHOT300
set failsafe_delay = 5
set failsafe_switch_mode = STAGE2
set failsafe_throttle_low_delay = 10
set bat_capacity = 1300
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set pid_process_denom = 2
set osd_warn_batt_not_full = OFF
set osd_rssi_alarm = 35
set osd_rssi_dbm_alarm = -60
set osd_cap_alarm = 1200
set osd_alt_alarm = 85
set osd_tim1 = 0
set osd_tim2 = 1537
set osd_vbat_pos = 6562
set osd_rssi_pos = 6200
set osd_link_quality_pos = 6198
set osd_tim_2_pos = 6262
set osd_flymode_pos = 44
set osd_g_force_pos = 4557
set osd_throttle_pos = 449
set osd_ah_sbar_pos = 4302
set osd_ah_pos = 4174
set osd_current_pos = 6581
set osd_mah_drawn_pos = 6615
set osd_craft_name_pos = 456
set osd_gps_speed_pos = 6210
set osd_gps_lon_pos = 464
set osd_gps_lat_pos = 449
set osd_gps_sats_pos = 6232
set osd_home_dir_pos = 6221
set osd_home_dist_pos = 6223
set osd_flight_dist_pos = 4138
set osd_altitude_pos = 6177
set osd_power_pos = 4482
set osd_warnings_pos = 6185
set osd_avg_cell_voltage_pos = 6594
set osd_disarmed_pos = 6507
set osd_core_temp_pos = 4504
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_band = 1
set vtx_channel = 7
set vtx_power = 1
set vtx_freq = 5745
set vcd_video_system = PAL
set displayport_max7456_wht = 1
set gyro_1_align_yaw = 1800
set name = Realacc220

profile 0

rateprofile 3

# end the command batch
batch end
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: LED 2
A15: SERIAL_TX 11
B00: FREE
B01: ADC_RSSI
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: LED 1
B10: OSD_CS
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: SDCARD_CS
C02: GYRO_CS 1
C03: GYRO_EXTI
C04: ADC_CURR
C05: ADC_BATT
C06: MOTOR 1
C07: MOTOR 2
C08: MOTOR 3
C09: MOTOR 4
C10: SERIAL_TX 3
C11: SERIAL_RX 3
C12: SERIAL_TX 5
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH1: SERIAL_TX 11
TIM3:
    CH1: MOTOR 1
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH2: MOTOR 2
    CH3: MOTOR 3
    CH4: MOTOR 4
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: TIMUP 8
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
Betaflight / STM32F405 (S405) 4.2.8 Feb 15 2021 / 12:08:36 (101738d8e) MSP API: 1.43
config: manufacturer_id: MTKS, board_name: MATEKF405STD, version: 310c299f, date: 2020-05-22T01:08:06Z
board: manufacturer_id: MTKS, board_name: MATEKF405STD
Frsky R-XSR rx 
Analog RSSI enabled

etracer65, 2021-03-07:
This is a bug in the OSD menus. Working on a fix.

etracer65, 2021-03-07:
See #10603

PR #10603 (2021-04-26): Fix CMS RSSI dBm alarm data type and range
Fixes #10601 

Related to #9550

Data type was `uint8` and needed to be `int16` to accomodate the range change to negative numbers.
closed 2021-04-26 by mikeller