betaflight/betaflight #10604: Request for user defined arm delay

DutchFPV, 2021-03-07:
Problem:  
Spinning propellers are dangerous.  Even when rigid safety procedures are followed accidents happen.  I have a rigid procedure for arming and disarming my quads.  In over 5 years I've never taken a prop...until the other day.  When a wrinkle in the universe caused my controller to bump against my chest, somehow simultaneously pushing the left stick down and hitting the arm switch while the quad was in my hand.  I was holding the quad the safest possible way, I heard the arm beep and knew what was happening, but a prop still knicked me.  It could have been worse.  Its sounds random but accidents like this are out there.  

Solution:
In addition to the 'beep on arm' option already installed, it would be great to set an arm delay.  Maybe 0-10 seconds.  So that you get a beep acknowledging the arm sequence, but still have a few seconds before the motors start.  Beyond my unique experience I think there are other good reasons for it.  At races for instance, it would buy a start official a few extra seconds to clear the field before anyone goes hot.  I feel like its a rare occasion when instantaneous arming is really necessary.  And it could still be an option. 

Alternative:
I know the pre arm option already exists.  Its a great additional safety gate.  But there is still a point where one switch flip and you are instantly live.  I feel like a built in delay between the last arm command and the props actually spinning, is a purposeful additional safety measure.  Should one desire this additional measure of safety.

Additional Info:
I will refrain from posting pictures of my bloody thumb to further compel you.

closed 2021-03-07 by github-actions[bot]

haslinghuis, 2021-03-10:
@DutchFPV That was a bad experience, but as your mentioned we already have a `pre-arm` option. A delay would only enrage racers unless the seconds could be defined, but I guess it's just bad luck and hopefully just once in a life time.

DutchFPV, 2021-03-11:
Thanks for the follow up.  It was just a thought.  To be clear, what I was suggesting was a user defined interval from 0-X seconds.  So racers could set it to 0 and have nothing to worry about.  But in general I'd argue that instant arming is the most dangerous, least useful option.  Just take 5 seconds, forget 'how its always been' and walk through it in your mind.  When do you really need instant arming?  You just don't.  What is the real merit?  When is it really absolutely necessary?  Pretty much never.  But when is it safer to have a few extra seconds warning before the motors go hot?  Every time.  Really think about it and I think you'll see its inarguably safer, and with the option to set the interval to 0, you really don't lose any functionality.  You just gain safety for every other case.  If you were making a product that you owned the liability for, you would have done this the first day.  

That said, no hard feelings.  My incident got me thinking about it, and I thought it was worth sharing.  You guys do an amazing job with this software.  I've been using it since the first days of Cleanflight and let's be honest, every version to date puts everything Microsoft has put out in 30 years to shame, LOL.  So thanks for hearing me out and keep up the good work.
Best,Derek 
DutchFPV

Derek C. Donohue 
***@***.***

DutchFPV, 2021-03-16:
Hey FYI I've gotten a little more insight into my incident and I've realized its worse than just a missing feature.  You guys have a major arming issue in version 4.2 with F7 boards.  Specifically that you either get no arming or you get a full throttle arm.  Which taken together means you're doing everything trying to arm, then when you get it to arm it unexpectedly goes to full throttle.  I've experienced both with 2 brand new F7 boards, 1 Matek and 1 Holybro.  Its noted in multiple threads on Reddit too.  Its been raised on GitHub multiple times but your team seems very dismissive of it.  This is a very dangerous situation and a major liability to board manufacturers.  I've already written both companies to warn them.  Both boards reverted to BF 4.1 arm predictably and flawlessly.  Its definitely a problem in 4.2 (both had 4.2.8 installed BTW)

I'm not going to waste my time reporting this.  Only to have my post given a failing grade.  You've already been told.  I can't force you to take this serious.  But I'm going to take the path of least resistance and tell everyone not to use 4.2 and tell everyone they know not to use it either.  It's not to be a jerk or try to 'cancel' you guys, but because its a legitimate safety issue.  When I had my incident the drone armed full throttle and it could have cost me my thumb.   Regardless of who or what's to blame people deserve to know there's a serious risk involved.  I love BF and am not against you in any way.  But you guys need to fix this.  And be transparent about the fact that its a risk that exists.  Seriously.  

Best,Derek

Derek C. Donohue ***@***.***

mikeller, 2021-03-23:
@DutchFPV: If you are not providing any information to help debug your problem, and only provide hearsay about a problem without providing references then you should not be surprised if nobody is helping you getting it fixed.

DutchFPV, 2021-03-23:
The hole in my thumb is not 'hearsay'.  Got plenty of gruesome pictures if you want them. But sadly you closed my request because of formatting.  So you shouldn't be surprised that I switched to FalcoX this weekend.  

Thanks for the years of enjoyment.  Best of luck!

Derek

Derek C. Donohue ***@***.***

haslinghuis, 2021-03-23:
@DutchFPV, Betaflight uses a bot to close issues which are not following template, it's nothing personal. To look into an issue we need some kind of information.