betaflight/betaflight #10607: Improvement of auto RC filtering with Crossfire Receivers

N4V1G4T0R, 2021-03-08:
### Is your feature request related to a problem? Please describe.

Auto RC signal filtering is optimized for the rc signal at arming. In most cases this will be the 150hz mode, because of a short distance between TX and RX. But during normal flights, crossfire will switch accordingly between the modes for optimal transmission. At a decent distance, the 50hz mode will be active most of the time. So, actually rc signal filtering is not optimal for most of the time.

### Describe the solution you'd like

Better handling of the crossfire switching and rc signal auto filtering.

### Describe alternatives you've considered

Ppl fix crossfire to 50hz, but missing out on 150hz and 4hz.
Or switch filters to manual, which requires some tuning.

I hope I got everything right. ;-)

etracer65, 2021-03-08:
RC smoothing detects the signal change and automatically adjusts the filtering (assuming you've set the cutoffs to AUTO). What you are asking for already exists.

N4V1G4T0R, 2021-03-08:
OK, thanks for the fast answer. That‘s good news. So there is no need to switch off crossfire switching (anymore). 👍

But now I am confused. We were just recently discussing this. Maybe it‘s just the switching itself that makes issues?

etracer65, 2021-03-08:
I'm not sure what your confusion is about. The ability to adapt to changing frame rates has been a feature from the beginning.

N4V1G4T0R, 2021-03-09:
Haha, I‘m not confused about the feature, Bruce.

I am confused about the glitch in the rc signal trace we have been discussing in the FB Group. If it is not comming from the crossfire switching, what is it coming from then... 🤔

N4V1G4T0R, 2021-03-09:
Here's the confusing thread. I apologize in advance to @betaflight @tbs if I got that wrong.

https://www.facebook.com/groups/291745494678694/permalink/1048745628978673/?comment_id=1048803805639522&reply_comment_id=1049928718860364

etracer65, 2021-03-09:
That FB link is to a closed group and I'm not a member. Can't see the post.

N4V1G4T0R, 2021-04-09:
Probably solved by #10650

closed 2021-04-09 by N4V1G4T0R