betaflight/betaflight #10614: SpeedyBee TX500 vtxtable incorrect - linked from Betaflight wiki

etracer65, 2021-03-11:
Have tried contacting the manufacturer through their website support pages with no response.

The power level sections of the Speedybee TX500 Tramp `vtxtable` hosted by the manufacturer and linked to from the Betaflight wiki is incorrect and nonfunctional. In particular, the 500mw power setting will not work. The corrected power level section should be:
```
		"powerlevels_list": [{
				"value": 25,
				"label": "25"
			},
			{
				"value": 200,
				"label": "200"
			},
			{
				"value": 400,
				"label": "500"
			}
		]
```
Since the file is hosted by the manufacturer so there's no way for us to correct this. As the manufacturer hasn't responded, I've removed the links to the incorrect tables from the wiki and directed users to this issue instead until such time the manufacturer responds and corrects the files.

The links to the incorrect files are as follows:
https://speedybee.s3.amazonaws.com/vtxtable/betaflight/TX500/speedybee_tx500_vtx_table_irc_tramp_us.json
https://speedybee.s3.amazonaws.com/vtxtable/betaflight/TX500/speedybee_tx500_vtx_table_irc_tramp_eu.json

mikeller, 2021-03-22:
Hey @azolyoung do you still have contacts within SpeedyBee?

azolyoung, 2021-03-26:
@etracer65  thanks for your help! The issue was fixed now

etracer65, 2021-03-26:
Restored the links in the vtxtable wiki page.

closed 2021-03-26 by etracer65