betaflight/betaflight #10615: Is there anyone interested in Express LRS 2.4 GHz SPI-RX implementation?

vodka-bears, 2021-03-12:
### Is your feature request related to a problem? Please describe.

SPI-RX range with cc2500 is kinda low. The proposed functionality is supposed to provide an opportunity for manufacturers to produce longer-range FCs with SPI-RX.

### Describe the solution you'd like

As many people here might know there's an open source project called ExpressLRS (https://github.com/ExpressLRS/ExpressLRS). They make a faster RC system for both 868/915 MHz and 2.4 GHz bands. 2.4 GHz system for now performs better than brand new systems like Ghost and Tracer (they're based on the same chip). I'm gonna implement Express LRS rx-side functionality in Betaflight. For tests I'm gonna get a Betafpv AIO 1-2S whoop-slyle flight controller. It has a pin header for an "external" SPI-RX. I'm gonna make a custom SPI-RX board with sx1280 chip for debugging and testing.

### Describe alternatives you've considered

I didn't consider any alternatives. The main reason I'm writing this is to find out if there's interest in this and make sure that it's gonna be worth it or not.

etracer65, 2021-03-12:
I think it's an interesting idea. But targeting the existing cc2500 F411 audience might be difficult. The F411 is very close to running out of flash so I'm not sure it's going to be practical to add another significant RX stack there. But I guess it depends on how much code it ends up being.

mikeller, 2021-03-22:
Looking into stats, Redpine and Spektrum that are currently included do not seem to be having many takers, so we might want to try swapping them for Express LRS:

[image]

And even if not, this protocol can always be used on custom builds if an implementation for it is contributed to Betaflight.

vodka-bears, 2021-03-25:
@mikeller as a Redpine user I'm strongly against its removal if my opinion counts.

mikeller, 2021-03-25:
@vodka-bears: There's always the option for the maintainer of Redpine to provide custom firmware binaries that support a different set of features, including Redpine.

JyeSmith, 2021-03-25:
@etracer65 I think ExLRS will be great on FCs with F405, F7 etc as a proper 'full range' rx.  Do you see any issue with these MCUs?

etracer65, 2021-03-26:
> I think ExLRS will be great on FCs with F405, F7 etc as a proper 'full range' rx. Do you see any issue with these MCUs?

F405's no problem. F722 are 512KB targets so they are problematic, F745 no problem.

But I really think your priorities are wrong. I'm not sure that any significant manufacturer is going to be willing to produce such a board because user adoption is basically nonexistent at this time. Frankly ExLRS is difficult and confusing to use at this time for the average user. Flashing or hacking other hardware to install an alternate firmware is never going to be widely adopted. Until there are off-the-shelf options that are easy to access it's going to remain niche.

IllusionFpv, 2021-03-26:
@etracer65 there are already 3 manufactor for elrs hardware (happy model, neutron RC, Namimno RC)

etracer65, 2021-03-26:
> there are already 3 manufactor for elrs hardware (happy model, neutron RC, Namimno RC)

Doesn't change the fact that adoption is very low and most consumers don't know anything about it. So I can't see how it would be economically viable or a manufacturer to product a flight controller with onboard support at this time. Maybe in the future, but the focus needs to be on expanding user awareness and adoption.

liyouluo, 2021-04-09:
Regarding the issue of flash, I found that STM32F412 and STM32F413 have 1024K flash. Can they run betaflight firmware?

vodka-bears, 2021-05-07:
Received the betafpv hardware today. The custom hardware is being developed. Expecting to begin software development in several weeks.

@etracer65 Happymodel recently released 2.4 GHz ELRS tx module and receivers. And there's already a shortage on them.

JBKingdon, 2021-05-08:
I'd be very interested to see this done. I'd like to learn how the spi receivers integrate into betaflight.

nightflasher, 2021-05-18:
Happymodel also released 1s whoop aio boards with 900MHz and 2.4GHz receivers, as far as my understanding goes, the receivers won't ever be spi ones, because the lora part uses the spi port of the esp/stm. Also, the ExpressLRS configurator helps building the firmware as well as flashing the receivers (through betaflight) and transmitters as well.

vodka-bears, 2021-05-18:
@nightflasher haven't heard about 2.4 GHz version release but yeah, 900 MHz version has a separate MCU for ELRS. As far as I heard they announced that 2.4 GHz version will have SPI-RX but I haven't seen any implementation attempts yet. The custom PCBs and components are still on their way to me. Received the module and ep1 receivers recently, have a very good impression about how my quads fly now.

JBKingdon, 2021-05-18:
I hear it's a done deal now: https://github.com/phobos-/betaflight/tree/express-lrs

vodka-bears, 2021-05-18:
@JBKingdon wow, that very nice person was first. Well, I won't even need to do anything, awesome. @phobos- please mention "closes 10615" in a pr. Nice work, will test soon.

hydra, 2021-05-18:
I have an H7 board with an SX1280 on-board, I was going to do the port but @phobos- beat me to it while I was in hospital and dealing with H730 stuff.

You can see the board in the video on #10695

phobos-, 2021-05-19:
Yup, the integration is ready, but it's still not close to merging back to betaflight. One main reason is that expresslrs code itself is still in RC phase. Please be patient.

PR #10788 (2021-12-22): ExpressLRS over SPI - sx1280 and sx1276 support
Fixes #10615 

### **Support**
If you have support questions, please feel free to join Discord, me (Phobos#4328@discord) or someone else can help you there. -- https://discord.gg/dS6ReFY

### Intro

This is an initial implementation of SPI ExpressLRS receiver. It serves the purpose of integrating ExpressLRS into whoop aio boards with short to medium range. It has an additional benefit of not using crsf protocol and improving overall end to end latency. Huge thanks to @githubDLG @AlessandroAU @JyeSmith @wvarty @hydra and others for the hardware an [... cut]
closed 2021-12-22 by haslinghuis