betaflight/betaflight #10618: Flywoo411 board Sbus not working

bigtreeteche3, 2021-03-12:
### Describe the bug
I have a Noxe F4 board i'm using for years, now i made a new drone using that specific board. I'm 100% shure to have solderd the wires to the right pins for SBUS (Uart 2) i'm pretty shure to have set it up right in Betaflight but i'm not able to get it to work (I can see the xm plus receiver turning grean so i know its connecting to the radio... Already tried to change uarts, nothing, tried the old target Nox and STM32F411, tried a older version (also Beta 4.1 with one i'm pretty shure it worked before) of the acutal target FLYWOO411.
### To Reproduce
Nothing, always there

### Expected behavior
should show my radio under the receiver tab

### Flight controller configuration
see first part 

PASTE THE OUTPUT OF 'diff' HERE

Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / STM32F411 (S411) 4.2.8 Feb 15 2021 / 12:09:04 (101738d8e) MSP API: 1.43
# config: manufacturer_id: FLWO, board_name: FLYWOOF411, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

board_name FLYWOOF411
manufacturer_id FLWO

# resources
resource MOTOR 1 B00
resource MOTOR 2 A10
resource MOTOR 3 A09
resource MOTOR 4 A08
resource LED_STRIP 1 NONE

# feature
feature -SOFTSERIAL
feature -TELEMETRY
feature -RX_PARALLEL_PWM

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 1 64 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 3 900 1300 0 0
aux 2 13 2 1700 2100 0 0
aux 3 35 1 1700 2100 0 0

# master
set acc_calibration = 18,-20,-166,1
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 16
set rssi_scale = 180
set rssi_offset = -100
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set vbat_warning_cell_voltage = 340
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 4
set osd_rssi_alarm = 25
set osd_rssi_pos = 2477
set osd_warnings_pos = 2473
set osd_avg_cell_voltage_pos = 2092
set osd_stat_used_mah = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set gyro_1_align_yaw = 1800

profile 0

rateprofile 3

# end the command batch
batch end
PASTE THE OUTPUT OF 'resource show all' HERE
$M>	nf
Entering CLI Mode,
# Building AutoComplete Cache ... Done!
# 
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: ADC_BATT
A01: ADC_CURR
A02: FREE
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 4
A09: MOTOR 3
A10: MOTOR 2
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 1
B01: FREE
B02: PREINIT 2
B03: GYRO_EXTI
B04: FREE
B05: PINIO 1
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: CAMERA_CONTROL
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: BEEPER
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: DSHOT_BITBANG 2
    CH2: DSHOT_BITBANG 1
TIM2:
    CH3: CAMERA_CONTROL
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: DSHOT_BITBANG 2
DMA2 Stream 2: DSHOT_BITBANG 1
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
### Setup / Versions
 https://www.banggood.com/20x20mm-Upgrade-Betaflight-F4-Noxe-V1-Flight-Controller-AIO-OSD-5V-8V-BEC-w-or-Barometer-and-Blackbox-for-RC-Drone-FPV-Racing-p-1310419.html?rmmds=myorder&cur_warehouse=CN&ID=17
 
 With mamba F30 mini esc, XM Plus receiver.  Wired to the pins under RX2 (Sbus, grund 4,5v)
[dump.txt](https://github.com/betaflight/betaflight/files/6129362/dump.txt)

etracer65, 2021-03-12:
This is not the place to request help with your configuration or hardware. See https://github.com/betaflight/betaflight#support-and-developers-channel

closed 2021-03-12 by etracer65

mikeller, 2021-03-22:
@bigtreeteche3: You are using the wrong target for your board, this is what it should be:

[image]