betaflight/betaflight #10619: Can't enable Bidirectional Dshot on all ESCS

Afteraffekt, 2021-03-13:
### Describe the bug
Flashing ESCs to latest BLHeli-S, or JESC does not enable Bidirectional Dshot for all 4 motors, 2 or 1 motors wont spin when Bidirectional is set

### To Reproduce
flash bidirectional dshot esc to firmware, and enable in betaflight

### Expected behavior
All 4 motors should be able to spin and show RPM.

### Flight controller configuration

```
diff

# version
# Betaflight / IFLIGHT_H743_AIO (IF7A) 4.3.0 Oct 18 2020 / 23:31:14 (e9952f2747) MSP API: 1.44

# start the command batch
batch start

board_name IFLIGHT_H743_AIO

# name: Cheddar

# feature
feature -AIRMODE
feature GPS

# serial
serial 0 64 115200 57600 0 115200
serial 1 1 115200 57600 0 115200
serial 3 2 115200 57600 0 115200

# beeper
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_SET

# map
map RETA1234

# led
led 0 7,11::CTOBVIW:0

# aux
aux 0 0 5 1700 2100 0 0
aux 1 1 4 900 1300 0 0
aux 2 2 4 1300 1700 0 0
aux 3 46 2 1700 2100 0 0
aux 4 13 0 1300 1700 0 0
aux 5 35 0 1700 2100 0 0
aux 6 36 7 1700 2100 0 0
aux 7 37 2 1300 1700 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 BAND_D   D CUSTOM  5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# master
set gyro_lowpass2_hz = 375
set dyn_notch_q = 250
set dyn_notch_max_hz = 350
set dyn_lpf_gyro_min_hz = 300
set dyn_lpf_gyro_max_hz = 750
set dyn_lpf_gyro_curve_expo = 0
set acc_calibration = -27,-27,-74,1
set mag_hardware = NONE
set serialrx_provider = CRSF
set blackbox_sample_rate = 1/16
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set failsafe_procedure = GPS-RESCUE
set ibata_scale = 100
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_set_home_point_once = ON
set osd_vbat_pos = 2082
set osd_rssi_pos = 2136
set osd_current_pos = 2177
set osd_gps_speed_pos = 2146
set osd_gps_lon_pos = 2050
set osd_gps_lat_pos = 2063
set osd_gps_sats_pos = 2167
set osd_home_dir_pos = 2126
set osd_home_dist_pos = 2199
set osd_compass_bar_pos = 2506
set osd_avg_cell_voltage_pos = 14402
set debug_mode = GYRO_SCALED
set name = Cheddar

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 105
set dyn_lpf_dterm_max_hz = 255
set dterm_lowpass2_hz = 225
set d_min_roll = 0
set d_min_pitch = 0

rateprofile 0

# end the command batch
batch end

# 
```
```
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: PREINIT 2
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_EXT
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: ADC_RSSI
C06: FREE
C07: FREE
C08: CAMERA_CONTROL
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: LED 1
C14: FREE
C15: FREE
D00: GYRO_EXTI
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: LED_STRIP
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: FREE
E02: SPI_SCK 4
E03: FREE
E04: OSD_CS
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: FREE
E09: MOTOR 3
E10: FREE
E11: MOTOR 4
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE
G00: FREE
G01: FREE
G02: FREE
G03: FREE
G04: FREE
G05: FREE
G06: FREE
G07: FREE
G08: FREE
G09: FREE
G10: FREE
G11: FREE
G12: FREE
G13: FREE
G14: FREE
G15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: MOTOR 3
    CH2: MOTOR 4
TIM2: FREE
TIM3:
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM4:
    CH1: LED_STRIP
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15: FREE
TIM16: FREE
TIM17: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 1
DMA1 Stream 1: MOTOR 2
DMA1 Stream 2: MOTOR 3
DMA1 Stream 3: MOTOR 4
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: LED_STRIP
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Setup / Versions
 
 - Flight controller: iFLight H7 Beast - Amazon.com
 
 - Other components: GPS BN-180, Vista Nebula Pro, Tracer Nano RX
 
 - How are the different components wired up:
RX is to R1/T1
GPS to R4/T4
Vista to R2/T2

Itsblunty mentioned DMT Timings being possible.

pilkinn, 2021-03-19:
I have the same issue. The only way I ever got bidirectional DSHOT working was with BTF 4.2.6 + a version of BLheli_M. However the motors sounded like a bag of gravel. With JESC 2 motors wouldn't spin at all. With Bluejay one motor refused to spin. 

Might try the very latest development version of BTF 4.3 with the BLHeli_M foirmware again (but the board ships with 4.3 so not holding out much hope).

mikeller, 2021-03-21:
This was discussed in https://github.com/betaflight/betaflight/pull/10383#issuecomment-776199710 - it appears there is a version of the IFLIGHT_H743_AIO that has got a wrong resistor on one of the ESCs, resulting in bidirectional Dshot not working properly.

Afteraffekt, 2021-03-21:
> This was discussed in #10383 (comment)

Is this all of them, or just a few select?

mikeller, 2021-03-22:
@Afteraffekt: See the original comment. Since we do not have schematics for this board we do not know what exactly is going on, just that some ESCs seem to be having a hardware problem.

bexamous, 2021-04-11:
Later comments seem to indicate wrong resistor may not be cause, in https://github.com/betaflight/betaflight/pull/10383#issuecomment-803608826

> I have two of these boards - one received in the post a couple of days ago. I have measured all of the resistors and they are all 1k, so feels like the resistor idea might be a red herring.

In last comments seems like issue is being avoided at least in ArduCopter, and WAR? is sorta suggested in 
 https://github.com/betaflight/betaflight/pull/10383#issuecomment-806065082:
> So my solution works by essentially setting the pin to low speed once at the beginning rather than every pulse (and avoiding DMA problems with reading the GCR pulses).
Now I am guessing you are setting the speed higher than low under the assumption that DSHOT1200 (for example) needs a faster rise time, but in my testing on the scope I'm not seeing a material difference - DSHOT1200 works just as well with low speed as mid speed. So my suggestion would be to try the same.

I have no idea if this is trivial change to make or not.  If issues like this only affect some random flight controller and it seems possible its a design issue, is any effort put into getting things working?  -- Or should I write off this FC as 'unlikely rpm filter is ever going to work on it in betaflight.'

Afteraffekt, 2021-05-11:
The issue still exists, and still needs resolved...

abwyatt, 2021-05-24:
I do hope there is interest to fix this ESC RPM Filtering related problem on the iFlight H743 Beast AIO. I for one have just come across the issue. At least one popular new quad, the Ummagawd 2Fiddy build is using the this FC and there are a host of owners trying to find a solution (https://www.reddit.com/r/fpv/comments/ldl0j2/iflight_beast_h7_and_rpm_filtering_issue/).

I've stumbled on an apparent workaround which has worked for other problem ESCs and which I confirm seems to work on the iFlight H743 Beast, so I thought I would contribute it here to see if it holds any clues for devs. See (https://www.reddit.com/r/TinyWhoop/comments/m16u5i/unable_to_get_rpm_filtering_working/)

I am on latest BF 4.3 development build. I've used BLHeli_M Configurator to flash the ESCs with latest 16.9 JazzMaverick Official firmware. Set PWM freq option to 48 KHz. Turned on RPM Filtering in BF 10.8 Configurator. Motor 3 shows 100% Errors and doesn't spin.

After 'set dshot_bitbang = OFF' in the CLI, Motor 3 still shows 100% error, but the motor is spinning. No gravelly sounds as others have reported. I've test flown the quad in my garage as is and it seems fine. The motors sound smooth and motors are cool on stock filters. 

Hope this provides a clue that a dev could look into.

abwyatt, 2021-05-25:
I am wondering why this would be a hardware issue with the iFlight H743 Beast, if it is indeed that. The FC was working fine on BF 4.2.6 with Bidirectional Dshot. It broke with the next release to 4.2.7/4.2.8 where I notice new features were being added to the target (IFLIGHT_H743_AIO). So that suggests to me that this is not hardware related, but firmware related.

Interestingly, I just tried BF 4.2.8 and found that the Motor 3 100% error is showing, and yet the motor spins up and seems to be operating fine in a test flight. Digging in the CLI, I note that there is no Dshot_Bitbang option in 4.2.8. So the intro of Dshot_Bitbang option in BF 4.3.0 appears to be the cause of the Motor 3 not spinning up.

So what's the practical consequence of just turning off Dshot_Bitbang in the target? 

And is the apparent Motor 3 error indicative of a false report, or is there a real effect on the operation of the RPM Filter?

abwyatt, 2021-05-25:
Here's the blackbox explorer analysis showing that the RPM filter seems to be working fine. Sorry, I don't have comprehensive data to analyse. I said I was testing in my garage ... can only get up to 40% throttle and I'm not about to do flips and rolls in my tiny garage with a 5 inch quad :D  It's raining non-stop here at moment and I live in a location far from my flying fields. Hope someone else can replicate and have a peek into the blackbox.

[image]

abwyatt, 2021-05-27:
> I am wondering why this would be a hardware issue with the iFlight H743 Beast, if it is indeed that. The FC was working fine on BF 4.2.6 with Bidirectional Dshot. It broke with the next release to 4.2.7/4.2.8 where I notice new features were being added to the target (IFLIGHT_H743_AIO). So that suggests to me that this is not hardware related, but firmware related.
> 
> Interestingly, I just tried BF 4.2.8 and found that the Motor 3 100% error is showing, and yet the motor spins up and seems to be operating fine in a test flight. Digging in the CLI, I note that there is no Dshot_Bitbang option in 4.2.8, or maybe it was not accessible. In any case, the Dshot_Bitbang auto default in BF 4.3.0 appears to be the cause of the Motor 3 not spinning up.
> 
> So what's the practical consequence of just turning off Dshot_Bitbang in the target?
> 
> And is the apparent Motor 3 error indicative of a false report, or is there a real effect on the operation of the RPM Filter?

OK, I think we can close this, at least for the iFlight Beast H743. I was put in touch with Patrick at iFlight and he tells me the problem is hardware and there is no fix. So that's it! So we shouldn't anymore time on this.

closed 2021-06-27 by haslinghuis

dukevah, 2022-01-13:
I have installed Betaflight 4.3 RC2 and it didn't allow me to spin two motors. I set `set dshot_bitbang = OFF` and now I can spin all motors. The only issue I am facing now is that the Betaflight configurator doesn't report the error rate. I have not flown the quad yet.