betaflight/betaflight #10620: Show consumed Watthours in OSD

BlackPixxel, 2021-03-13:
### Is your feature request related to a problem? Please describe.

For more endurance-oriented quads I would like to be able to estimate the remaining flight time by looking at the current flight time and the consumed energy. Using the consumed amphours is inaccurate, because the constantly reducing voltage means that later in the flight the current consumption will go up. Especially with Li-Ion packs, as their voltage decreases alot throughout the flight.

### Describe the solution you'd like

Similar to how the used amphours can be shown in the OSD, there should be another OSD element to show the watthours.
Compared to amphours the watthours will grow with a constant rate under constant load, allowing for very accurate flight time estimation.
When for example 1/4th of the usable watthours of the battery are consumed, the user can look at the flight time in the OSD and multiply it by 4. This gives him a very accurate estimate of the remaining flight time, much more accurate than doing the same with the amphours.
### Describe alternatives you've considered

There is already the flight time estimation OSD element. But when using batteries of different capacities it will not work correctly unless each time the capacity setting in Betaflight is changed.

mikeller, 2021-03-15:
@etracer65: This looks like a candidate for a variant to amperehours OSD element.

etracer65, 2021-03-15:
I don't think this is a good candidate for the OSD element variants. They are meant to be used to display the **same information** in varying formats. This request requires a different set of calculations and accumulations (with changes outside the OSD to accomplish this) resulting in different information than mAh consumed. Further I don't think an argument can be made that this request and mAh consumed are mutually exclusive and the user might not want to display both.

mikeller, 2021-03-15:
@etracer65: Makes sense - let's get this added as an extra element.

KarateBrot, 2021-04-03:
Ok, so I did the math _(unit of time in seconds)_:
```
dE / dt = P(t)
dE = P(t)·dt
dE = V(t)·A(t)·dt
E - E_prev = V·A·(t - t_prev)

⇒ E = E_prev + V·A·(t - t_prev)
```

Which yields:
```C
wattHours += battVoltage * battCurrent * deltaTime / 3600.0f;
```
Or using the existing variable for `milliAmpHours`:
```C
wattHours += battVoltage * milliAmpHours / 3.6f;
```

jonmahoney15, 2021-10-20:
yeah sure I'll wait to PR until thats merged to avoid any merge conflicts. I do see most pilots just using it as a post flight stat, but like mentioned in the feature request if someone is using it for a long range flight and they just want a different way to judge battery consumption I don't see why not.

KarateBrot, 2022-09-24:
Done in https://github.com/betaflight/betaflight/pull/11843

KarateBrot, 2022-09-24:
Completed in https://github.com/betaflight/betaflight/pull/11843

closed 2022-09-24 by KarateBrot