betaflight/betaflight #10621: What is currently going on with this project?

TheIsotopes, 2021-03-13:
### Is your feature request related to a problem? Please describe.

not really?! 
### Describe the solution you'd like

is below 
### Describe alternatives you've considered

For more than two weeks things have not seemed to go quite smoothly here (the same applies to the configurator).
As a team, you should consider that the merging of pull requests should not only be allowed by one person aka @mikeller .
This would keep the project moving forward without any major delays. 

This is just a suggestion. Thx @all

haslinghuis, 2021-03-13:
Yes, I already complained a month ago about this, as everything is voluntary, we need more members. @mikeller is our BDFL but should allow more sub servants

TheIsotopes, 2021-04-07:
Oh guys, is this going to start again?

haslinghuis, 2021-04-09:
It's just the weather and people need to relax and recharge. Everything will work out. No worries.

TheIsotopes, 2021-04-09:
@haslinghuis As I said before, there is a second man missing with the same rights as @mikeller .

haslinghuis, 2021-06-08:
@TheIsotopes you have been heard. Discussion has been going. I like to introduce mandatory testing for approving configurator PR's, can I ask you for your contribution on testing here?

TheIsotopes, 2021-06-18:
@haslinghuis only if it's still worth.
imho it seems to be coming to an end with betaflight.
I will watch this for a while and come back to it later. 

reopened for further discussion

KarateBrot, 2021-06-18:
@TheIsotopes If no one wants to test because of this mindset then it's a self-fulfilling prophecy. But it's definitely not coming to and end. Working hard on BF 4.3.

TheIsotopes, 2021-06-18:
@KarateBrot I'm still testing, you don't need to be afraid.
But if everything only comes up in bits and pieces and at fairly long intervals (several days or weeks), you start thinking about the project.

KarateBrot, 2021-06-18:
Fair enough. I'm sure it gets better again. Our gatekeeper is just very busy at the moment :)

haslinghuis, 2021-06-24:
Give it a month. As we are approaching a new release we already need to think about creating a public release.

TheIsotopes, 2021-07-13:
Nothing happened again for almost three weeks. who or what is this time to blame?

daleckystepan, 2021-07-13:
@TheIsotopes what would like to see here? New BF release ASAP?

I am pretty sure that guys are doing great job. I and definitely a lot of others spent a few hours a day by testing things, writing PR, documentation and review. We are also held a discussion with vendors. No worries about progress maybe it is just not visible outside.

You can join us a help us with it. Sure that it can be better but it's good anyway.

closed 2021-07-25 by TheIsotopes