betaflight/betaflight #10625: BMP390 no communication

sedlacekdavid, 2021-03-19:
### Describe the bug
It's not really a bug since only BMP388 is supported, but I'd like know your opinion. 
I have a BMP390 sensor and it just doesn't work.

### To Reproduce

connect an adafruit BMP390 board https://www.adafruit.com/product/4816 to the default (verified) i2c bus

### Expected behavior

The sensor should be detected

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.8 Feb 15 2021 / 12:10:35 (101738d8e) MSP API: 1.43
# config: manufacturer_id: TMTR, board_name: TMOTORF7, version: e02dd6f2, date: 2020-11-04T11:31:26Z

# start the command batch
batch start

board_name TMOTORF7
manufacturer_id TMTR

# resources
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 C06
resource MOTOR 4 C07

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS
feature LED_STRIP
feature ESC_SENSOR

# beeper
beeper -SYSTEM_INIT
beeper -ON_USB

# serial
serial 1 1 115200 57600 0 115200
serial 2 2 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200
serial 4 64 115200 57600 0 115200

# aux
aux 0 0 0 1350 1625 0 0
aux 1 1 0 1875 2100 0 0
aux 2 13 1 1300 1700 0 0
aux 3 35 3 1300 1700 0 0

# adjrange
adjrange 0 0 2 900 2100 29 2 0 0

# master
set acc_calibration = 60,0,-13,1
set mag_calibration = 423,-1,617
set baro_bustype = I2C
set baro_spi_device = 0
set baro_i2c_address = 119
set baro_hardware = BMP388
set serialrx_provider = SBUS
set dshot_idle_value = 250
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_procedure = GPS-RESCUE
set current_meter = ESC
set battery_meter = ESC
set gps_sbas_mode = AUTO
set gps_auto_baud = ON
set osd_warn_rssi = ON
set osd_warn_link_quality = ON
set osd_vbat_pos = 459
set osd_remaining_time_estimate_pos = 14570
set osd_current_pos = 2061
set osd_gps_speed_pos = 2055
set osd_gps_sats_pos = 6170
set osd_home_dir_pos = 2549
set osd_home_dist_pos = 2070
set osd_compass_bar_pos = 81
set osd_altitude_pos = 449
set osd_warnings_pos = 14761
set osd_disarmed_pos = 14570
set osd_esc_tmp_pos = 333
set osd_core_temp_pos = 135
set osd_stat_max_alt = ON
set osd_stat_total_time = ON
set osd_stat_total_dist = ON

profile 0

rateprofile 0

# end the command batch
batch end

```
```
PASTE THE OUTPUT OF 'resource show all' HERE

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: LED_STRIP
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: LED 1
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: FREE
B09: CAMERA_CONTROL
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: MOTOR 3
C07: MOTOR 4
C08: FLASH_CS
C09: FREE
C10: SERIAL_TX 3
C11: SERIAL_RX 3
C12: FREE
C13: BEEPER
C14: PINIO 1
C15: FREE
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: LED_STRIP
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH2: DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11:
    CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 3
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: LED_STRIP
DMA2 Stream 7: FREE

```

### Setup / Versions
 
 - Flight controller:  T-motor F7 (F722)
 
FSC T-motor F55A
 - Other components:
 
 - How are the different components wired up:
The sensor is connected to standard SDA/SCL lines together with a magnetometer which works.

daleckystepan, 2021-03-19:
Try this
```
set baro_i2c_device = 2
```

sedlacekdavid, 2021-03-19:
Doesn't work

sedlacekdavid, 2021-03-19:
Not sure if it will help but in BF configurator it says "I2C error: 5" at the very bottom status bar.

sedlacekdavid, 2021-03-19:
It's really fun I just tried a BMP280 I found at my workshop and it doesn't work as well. 

And you are right about that device=2 because the magnetometer works with device=2 without any issues and they both share the same bus.

etracer65, 2021-03-19:
Betaflight doesn't support the BMP390.

etracer65, 2021-03-19:
The reason that you can't get the BMP280 to work is because your config is incorrect. Doing the following in the CLI should allow the BMP280 (or BMP388) to work.
```
set baro_bustype = I2C
set baro_i2c_device = 2
set baro_i2c_address = 0
baro_hardware = AUTO
save
```
However connecting to the same I2C bus where you already have a magnetometer connected could be problematic depending on the bus termination (pull-ups). So if it still doesn't work or has errors then try disconnecting the magnetometer for testing.

The BMP390 will not work as Betaflight doesn't have a driver for that (new) sensor.

sedlacekdavid, 2021-03-19:
Thank you very much for clarification - it thought BMP390 is nothing more than more accurate BMP388 and it is 100 % compatible, but obviously it's not. So I'll order BMP388.

Regarding my second try with a BMP280 --- it's actually a BME280 not BMP280. 

Have a nice weekend guys!

closed 2021-03-19 by sedlacekdavid

etracer65, 2021-03-19:
BME280 is supported in Betaflight starting with 4.3. So you can try a nightly development build if you want (at your own risk).

sedlacekdavid, 2021-03-27:
@etracer65 hi, is it? 

# get baro_hardware
baro_hardware = AUTO
Allowed values: AUTO, NONE, BMP085, MS5611, BMP280, LPS, QMP6988, BMP388, DPS310

sedlacekdavid, 2021-03-27:
It works indeed! :) Very good job guys.

[image]