betaflight/betaflight #10635: Betaflight for ESP32

tausif-sama, 2021-03-24:
### Is your feature request related to a problem? Please describe.

### Describe the solution you'd like

I was trying to build a flight controller with the ESP32 using MPU6050. But I just couldn't find suitable flight control software available for this board. Has betaflight been successfully ported for ESP32? Where can I get it?
### Describe alternatives you've considered

I've considered using MultiWii and hackflight and honestly had no luck so far.

mikeller, 2021-03-24:
@tausif-sama: No, Betaflight has not been ported to ESP32, and there are (to my knowledge) no efforts being made to do so.
In general, I am not sure if the I/O capabilities of the ESP32 would be sufficient to meet the requirements of a flight controller (multiple high-throughput SPI peripherals, multiple fast I/O channels for motor control, multiple UARTS).
You might be better off building a hybrid system consisting of an STM32 based flight controller running Betaflight and an ESP32 running custom firmware and controlling the flight controller over an UART with MSP or an RC protocol.

ArshaanTech, 2023-11-19:
### Is your feature request related to a problem? Please describe.
Hello, I just came across this and I wanted to say that:
The ESP32 has four SPI peripheral devices, called SPI0, SPI1, HSPI and VSPI.
The ESP32 chip has 3 UART controllers.
ESP32 has 2 I2C controllers.
### Describe the solution you'd like
I actually think it would be nice to try and make a betaflight firmware for ESP32 boards. 
Sensors that can be used:
Gyro - MPU6050 i2c OR MPU6500 SPI etc
Baro - BMP280, i2c
magnetometer - HMC5883 i2c, QMC5883 i2c
GPS - neo 6m UART
Receiver - PPM / SBUS or CRSF UART

### Describe alternatives you've considered
I have seen a firmware for betaflight on ESP32, but it needs some improvements which I think should be made.

playztag, 2024-05-12:
Would anyone be interested in porting betaflight on the esp32 with me with the assistance of LLM?

ArshaanTech, 2024-05-13:
I'm not really a great programmer but I would like to help.

playztag, 2024-05-13:
Any experience with LLM generating code?

[image: This is Quan Gan's card. Their email is ***@***.*** Their phone
number is +1 661 755 0275.] <https://hihello.me/hi/ztag>

JonRobo, 2024-09-17:
The ESP32-P4 is a new variant that may be excellent for a FC.
55 GPIO
Up to 400MHz clock speed
4 x SPI
5 x UART
2 x I2C / 1 x I3C
2 × Motor Control PWM (MCPWM), up to 6
channels
Image and voice processing interfaces

and much more.

playztag, 2024-09-17:
> @ github. com>

Anyone interested in porting?

This is Quan Gan's card. Their email is ***@***.*** Their phone number is +1 661 755 0275. ( https://hihello.me/hi/ztag )

masterhi76, 2025-01-25:
Уважаемые! 
Данный проект [https://github.com/rtlopez/esp-fc](url) был реализован мной в железе. SPI - MPU6500; I2C - BMP180 . Все прекрасно работает, уровень держит, полет плавный, однако не хватает функционала  OSD (telemetry), а так же хотелось бы увидеть GPS.
Планируется ли портация Betafly на указанную платформу официально.
Если нет, готов предложить свое участие в плане сборки прошивки на платформу  ESP32.
P.S. Если есть какие то рекомендации или проекты по сборке собственной прошивки на ESP32, прошу по возможности поделиться, буду благодарен.
С Уважением!

closed 2026-09-26 by blckmn