betaflight/betaflight #10637: Betaflight 4.28 and Matek f722 se - barometer not recognized

ducatidaddy, 2021-03-24:
### Describe the bug
 Tabs in configurator are on. Barometer is BMP280.

### To Reproduce
 when plugged into usb, barometer not highlighted. other sensors are: gps, mag

### Expected behavior
 want barometer altitude reading, not gps

### Flight controller configuration

```
PASTE THE OUTPUT OF 'diff' HERE
```
```
PASTE THE OUTPUT OF 'resource show all' HERE
```

### Setup / Versions
 
 - Flight controller: 
 
 - Other components:
 
 - How are the different components wired up:

etracer65, 2021-03-24:
Barometer is actually probably the DPS310 and isn't supported until Betaflight 4.3. Contact the manufacturer to find out which version you have. I believe they also have published interim firmware versions that add preliminary support for the DPS310.

dkustec, 2021-03-24:
@ducatidaddy , New batch of that FC contains a DPS310 baro. This baro is not supported currently (will be in 4.3). Until then, you can download firmware from this link [betaflight_4.2.8_MATEKF722SE.hex](http://www.mateksys.com/downloads/FW/betaflight_4.2.8_MATEKF722SE.hex).

ducatidaddy, 2021-03-24:
Thanks. I will try this hex. Based on Banggood's specs for this board, it claims it is the BMP280.

etracer65, 2021-03-24:
Don't ever believe anything written on a Banggood product page. Go to the actual Matek product page. That's assuming you didn't get a counterfeit board with unknown components.

dkustec, 2021-03-24:
First batches of that board (Matek F722-SE) really had the BMP280 installed onboard. Later, Mateksys decided to switch on the DSP310. The footprint and pinouts are the same as BMP280 so no PCB redesign taken in its production. The only difference between "old" and "new" boards is a baro and a battery pads style.
Nevertheless, Banggood has an old description of those first batches, indeed.

ducatidaddy, 2021-03-25:
Reflashed the board with the hex above. Barometer recognized! Please consider this issue closed. Thanks for everyone's help. Will send a review and comment on Banggood's site.

closed 2021-03-25 by etracer65