betaflight/betaflight #10641: Mixer mode issue

haslinghuis, 2021-03-26:
### Describe the bug
Found the following issue while working on https://github.com/betaflight/betaflight-configurator/pull/2436:

1. some mixer modes cannot be set using CLI or GUI
2. some mixer modes have no outputs in firmware before defining appropriate `mmix` settings. (Only STM32F411 needs `mmix` settings defined before, STM32F7X2 is more relax and does not need a `mmix` defined.)

Tested with BFC 10.7/8 and FW 4.04 - 4.3

### To Reproduce
When selecting a different models these are the problematic choices:

1. `Heli90`, `Custom Airplane` and `Custom Tricopter` returns zero output from firmware
2. `Heli90` and `PPM to servo` changes into `Custom Airplane`
3. `Singlecopter` changes into `Custom`

### Expected behavior
Obvious these are not much used models in Betaflight and could be overlooked over time. Did have a look at the mixer code in firmware but could not find a reason for this behavior. 

### Flight controller configuration
```
# version
# Betaflight / STM32F7X2 (S7X2) 4.3.0 Mar 24 2021 / 19:47:01 (norevision) MSP API: 1.44
# config: manufacturer_id: UNKN, board_name: MTKS-MATEKF722MINI.c, version: unknown, date: 2021-02-20T23:43:05.

# start the command batch
batch start

board_name MATEKF722MINI
manufacturer_id MTKS

profile 0

rateprofile 0

# end the command batch
batch end
```

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: GYRO_CS 2
D00: FREE
D01: FREE
D02: FLASH_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1: MOTOR 1
    CH2: MOTOR 2
    CH3: MOTOR 3
    CH4: MOTOR 4
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```
### Setup / Versions
 - Flight controller: 
 Matek STM32F722
 Matek STM32F411

etracer65, 2021-03-26:
Seems to be working as designed. The mixers you identify as problematic are defined to have at least one motor, but don't have a default motor mixing defined. So the config check code resets them to "CUSTOM".

https://github.com/betaflight/betaflight/blob/dc76125e699ac7b296d971752e187bc9dabde865/src/main/config/config.c#L210-L225

The linkage from the mixer mode to the motor mixes is here:

https://github.com/betaflight/betaflight/blob/dc76125e699ac7b296d971752e187bc9dabde865/src/main/flight/mixer_init.c#L219-L248

haslinghuis, 2021-03-26:
Yeah but even after applying example 3 in github.com/betaflight/blob/master/docs/Mixer.md#example-3-custom-tricopter it will change to the modes mentioned.

etracer65, 2021-03-26:
Seems fine to me.
```
# mixer
mixer CUSTOMTRI
mmix 0  1.000  0.000  1.333  0.000
mmix 1  1.000 -1.000 -0.667  0.000
mmix 2  1.000  1.000 -0.667  0.000

# servo
servo 5 1000 2000 1500 100 -1
smix reverse 5 2 r

# servo mixer
smix 0 5 2 100 0 0 100 0
```

And the `CUSTOMTRI` mixer is excluded from this "reset to CUSTOM" logic.

haslinghuis, 2021-03-26:
No problem for me but with the mmix setting for tricopter in place I still observe 
- Heli90 and PPM to servo changes into Custom Airplane
- Singlecopter changes into Custom

etracer65, 2021-03-26:
Yes, look at the config check code. The logic has nothing to do with whether you have defined `mmix` entries. It only checks whether there is or isn't a default motor mix. So `SINGLECOPTER` has 1 motor and no motor mix:
```
     { 1, true,  NULL },                // MIXER_SINGLECOPTER 
```
So this code resets it to `CUSTOM`:
```
         if (mixers[mixerMode].motorCount && mixers[mixerMode].motor == NULL) 
             mixerConfigMutable()->mixerMode = MIXER_CUSTOM; 
```

For the others they have servos but no default servo mix so they run into this check
```
 #ifdef USE_SERVOS 
         if (mixers[mixerMode].useServo && servoMixers[mixerMode].servoRuleCount == 0) 
             mixerConfigMutable()->mixerMode = MIXER_CUSTOM_AIRPLANE; 
 #endif 
```
`SINGLECOPTER` does have a default servo mix so it doesn't get caught in the second condition. Again, the checks are based on the defaults - not what the user has configured.

https://github.com/betaflight/betaflight/blob/dc76125e699ac7b296d971752e187bc9dabde865/src/main/flight/servos.c#L170-L198

I'm not saying the code couldn't be improved, but I don't know at this point. But it does seem to be working as written. It's possible that the code is so old that it doesn't really properly apply anymore but I haven't taken the time to analyze what should happen.

haslinghuis, 2021-03-26:
Thanks for all input. Resetting to another mode should not happen (user perspective), but I have a better understanding how it works now. As there are no pressing issues I'm happy to close this.

closed 2021-03-26 by haslinghuis