betaflight/betaflight #10642: Use Fport on F411 with FrSky Archer RS

bdn76, 2021-03-26:
### Describe the bug
Unable to use fport with FrSky Archer RS on F411

### To Reproduce
Betaflight 4.2.8. FC HGLRC Zeus 5 AIO. Receiver FrSky Archer RS. Connect ~sport of Archer RS to T2 on F411 FC (according to https://github.com/betaflight/betaflight/wiki/The-FrSky-FPort-Protocol). Place mark on Fport in RS option via transmitter.

### Expected behavior
Work Receiver->FC stream and FC->Receiver telemetry.

### Flight controller configuration
```
# diff

# version
# Betaflight / STM32F411 (S411) 4.2.8 Feb 15 2021 / 12:09:04 (101738d8e) MSP API: 1.43
# config: manufacturer_id: HGLR, board_name: HGLRCF411, version: 18c5b9b5, date: 2020-05-25T12:32:42Z

# start the command batch
batch start

board_name HGLRCF411
manufacturer_id HGLR

# resources
resource MOTOR 5 NONE
resource SERIAL_TX 11 B03
resource PINIO 1 B10

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature SOFTSERIAL
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 1 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 30 8192 115200 57600 0 115200

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 1 1800 2100 0 0
aux 2 2 1 900 1200 0 0
aux 3 13 2 1800 2100 0 0

# master
set acc_calibration = 3,-60,165,1
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = FPORT
set dshot_idle_value = 450
set motor_pwm_protocol = DSHOT600
set yaw_motors_reversed = ON
set pinio_config = 129,1,1,1
set pinio_box = 0,255,255,255
set gyro_1_align_yaw = 1800

profile 0

rateprofile 0

# end the command batch
batch end

# 
```

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: PREINIT 2
B00: ADC_BATT
B01: ADC_CURR
B02: BEEPER
B03: SERIAL_TX 11
B04: MOTOR 1
B05: MOTOR 2
B06: MOTOR 3
B07: MOTOR 4
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: PINIO 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2: SERIAL_TX 11
TIM3:
    CH1: MOTOR 1
    CH2: MOTOR 2
TIM4:
    CH1: MOTOR 3
    CH2: MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: MOTOR 4
DMA1 Stream 4: MOTOR 1
DMA1 Stream 5: MOTOR 2
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Setup / Versions
 - Flight controller: HGLRC Zeus 5 AIO (HGLRCF411), BTFL 4.2.8
 - Other components: FrSky Archer RS
 - How are the different components wired up: FC T2 <-> receiver ~sport

I'm connect oscilloscop to receiver and see inverted signal on ~sport pin but i'm not see any response from FC nor on T2 nor on R2 port. Try to change sport to ~sport, R2 to T2 without success and do not

etracer65, 2021-03-26:
With a F4 flight controller you must either supply an **uninverted signal, or must have a bidirectional inverter on the flight controller to uninverted the signal**. The F4 processor cannot handle an inverted signal on its own like a F7 can. That's why the page you referenced refers to a "hacked" receiver with an uninverted mod. If you cannot get an uninverted signal from that receiver then you'll be unable to get Fport to work with that flight controller.

closed 2021-03-26 by etracer65

bdn76, 2021-03-26:
I'm use ~sport on Archer RS with **uninverted signal** (i'm check it via oscilloscop).

etracer65, 2021-03-26:
> I'm use ~sport on Archer RS with **uninverted signal** (i'm check it via oscilloscop).

Then follow the instructions on the page in the wiki that you already linked to.

> F4 with 'uninverted SmartPort' hacked receiver: 
> 
> ```
> set serialrx_provider = FPORT
> set serialrx_halfduplex = ON
> set serialrx_inverted = OFF
> ```
>

This is not the place to get help with hardware of configuration. See https://github.com/betaflight/betaflight#support-and-developers-channel

bdn76, 2021-03-26:
Of course i'm setting up params from doc:

```
$ grep serialrx BTFL_cli_20210325_205225.txt
set serialrx_provider = FPORT
set serialrx_inverted = OFF
set serialrx_halfduplex = ON
```

without success. I'm try all combination inverted/halfduplex without success and i'm check output from FC TX pin via oscilloscop - empty. Is it empty output from TX normal with enabled fport and telemetry?

mikeller, 2021-03-28:
@bdn76:

> see inverted signal on ~sport pin 

If this is the case (with 'inverted signal' being idle == low) then there is no uninverted output from this RX, and you will need an external inverter.
However, if the signal you see is idle == high then you should be able to connect this to the TX pin on the flight controller, and use the following settings:

```
set serialrx_provider = FPORT
set serialrx_inverted = OFF
set serialrx_halfduplex = ON
```

bdn76, 2021-03-28:
Problem resolved via full powercycle of receiver + FC and transmitter. Soldering/options have been correct.

Correct recipe: solder HGLRC ZEUS5 AIO T2 <-> FrSky Archer RS inverted s.port, remove jumper between RC pin and SBUS pin on FC, remove jumper between RC pin and R2 pin on FC, **mark Fport** in transmitter RS menu, **unmark Fport2** in transmitter RS menu, setting up betaflight

```
set serialrx_provider = FPORT
set serialrx_inverted = OFF
set serialrx_halfduplex = ON
```

**powercycle** (bat and usb) FC+receiver, **powercycle** transmitter.

haslinghuis, 2021-03-28:
Thanks for reporting back and great you were able to get it working. Sometimes a configuration needs a cold boot.