betaflight/betaflight #10643: Fly to Moon issue with MATEKF411& BF 4.2.8

nestlef1, 2021-03-26:
Hi, 

I recently updated my 3S toothpick from 3.5.5 to 4.2.8 to leverage RPM filter.
(The 4 in 1 ESC (BLHeli_S) has version G_H_30, from JazzMaverick’s firmwares it has been upgraded to 16.73 to support RPM filtering)

3.5.5 flew o.k. !

Now on 4.2.8 when I hit arm, and sligtly (1mm) raise the throttle the quad launches itself with full throttle to the sky.

diff on 4.2.8:
 ``` 
# diff

# version
# Betaflight / STM32F411 (S411) 4.2.8 Feb 15 2021 / 12:09:04 (101738d8e) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF411, version: be0c9205, date: 2020-01-15T19:44:32Z

# start the command batch
batch start

board_name MATEKF411
manufacturer_id MTKS

# map
map TAER1234

# serial
serial 1 2048 115200 57600 0 115200

# aux
aux 0 0 7 1625 2100 0 0
aux 1 1 1 1300 1625 0 0
aux 2 2 1 1800 2100 0 0

# master
set acc_calibration = -97,35,-133,1
set rssi_channel = 16
set rc_smoothing_auto_smoothness = 7
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set bat_capacity = 450
set current_meter = VIRTUAL
set ibatv_scale = 350
set ibatv_offset = 40
set osd_rssi_alarm = 40
set osd_cap_alarm = 450
set osd_rssi_pos = 2105
set osd_tim_1_pos = 2551
set osd_tim_2_pos = 2519
set osd_flymode_pos = 2082
set osd_current_pos = 2541
set osd_mah_drawn_pos = 2498
set osd_warnings_pos = 2378
set gyro_1_align_yaw = 1800
set gyro_rpm_notch_q = 700

profile 0

# profile 0
set dyn_lpf_dterm_curve_expo = 7
set iterm_relax_cutoff = 20
set yaw_lowpass_hz = 100
set throttle_boost = 7
set throttle_boost_cutoff = 25
set ff_spike_limit = 70
set ff_smooth_factor = 20

rateprofile 0

[... 786 lines cut ...]
set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 114
set pitch_rc_rate = 113
set yaw_rc_rate = 204
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 0
set roll_srate = 76
set pitch_srate = 70
set yaw_srate = 38
set tpa_rate = 10
set tpa_breakpoint = 1650
set throttle_limit_type = OFF
set throttle_limit_percent = 100

# 
``` 

I have no blackbox recording possibility :(

Any clue?
Thanks,
Peter

etracer65, 2021-03-26:
https://drones.stackexchange.com/questions/121/freshly-built-quadcopter-drone-flips-on-takeoff

And don't say you checked these. Really check them - particularly the motor directions.

This is not the correct place to seek help for hardware or configuration issues. See https://github.com/betaflight/betaflight#support-and-developers-channel

closed 2021-03-26 by etracer65

nestlef1, 2021-03-26:
I did not say the drone flips, I'm saying it lifts up to the sky.
I checked the motor directions, one by one, obviously it is okay, as I usually disarm around 20meters high.
It couldn't get that high if it is propeller or motor direction issue.

Gyro: Also, how the 3D model moves that perfectly matches the reality.

I also said that on previous BF release the same hardware is ok, so I do not think it is a hardware issue.
As you can see the default configuration is only slightly changed, I do not thing this has anything to configuration.

ctzsnooze, 2021-03-28:
Can you make a log of the event?

ctzsnooze, 2021-03-28:
I don't see anything wrong with the diff all in 4.x so I can't really help.  A log would make the fixing easy.
Have you taken props off and confirmed that the motors spin smoothly and accurately and in the correct direction via the motors tab?

mikeller, 2021-03-28:
@nestlef1:

> The 4 in 1 ESC (BLHeli_S) has version G_H_30, from JazzMaverick’s firmwares it has been upgraded to 16.73 to support RPM filtering)

JazzMaverick firmware is known to be unstable and exist in countless variations with experimental features and bugs, all released under the same 'version number'. Most likely what you have installed isn't working properly. Try an alternative firmware like BlueJay (https://github.com/mathiasvr/bluejay) or JESC instead.