betaflight/betaflight #10647: on latest builds the rx is not recognized

TheIsotopes, 2021-03-28:
### Describe the bug

since `build #2426` the rx is no more recognized 
@SteveCEvans pls check #10609
### To Reproduce

using `build #2426` or newer 

### Expected behavior

a working rx and telemetry

### Flight controller configuration

```
# Betaflight / STM32F405 (S405) 4.3.0 Mar 28 2021 / 03:52:17 (2278ffd1d) MSP API: 1.44
# config: manufacturer_id: MTKS, board_name: MATEKF405MINI, version: f173f9bc, date: 2019-10-02T14:43:13Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF405MINI
manufacturer_id MTKS
mcu_id 0028001d3436510b33323835
signature 

# name: EGODRIFT

# feature
feature -SOFTSERIAL
feature -TELEMETRY
feature -OSD
feature LED_STRIP

# serial
serial 1 0 115200 57600 0 115200
serial 3 64 115200 57600 0 115200

# beeper
beeper -BAT_CRIT_LOW
beeper -BAT_LOW

# led
led 0 6,5::AW:0
led 1 7,5::AW:0
led 2 8,5::AW:0
led 3 9,5::AW:0
led 4 9,2::AW:0
led 5 8,2::AW:0
led 6 7,2::AW:0
led 7 6,2::AW:0

# mode_color
mode_color 6 1 3

# aux
aux 0 0 1 1900 2100 0 0
aux 1 13 2 900 1100 0 0
aux 2 26 2 1900 2100 0 0
aux 3 36 0 1900 2100 0 0

# master
set gyro_lowpass2_hz = 0
set acc_hardware = NONE
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set bat_capacity = 1550
set use_cbat_alerts = ON
set cbat_alert_percent = 20
set vbat_duration_for_warning = 50
set vbat_duration_for_critical = 20
set pid_process_denom = 2
set telemetry_disabled_mode = ON
set telemetry_disabled_acc_x = ON
set telemetry_disabled_acc_y = ON
set telemetry_disabled_acc_z = ON
set telemetry_disabled_pitch = ON
set telemetry_disabled_roll = ON
set telemetry_disabled_heading = ON
set telemetry_disabled_altitude = ON
set telemetry_disabled_vario = ON
set telemetry_disabled_lat_long = ON
set telemetry_disabled_ground_speed = ON
set telemetry_disabled_distance = ON
set name = EGODRIFT

profile 0

# profile 0
set profile_name = 3S_Bat
set p_pitch = 40
set i_pitch = 79
set d_pitch = 33
set f_pitch = 83
set p_roll = 33
set i_roll = 67
set d_roll = 27
set f_roll = 71
set d_min_roll = 18
set d_min_pitch = 22
set auto_profile_cell_count = 3
set simplified_pids_mode = RP
set simplified_master_multiplier = 80
set simplified_roll_pitch_ratio = 110

profile 1

# profile 1
set profile_name = 4S_Bat
set p_pitch = 35
set i_pitch = 69
set d_pitch = 29
set f_pitch = 73
set p_roll = 29
set i_roll = 59
set d_roll = 24
set f_roll = 63
set d_min_roll = 16
set d_min_pitch = 19
set auto_profile_cell_count = 4
set simplified_pids_mode = RP
set simplified_master_multiplier = 70
set simplified_roll_pitch_ratio = 110

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set rates_type = QUICK
set roll_rc_rate = 150
set pitch_rc_rate = 150
set yaw_rc_rate = 130
set roll_srate = 126
set pitch_srate = 126
set yaw_srate = 99

# restore original rateprofile selection
rateprofile 0

# end the command batch
batch end
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: FREE
A02: FREE
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: LED 2
A15: FREE
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: LED_STRIP
B07: I2C_SDA 1
B08: FREE
B09: LED 1
B10: PREINIT 3
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FLASH_CS
C01: FREE
C02: GYRO_CS 1
C03: GYRO_EXTI
C04: ADC_CURR
C05: ADC_BATT
C06: MOTOR 1
C07: MOTOR 2
C08: MOTOR 3
C09: MOTOR 4
C10: FREE
C11: FREE
C12: FREE
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4:
    CH1: LED_STRIP
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

### Setup / Versions
 
 - Flight controller: Matek F405-Mini
 
 - Other components: FrSky R-XSR with FPort lbt firmware
 
 - How are the different components wired up:

SteveCEvans, 2021-03-28:
Tested on both FURYF4OSD and TMOTORF7 targets with Crossfire and working OK for both control and telemetry. Is this affecting control, telemetry or both on FrSky?

TheIsotopes, 2021-03-28:
no telemetry is displayed and the sliders in the receiver tab do not move when i enter data with the sticks.

haslinghuis, 2021-03-28:
Tested with Matek F722 Mini and TBS Crossfire still works. @TheIsotopes did you only test FPort or also SBUS configuration on Frsky?

[image]

TheIsotopes, 2021-03-28:
@haslinghuis only FPort testet ... for SBus i must flashing another firmware on my rx's

haslinghuis, 2021-03-28:
No problem, just trying to narrow it down further. Please, If anyone could test SBUS / S.Port with latest developement build would be appreciated.

TheIsotopes, 2021-03-28:
tested again with SPRACINGH7NANO and FrSky XM+ ... SBUS is working without problem ... just can't test telemetry, the rx doesn't have any. ;-)

mikeller, 2021-03-29:
I can reproduce this - FPort on r-XSR with BEEROTORF4 does not work with latest on `master`.

alteman, 2021-03-30:
FYI: this also affects SPI receivers (reproducible on MATEKF411RX)

SteveCEvans, 2021-03-30:
My R-XSR has arrived... I’ll look into this later this evening.

TheIsotopes, 2021-03-30:
I tested again the SPracingH7Nano with XM+.
SBUS seems to work, yes ... but I still don't like the RX task output in the task list. 

connected with the transmitter, I come to these values:
```
# tasks
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms   late    run reqd/us
00 - (         SYSTEM)     10       1       0    0.5%    0.0%         1      0     15       5
01 - (         SYSTEM)    999       2       0    0.6%    0.0%       126      0   1489       5
02 - (           GYRO)   8000      11       8    9.3%    6.9%     28021      0  11913       0
03 - (         FILTER)   8000       9       4    7.7%    3.7%     15906      0  11913       0
04 - (            PID)   8000      26      19   21.3%   15.7%     64845      0  11913       0
07 - (             RX)  62500       8       7   50.5%   44.2%      1099      0    660      12
08 - (         SERIAL)    100     978       1   10.2%    0.5%       235      1    149       6
09 - (       DISPATCH)    999       2       0    0.6%    0.0%       161      0   1489       5
10 - (BATTERY_VOLTAGE)     50       2       2    0.5%    0.5%        39      0     74       7
11 - (BATTERY_CURRENT)     50       1       1    0.5%    0.5%         8      0     74       5
12 - ( BATTERY_ALERTS)      5       1       1    0.5%    0.5%         1      0      8       5
13 - (         BEEPER)    100       2       1    0.5%    0.5%        27      0    149       5
20 - (       LEDSTRIP)    100       8       1    0.5%    0.5%        53      0    149       6
24 - (            CMS)     20       1       1    0.5%    0.5%         7      0     30       6
25 - (        VTXCTRL)      5       1       0    0.5%    0.0%         0      0      8       5
26 - (        CAMCTRL)      5       1       1    0.5%    0.5%         0      0      8       5
28 - (    ADCINTERNAL)      1       2       1    0.5%    0.5%         0      0      2       6
RX Check Function                   6       1                       193
Total (excluding SERIAL)                        95.0%   74.5%
```
without the transmitter switched on 
```
# tasks
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms   late    run reqd/us
00 - (         SYSTEM)     10       1       1    0.5%    0.5%         3      0     10       6
01 - (         SYSTEM)    999       2       0    0.6%    0.0%       273      0   1009       6
02 - (           GYRO)   8000      11       8    9.3%    6.9%     60487      0   8073       0
03 - (         FILTER)   8000       8       4    6.9%    3.7%     34266      0   8073       0
04 - (            PID)   8000      25      19   20.5%   15.7%    138824      0   8073       0
07 - (             RX)  24390       8       7   20.0%   17.5%      2476      0    132      12
08 - (         SERIAL)    100     988       2   10.3%    0.5%       283      1    101       7
09 - (       DISPATCH)   1000       2       1    0.7%    0.6%       348      0   1009       5
10 - (BATTERY_VOLTAGE)     50       2       1    0.5%    0.5%        85      0     50       7
11 - (BATTERY_CURRENT)     50       1       1    0.5%    0.5%        17      0     50       5
12 - ( BATTERY_ALERTS)      5       1       1    0.5%    0.5%         3      0      5       5
13 - (         BEEPER)    100       2       0    0.5%    0.0%        59      0    101       5
20 - (       LEDSTRIP)    100       9       2    0.5%    0.5%       114      0    101       7
24 - (            CMS)     20       1       1    0.5%    0.5%        16      0     20       6
25 - (        VTXCTRL)      5       1       0    0.5%    0.0%         2      0      5       5
26 - (        CAMCTRL)      5       0       0    0.0%    0.0%         2      0      5       5
28 - (    ADCINTERNAL)      1       1       1    0.5%    0.5%         0      0      1       6
RX Check Function                   2       0                       445
Total (excluding SERIAL)                        62.5%   47.9%
```

SteveCEvans, 2021-03-30:
The high update rate for the rx task is due to the task being called many times over to handle each state. Unfortunately the load calculating code isn't handling this too well. Needs some thought. Same issue for the baro task.

TheIsotopes, 2021-03-31:
@SteveCEvans does that mean that this is okay?
I don't think so, because i can't do a gyro calibration via stick command with it.

TheIsotopes, 2021-04-17:
fix in #10659

PR #10649 (2021-04-25): Fixed RX task check.
Fixes #10647.

We need to run the entire task check every time, to avoid losing status updates while waiting for the main task to be scheduled.
closed 2021-04-17 by TheIsotopes