betaflight/betaflight #10648: Smartaudio not working on Succex-E F4

JumaAnher, 2021-03-28:
Hi team,

Here's an issue (I think) with iFlight Succex-E F4 TX1. 
Problem Description: Smartaudio not working wiring to TX1 of the iFlight SucceX-E F4 board.
FC iFlight Succex-E F4 V2.1, target IFLIGHT_SUCCEX_E_F4 (checked several versions). I have checked that with Emuflight the smartaudio works properly in the same pad.

VTX: TBS Unify Pro V3 5V.

Here's the diff:
```# 

# diff all

# version
# Betaflight / STM32F405 (S405) 4.2.8 Feb 15 2021 / 12:08:36 (101738d8e) MSP API: 1.43
# config: manufacturer_id: IFRC, board_name: IFLIGHT_SUCCEX_E_F4, version: c7fde51e, date: 2020-08-26T04:30:34Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name IFLIGHT_SUCCEX_E_F4
manufacturer_id IFRC
mcu_id 001c00365853500620303758
signature 

# feature
feature -LED_STRIP
feature -DISPLAY
feature SOFTSERIAL
feature GPS

# serial
serial 0 2048 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 5 16384 115200 57600 0 115200

# aux
aux 0 0 0 1925 2100 0 0
aux 1 1 1 1925 2100 0 0
aux 2 46 3 1900 2100 0 0
aux 3 13 3 1400 1600 0 0
aux 4 35 2 1900 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# master
set acc_calibration = 54,23,321,1
set fpv_mix_degrees = 30
set serialrx_provider = CRSF
set blackbox_device = NONE
set ibata_offset = 0
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set debug_mode = SMARTAUDIO
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 2
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5658
set gyro_1_align_yaw = 1800

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 130
set pitch_rc_rate = 130
set yaw_rc_rate = 130
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set yaw_srate = 65

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```

Best regards,
Juan Hernandez

haslinghuis, 2021-03-28:
According to https://shop.iflight-rc.com/index.php?route=product/product&product_id=978 you should use T1 for SmartAudio and set SmartAudio as protocol in ports tab for UART1.

Is it working on a previous firmware build?

[image]

JumaAnher, 2021-03-28:
That is just what I did. I'm using the pad instead of using the connector. I checked installing Emuflight and Smartaudio works properly.

No, it does not work in previous firmware versions, at least 4.1 and 4.2 versions, I did not try lower versions.

mikeller, 2021-03-29:
@JumaAnher: Can you please try this: 
[betaflight_4.2.9_STM32F405_763badc4f.zip](https://github.com/betaflight/betaflight/files/6221311/betaflight_4.2.9_STM32F405_763badc4f.zip)

JumaAnher, 2021-03-30:
> @JumaAnher: Can you please try this:
> [betaflight_4.2.9_STM32F405_763badc4f.zip](https://github.com/betaflight/betaflight/files/6221311/betaflight_4.2.9_STM32F405_763badc4f.zip)
@mikeller I've tried the firmware you attached and the issue is still present.

Tell me if you need any other information or test.

BR

JumaAnher, 2021-04-09:
Hi,
Do you need any other test?

BR

rafareusch, 2021-05-07:
@JumaAnher Did you fixed somehow this issue?
I'm having the same issue at 4.2.9 with AKK vtx, tried using UART 6 but no luck. 

Things I tried: 
Switching from T1 to T6, and changing the corresponding port config - didn't work.
Flashed same FW again and applied custom defaults, tried only configuring SmartAudio - didn't work
With the FW flashed applied my dump - didn't work

There's no way that the pad isn't making contact, tested continuity and seems to be fine. (battery is plugged)

JumaAnher, 2021-05-07:
> @JumaAnher Did you fixed somehow this issue?
> I'm having the same issue at 4.2.9 with AKK vtx, tried using UART 6 but no luck.
>
> Things I tried:
> Switching from T1 to T6, and changing the corresponding [...]

Hi!
No way to fix it, it's a Betaflight firmware problem since it works properly with Emuflight installed.
I use Smart Audio trough my Crossfire Nano RX.

Betaflight developers, please review this issue.

Best regards.

haslinghuis, 2021-05-07:
ping @etracer65

closed 2021-06-15 by github-actions[bot]