betaflight/betaflight #10654: NOGYRO on JHEMCUF7 after FW update

AlessandroFornasier, 2021-03-31:
### Describe the bug
After flashing latest betaflight firmware on the JHEMCUF7 board the gyroscope and accelerometer are not working - NOGYRO flag after status command in CLI.

### To Reproduce
Flash latest betaflight firmware on JHEMCUF7

### Expected behavior

### Flight controller configuration

```
# version
# Betaflight / JHEF7DUAL (JH7D) 4.1.0 Oct 16 2019 / 11:39:01 (c37a7c91a) MSP API: 1.42

# start the command batch
batch start

board_name JHEF7DUAL

# feature
feature -TELEMETRY
feature MOTOR_STOP

# serial
serial 2 8192 115200 57600 0 115200
serial 3 1 115200 57600 0 115200
serial 5 1 115200 57600 0 115200

# led
led 0 6,7::C:2
led 1 7,7::C:6
led 2 8,7::C:10
led 3 9,7::C:12

# aux
aux 0 40 2 1300 2100 0 0
aux 1 41 3 1300 2100 0 0

# master
set gyro_to_use = SECOND
set motor_pwm_protocol = DSHOT600
set osd_crosshairs_pos = 2253
set osd_current_pos = 2497

profile 0

rateprofile 0

# end the command batch
batch end
```
```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: SERIAL_RX 2
A04: PREINIT 2
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: LED_STRIP
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: LED 1
B00: MOTOR 1
B01: MOTOR 2
B02: PREINIT 1
B03: MOTOR 4
B04: MOTOR 3
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: CAMERA_CONTROL
B09: PINIO 2
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_RSSI
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: FREE
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FLASH_CS
C14: PINIO 1
C15: BEEPER
D02: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: LED_STRIP
TIM2:
    CH2: MOTOR 4
TIM3:
    CH1: MOTOR 3
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM4:
    CH3: CAMERA_CONTROL
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: LED_STRIP
DMA2 Stream 7: FREE
```

### Setup / Versions
 - Flight controller: JHEMCUF7 http://www.jhemcu.com/e_productshow/?21-JHEMCU-F7BT-5V-8V-DJI-Air-Flight-Controller-Dual-Gyro-BlueTooth-21.html
 - Other components:
 - How are the different components wired up:

### Steps done
 - Original backup from CLI with diff all (gyro was working)
 - Latest version FW flash 
 - Restore original firmware version (4.1.0)
 - Restore backup from CLI
 
 Problem (NOGYRO) still present.
 
 Thanks everybody in advance for your help.

mikeller, 2021-03-31:
@AlessandroFornasier: You are using an outdated legacy version of the firmware. Please use configurator to install the latest version of the firmware available for your hardware (4.2.8 at the time of writing).

AlessandroFornasier, 2021-03-31:
@mikeller thanks for the help.
I did re-flash the latest version (target: JHEF7DUAL - 4.2.8) and applied custom defaults. However, still NOGYRO and no moving quad on the gui.

mikeller, 2021-04-01:
@AlessandroFornasier: Interesting. Can you try to re-flash the firmware that work for you to verify that this is not a case of a defective board / gyro?

AlessandroFornasier, 2021-04-01:
@mikeller yes it is an interesting situation! Actually I don’t think there is a firmware which makes the gyro work. I tried different firmwares (both the legacy versions 4.0.x and 4.1.x and the latest versions 4.2.x) but for each of these the gyro was not working.

Let me recap the history a bit to make everything more clear:

1. I got a new board JHEMCU F7 dual gyro with betaflight firmware version 4.1.0 already installed that was working.
2. I did a backup from the CLI with `diff all`
3. I installed the latest version of the betaflight firmware (4.2.8) and applied the custom defaults
4.  When I connect back the flight controller, I got the NOGYRO flag

nwpulver, 2021-04-10:
@AlessandroFornasier Could you try switching to just one or the other gyro? I think it is in configuration tab under gyro. It says something like use gyro: (drop down menu) 1 or 2. Select one or the other and see if that fixes.

AlessandroFornasier, 2021-04-10:
@nwpulver I already did it but same results. For each of the gyro I also try the command `gyroregisters` and the result is 0xFF in all the fields, which let me think that gyro are not answering.

I am kinda thinking that some component got burned and the gyros are not powered correctly. I will try to check the voltages on the pins of the gyro.

closed 2021-05-18 by github-actions[bot]

KarateBrot, 2021-05-24:
I can confirm this. Gyros both not working on current master. BF 4.2.9 works just fine with JHEF7DUAL.