betaflight/betaflight #10667: Allow different OSD position of "Land now" BatteryCritical v.s. other warnings.

MBoerlage, 2021-04-05:
### Is your feature request related to a problem? Please describe.

I am so sorry. I am killing lipos, hard, fast, hot - especially during racing practice. I know I am not the only one. The power warnings are fantastic, and I use the thresholds carefully to protect sudden death of lipo cells. The way I use them know is that "low battery" (BatteryWarning) is a soft warning - finish lap and land comfortable. And "land now" (BatteryCritical) is a hard warning - do not complete lap, just obey and land now. Passing the 3.3V threshold I set for "land now" is in many cases a sure death of a lipo. 

The issue I am facing is that during an intense race practice, I struggle to register the difference between "low battery" and "land now" warning as they are flashing in front of you. Since they are two warnings tied to the same location in OSD, I have to choose to either 1) have "low battery" flashing in front of me while its not yet a hard warning". I then have to focus on the point where the flashing "low battery" changes into "land now". 2) have warnings in the corner somewhere which could lead to me missing out a sudden death warning (land now) of my lipos. In that case its even harder to notice when the warning changes from low battery to land now.

### Describe the solution you'd like

It would greatly help if I can give the "low battery" warning a different location as the "land now" warning. I would choose to have the "low battery" warning in the corner, right where I prefer all other warnings to show up. And then have the "land now" right in the middle of my osd, so its an immediate signal to obey the warning for me.

My request would be to allow "land now" a location different from other warnings. 

A more structural change could be to create a different class of either "warnings" or "critical warnings". And then allow the location of warnings to be different from "critical warnings".  Another idea, and I am not sure if that is feasible, is to change the OSD color of the critical warning so it stands out from the standard warning. 

### Describe alternatives you've considered

I have not been able to find a way to do this properly.

granolaFPV, 2021-04-05:
I would also love a solution for this issue. I often end up landing too soon or too late when racing, because I don't want the "Low Battery" warning positioned where it's a constant distraction — but that also results in missing the "Land Now" warning since it's difficult to visually differentiate between those two warnings when my gaze is focused intently on the next gate.

MBoerlage, 2021-04-05:
So some further discussion; if you have two elements, 
- one containing warnings as today. 
- second one contains the "land now" critical warning. 
Both elements can be put wherever on the screen as user want. The question came up if the first one should also show "land now", just to make sure that the critical warning is always visible. 

Flashing:
- today the cell voltage level is flashing when low battery or land now are flashing
- once you hit the battery critical level, you only see "land now" flashing, not "low battery"
So request is to have the new critical warning "battery critical" flashing as well.

limonspb, 2021-04-05:
interesting idea.
Should be pretty easy to implement a separate "Critical Warning" osd element with its own separate position.
I guess without any changes to the current warning element.

What are the other critical warnings? Or just a separate "LAND NOW" element?
lets see maybe @mikeller  got some ideas.

MBoerlage, 2021-04-05:
from some (not code review) research (yes it was google): 

osdWarningArmingDisabled: Reports the most severe reason for not arming.
osdWarningBatteryNotFull: Alerts when a battery is connected that is not fully charged.
osdWarningBatteryWarning: Alerts when the battery voltage falls below warning threshold.
osdWarningBatteryCritical: Alters when the battery voltage falls below the minimum average cell voltage.
osdWarningVisualBeeper: Shows a visualisation of the beeper (shows as 4 asterisks).
osdWarningCrashFlipMode: Warns when flip over after crash mode is activated.

My request would be to only allow to copy "osdWarningBatteryCritical" into a separate element.

McGiverGim, 2021-04-07:
What about simply to put a voltage element, not a warning, at any part of the screen, to be able to see the actual state of the battery, and disable the battery warning letting only the land now (battery critical)? Is not that what you are looking for?

granolaFPV, 2021-04-07:
@McGiverGim I’m not the originator of this issue, but I’ll respond based on my own experience. My OSD setup already has battery voltage at the top center of the screen, but when I’m racing I find that even a quick glance at the voltage often results in making an error on the track (and sometimes crashing). I leave the battery warning enabled because “LOW BATTERY” is _usually_ my indicator that it’s time to land. However, sometimes I’m in the groove and trying to eek out another fast lap. In those instances, I’m waiting to see “LAND NOW” to signal that I should not continue flying. Having some way to better differentiate between these two OSD warnings would be helpful, as I want to have both enabled, but one to be out of the way, and the other to be smack dab in my field of view.

lowngcawt, 2021-05-04:
As a racer having LAND NOW as it's own thing would be golden, these days low batt warnings hit after half a lap and is just constantly blinking the whole pack and it's annoying but land now is super useful to know when to back off to be able to finish the lap!

MBoerlage, 2021-05-14:
Thanks all for the great feedback! Hope our wonderful betalight developers pick it up! If you are a developer and read this, please let me know if I can help in any way.

mikeller, 2021-05-20:
@lowngcawt: You know you can enable / disable all of the warnings individually, right? :grinning: 

[image]

lowngcawt, 2021-05-20:
@mikeller 🤯🤯🤯 How have i not seen that or thought of it!?

closed 2021-06-28 by github-actions[bot]