betaflight/betaflight #10674: GPS rescue [GALILEO] allowed range 5-50

sedlacekdavid, 2021-04-07:
### Describe the bug

Hi, not sure if this is a bug. There is an option **gps_rescue_min_sats** and possible values are in an interval of 5-50, however from what I understood, you can have a solid fix from 4 satellites and more using Galileo navigation system. 

I set protocol to Ublox, then checked "Use galileo"

### To Reproduce

Configuration->GPS->Ublox protocol & Use galileo

then in Failsafe->Minimum sattelites (gps_rescue_min_sats ) the minimum value is 5.

### Expected behavior

gps_rescue_min_sats minimum value = 4 ?

### Flight controller configuration

```
# version
# Betaflight / STM32F7X2 (S7X2) 4.2.8 Feb 15 2021 / 12:10:35 (101738d8e) MSP API: 1.43
# config: manufacturer_id: TMTR, board_name: TMOTORF7, version: e02dd6f2, date: 2020-11-04T11:31:26Z

# start the command batch
batch start

board_name TMOTORF7
manufacturer_id TMTR

# name: David

# resources
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 C06
resource MOTOR 4 C07

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS
feature ESC_SENSOR

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# serial
serial 1 1 115200 57600 0 115200
serial 2 2 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200
serial 4 64 115200 57600 0 115200

# led
led 0 0,0::C:1
led 1 1,0::C:1
led 2 2,0::C:1
led 3 3,0::C:1
led 4 4,0::C:1
led 5 6,0::C:1
led 6 7,0::C:1
led 7 8,0::C:1
led 8 9,0::C:1
led 9 10,0::C:1

# aux
aux 0 0 0 1500 2100 0 0
aux 1 2 0 1800 2100 0 0
aux 2 27 3 1800 2100 0 0
aux 3 46 3 1400 1600 0 0
aux 4 13 2 1800 2100 0 0
aux 5 35 1 1800 2100 0 0
aux 6 37 2 1400 1600 0 0

# master
set gyro_lowpass2_hz = 300
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 125
set dyn_notch_max_hz = 350
set dyn_lpf_gyro_min_hz = 240
set dyn_lpf_gyro_max_hz = 600
set acc_calibration = 10,-11,-11,1
set align_mag = CW270
set mag_align_yaw = 2700
set mag_declination = 456
set mag_calibration = 37,342,153
set baro_bustype = I2C
set baro_i2c_device = 2
set fpv_mix_degrees = 30
set serialrx_provider = SBUS
set sbus_baud_fast = ON
set blackbox_p_ratio = 64
set dshot_idle_value = 400
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 0
set failsafe_off_delay = 200
set failsafe_throttle = 1250
set failsafe_procedure = GPS-RESCUE
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 340
set battery_meter = ESC
set ibata_scale = 153
set ibata_offset = -1500
set beeper_dshot_beacon_tone = 4
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_initial_alt = 30
set gps_rescue_descent_dist = 50
set gps_rescue_ground_speed = 1000
set gps_rescue_throttle_max = 2000
set gps_rescue_ascend_rate = 400
set gps_rescue_throttle_hover = 1400
set gps_rescue_min_sats = 5
set gps_rescue_min_dth = 50
set osd_flymode_pos = 2062
set osd_current_pos = 2533
set osd_mah_drawn_pos = 2501
set osd_gps_speed_pos = 2540
set osd_gps_lon_pos = 2112
set osd_gps_lat_pos = 2080
set osd_gps_sats_pos = 2074
set osd_home_dir_pos = 2069
set osd_home_dist_pos = 2070
set osd_flight_dist_pos = 140
set osd_compass_bar_pos = 425
set osd_altitude_pos = 2099
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 2546
set osd_disarmed_pos = 2316
set osd_nvario_pos = 2543
set osd_esc_tmp_pos = 96
set osd_rtc_date_time_pos = 2048
set osd_flip_arrow_pos = 236
set debug_mode = GYRO_SCALED
set gyro_rpm_notch_harmonics = 2
set gyro_rpm_notch_min = 125
set name = David

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 84
set dyn_lpf_dterm_max_hz = 204
set dterm_lowpass2_hz = 180
set vbat_pid_gain = ON
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 25
set p_pitch = 69
set d_pitch = 40
set f_pitch = 181
set p_roll = 63
set d_roll = 37
set f_roll = 171
set p_yaw = 68
set f_yaw = 171
set d_min_roll = 24
set d_min_pitch = 26
set d_min_boost_gain = 35
set d_min_advance = 10
set idle_min_rpm = 17

rateprofile 0

# rateprofile 0
set roll_rc_rate = 127
set pitch_rc_rate = 127
set roll_expo = 40
set pitch_expo = 40
set roll_srate = 74
set pitch_srate = 74
set yaw_srate = 71
set tpa_rate = 75
set tpa_breakpoint = 1750

# end the command batch
batch end

```
```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: LED 1
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: FREE
B09: CAMERA_CONTROL
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: ADC_CURR
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: MOTOR 3
C07: MOTOR 4
C08: FLASH_CS
C09: FREE
C10: SERIAL_TX 3
C11: SERIAL_RX 3
C12: FREE
C13: BEEPER
C14: PINIO 1
C15: FREE
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH2: DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11:
    CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 3
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 
 - Flight controller: 
 
 - Other components:
 
 - How are the different components wired up:

McGiverGim, 2021-04-07:
The 5-50 is because it was decided that 5 is the minimum "safe" value to get a correct fix. In earlier versions 7 satellites was the minimum Betaflight had to decide it had a correct and safe fix.

With 4 satellites you have a fix, but maybe the precision is not good. We simply decided to use 5 to be more in the safe side.

Maybe someone can demonstrate that 4 is enough, then we can change it, but until then, I think 5 is safer.

sedlacekdavid, 2021-04-07:
Ok it perfectly makes sense, thank you for the explanation man.

closed 2021-04-07 by sedlacekdavid