betaflight/betaflight #10678: Iflight Succex-E F4 Smart Audio on TX1 doesn't work

nwpulver, 2021-04-09:
### Describe the bug
Smart audio does not work on TX1 on the iFlight Succex E F4 FC. I suspect it is a target issue. A few people on this thread https://intofpv.com/t-f4-succex-e-f405-tbs-unify-pro32-smartaudio-not-working have the same issue (as do I, have not commented on that forum though)  and the last user to comment mentioned that they switched to emuflight and that SA on UART1 worked so they  suspect it as an issue with the target. Others had this issue with TBS pro 32HV, mine is with the AKK FX2. I have also tried flashing to the latest firmware (4.2.8) from the stock firmware (4.2.4) 

Note that I initially had this same board earlier in the week with a faulty gyro from the factory. The one with the fault gyro was also not able to get SA on UART1. So this is my second board with the same issue. 

### To Reproduce
Wire SA to UART1 on a iFlight Succex-E F4 FC. 

### Expected behavior
Smart audio to work after imputing correct VTX table 

### Flight controller configuration

```

Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / STM32F405 (S405) 4.2.8 Feb 15 2021 / 12:08:36 (101738d8e) MSP API: 1.43
# config: manufacturer_id: IFRC, board_name: IFLIGHT_SUCCEX_E_F4, version: c7fde51e, date: 2020-08-26T04:30:34Z

# start the command batch
batch start

board_name IFLIGHT_SUCCEX_E_F4
manufacturer_id IFRC

# serial
serial 0 2048 115200 57600 0 115200

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 400 800

# master
set acc_calibration = -25,28,196,1
set serialrx_provider = CRSF
set dshot_bidir = ON
set small_angle = 180
set gyro_1_align_yaw = 1800

profile 0

rateprofile 0

# end the command batch
batch end

```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: BEEPER
B05: LED 1
B06: LED_STRIP
B07: FREE
B08: FREE
B09: FREE
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: MOTOR 4
C09: MOTOR 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: INVERTER 2
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4:
    CH1: LED_STRIP
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH2: DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 3
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Setup / Versions
 
 - Flight controller: 
 - iFlight Succex-E F4 V2.1 (with baro) from pyrodrone. 
 
 - Other components:
 
 - How are the different components wired up:
 - VTX: UART1
 - RX: UART 2
 - GPS: UART 4
 - ESC: UART 6
 - iFlight 45A ESC - Bluejay firmware, bidirectional Dshot enabled

JumaAnher, 2021-05-07:
Same issue for me. Any progress?

nwpulver, 2021-05-24:
@JumaAnher I contacted Iflight about the issue as well. They don't really seem interested.

haslinghuis, 2021-05-24:
Did you try another TX?

nwpulver, 2021-05-24:
@haslinghuis yep, works fine on a different TX on both boards I have. A feature of this board is that it has plugs for everything so it would still be nice to be able to use TX1 for SA as the manual and plugs imply.

haslinghuis, 2021-05-24:
@Linjieqiang: As the target maintainer for iFlight, can you please have a look into this? Perhaps just changing documentation or investigate why SA isn't working on TX1

closed 2021-07-02 by github-actions[bot]